tbsdtv/linux_media #200: Error compiling on Ubuntu 5.6.13-050613-generic x86_64

Nalyden, 2020-05-20:
Hi, I'm trying to compile on a Ubuntu system with a 5.6.13-050613-generic x86_64 kernel and I am getting the following errors when I run make:

  LD [M]  /home/dylan/tbsdriver/media_build/v4l/fm_drv.o
  LD [M]  /home/dylan/tbsdriver/media_build/v4l/altera-stapl.o
  MODPOST 729 modules
ERROR: "fwnode_get_name" [/home/dylan/tbsdriver/media_build/v4l/v4l2-fwnode.ko] undefined!
scripts/Makefile.modpost:93: recipe for target '__modpost' failed
make[3]: *** [__modpost] Error 1
Makefile:1596: recipe for target 'modules' failed
make[2]: *** [modules] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.6.13-050613-generic'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/dylan/tbsdriver/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

BytEvil, 2020-05-22:
Hi. Same issue on Arch Linux kernel 5.6.13.

Paco125689, 2020-05-22:
Hi. Same issue on Fedora 32  Linux kernel 5.6.13-300.fc32.x86_64

Tnix76, 2020-05-22:
Hi.
Same issue on Debian Testing: 5.6.0-1-amd64 #1 SMP Debian 5.6.7-1 (2020-04-29) x86_64 GNU/Linux

L-S-D, 2020-05-25:
as a workaround edit media_build/v4l/compat.h and search for:
#ifdef NEED_FWNODE_GETNAME 

remove the line you just searched and the following #endif 

Working fine for me on 5.6.14

Nalyden, 2020-05-26:
Cheers, that's got it compiling for me.

closed 2020-05-26 by Nalyden

Tnix76, 2020-05-26:
Doesn't work for me in 5.6.0

zhang111chao, 2020-05-26:
hi ,

Please follow the steps below to install driver for tbs card .
#wget https://www.tbsdtv.com/download/document/linux/media_build-2020-05-22.tar.bz2
#tar jxvf media_build-2020-05-22.tar.bz2
#cd media_build
#./install.sh
#reboot

best regards!

jack@tbsdtv.com
 
From: Tnix76
Date: 2020-05-26 14:36
To: tbsdtv/linux_media
CC: Subscribed
Subject: Re: [tbsdtv/linux_media] Error compiling on Ubuntu 5.6.13-050613-generic x86_64 (#200)
Doesn't work for me in 5.6.0

Tnix76, 2020-05-26:
As far as I know the TBS official drivers doesn't support satellite, it isn't?

Tnix76, 2020-05-26:
Fails again to compile:

ERROR: "__devm_regmap_init_sccb" [/home/myuser/media_build/v4l/ov9650.ko] undefined!

zhang111chao, 2020-05-26:
hi ,
   I have successfully installed on ubuntu centos v5.6.0 v5.6.13 .
    please setup ssh connection or teamviewer , i will check,  

best regards!

jack@tbsdtv.com
 
From: Tnix76
Date: 2020-05-26 15:06
To: tbsdtv/linux_media
CC: zhang111chao; Comment
Subject: Re: [tbsdtv/linux_media] Error compiling on Ubuntu 5.6.13-050613-generic x86_64 (#200)
Fails again to compile:
ERROR: "__devm_regmap_init_sccb" [/home/myuser/media_build/v4l/ov9650.ko] undefined!

BytEvil, 2020-05-26:
@L-S-D, problem **NOT** solved...

L-S-D, 2020-05-26:
well I assume the original problem from the log was solved, depending on the exact OS and kernel, there might be more errors, like Tnix76 ov9650.ko undefined.
Anyway, that was just a dirty workaround to pass the error mentioned, I assume a proper fix will be available soon from TBS.

zhang111chao, 2020-05-27:
Please follow the steps below to install driver for tbs card .
#wget https://www.tbsdtv.com/download/document/linux/media_build-2020-05-22.tar.bz2
#tar jxvf media_build-2020-05-22.tar.bz2
#cd media_build
#./install.sh
#reboot

Paco125689, 2020-05-27:
This works for me
`Please follow the steps below to install driver for tbs card .
#wget https://www.tbsdtv.com/download/document/linux/media_build-2020-05-22.tar.bz2
#tar jxvf media_build-2020-05-22.tar.bz2
#cd media_build
#./install.sh
#reboot`
Thank you very much!

jogrue, 2020-06-09:
> 
> 
> Please follow the steps below to install driver for tbs card .
> #wget https://www.tbsdtv.com/download/document/linux/media_build-2020-05-22.tar.bz2
> #tar jxvf media_build-2020-05-22.tar.bz2
> #cd media_build
> #./install.sh
> #reboot

I had to uncomment the memtype.h kernel patch (this line in patch-kernel.sh: `#"v5.6_memtype_h.patch"`. Otherwise, I would receive this error:

```
media_build/v4l/ivtvfb.c:40:10: fatal error: asm/memtype.h: No such file or directory
```

After adding a few sudo calls in install.sh, I could build the driver again.

zhang111chao, 2020-06-10:
#ifdef CONFIG_X86_64
//#include <asm/memtype.h>
#endif

2. make 
3. make install