tbsdtv/linux_media #202: Unable to obtain signal lock with TBS6281SE

Nalyden, 2020-05-26:
Hi, I have two TBS6281SE cards in a Ubuntu box with kernel version 5.6.13-050613-generic x86_64. I'm trying to access free to air TV in Australia using DVB-T.

I'm running MythTV but my cards seem to be having reception issues. Running a channel scan reliably finds most channels but there are a couple transponders that won't tune reliably. Once in MythTV I can't get a signal lock on any of the available channels and their reported strength sits around 40-50%. 

Testing with a standard TV using the same cables and antenna point allows me to successfully tune all available channels with the TV reporting 100% signal strength (though I know this is a bit of a fuzzy value).

Testing with w_scan gets similar results with only two of the five available transponders tuning reliably.

zhang111chao, 2020-05-26:
hi ,

     Please establish a remote connection via ssh or teamviewe, I will check .
    Please add my skype account id jack_2431 . 

best regards!

jack@tbsdtv.com
 
From: Nalyden
Date: 2020-05-26 09:59
To: tbsdtv/linux_media
CC: Subscribed
Subject: [tbsdtv/linux_media] Unable to obtain signal lock with TBS6281SE (#202)
Hi, I have two TBS6281SE cards in a Ubuntu box with kernel version 5.6.13-050613-generic x86_64. I'm trying to access free to air TV in Australia using DVB-T.
I'm running MythTV but my cards seem to be having reception issues. Running a channel scan reliably finds most channels but there are a couple transponders that won't tune reliably. Once in MythTV I can't get a signal lock on any of the available channels and their reported strength sits around 40-50%.
Testing with a standard TV using the same cables and antenna point allows me to successfully tune all available channels with the TV reporting 100% signal strength (though I know this is a bit of a fuzzy value).
Testing with w_scan gets similar results with only two of the five available transponders tuning reliably.

allaboutmikey, 2020-07-14:
Did you get to the bottom of this? I have a 6281 as well (can't remember if it's SE or not, pciid is 6281:0002) and since I tried to update my kernel from 4.x to 5.x, I can't get the cards to tune anymore either.
The kernel modules load ok and the cards (I have two) are detected, they just won't tune in mythtv. :(
Also in Australia using DVB-T, btw.

Nalyden, 2020-07-14:
Hi Mikey,

Yeah, we did, I just completely forgot to update here.

Zhang111chao was able to tune by manually specifying the parameters in tzap (at least from memory it was tzap, could be wrong there). But I was still unable to reliably lock on to transponders using w_scan or MythTV. What I found was that MythTV wasn't detecting the broadcast standard properly, in each of the transponders it had selected "Unknown" (I think? Again my memory is rusty) instead of DVB-T. Once I set that manually it was all good to go.

allaboutmikey, 2020-07-14:
I just found something that maybe related half an hour ago. In MythSetup there is a selection in the capture card section that had to be made (in my just updated version 31) that wasn't there in previous versions (~ver27). It was set to DVBInput iirc, which looked fine to my eye. When I finally got desperate enough to go in there to look at the alternatives, I found that DVBInput wasn't one of the choices. It gave me DVB-T, DB-T2, DVB-S (and maybe DVB-C?). I chose DVB-T and now mythtvt works on the previously scanned channels. I haven't bothered to try scanning in MythSetup as my list of channels was already up to date.
I don't know if that selection fixes the broadcast standard detection or not, but it fixed my problem which sounded very similar to yours.