tbsdtv/linux_media #205: TBS 6985 problem with signal

semering, 2020-06-06:
I have 2 TBS cards - Linux 18.04 ( 6985 sat + 6285 cable )
Signals on cable adapters are always locked ( Astra Cesbo 5.64 ) and working perfect. But signals, or channels are dropping all the time. One time, signals are OK on all 4 adapters, but the channels go on and off. Other time, 1 of the 4 adapters is just gone. I think that on the video I can send to somebody, everything will be clear to somebody who knows the problems with astra+TBS.

[image]
[image]
[image]

zhang111chao, 2020-06-08:
hi ,
    please add my skype jack_2431 . 
    or write email to jack@tbsdtv.com .    
best regards!

egrekov, 2020-09-09:
Good day!
We also have 6985 cards, and the same problem after restarting the astra, we have to restart the adapters manually. Can you have a recipe to cure this sore?