tbsdtv/linux_media #206: ADD THIS DRIVER IN RPMFusion REPO ...

sajjads24, 2020-06-06:
hi
please add this driver like rpm package in rpmfusion for fedora 32 or ...?
for example :
dnf install TBS-Frimware
it's very good and very easy for install .
please add it to rpmfusion.

thanks

zhang111chao, 2020-06-08:
hi ,
     sorry,  It's hard to do that . The installation of each driver needs to match the current kernel version .  
Please follow the steps below to install driver for tbs6209 card .
#wget https://www.tbsdtv.com/download/document/linux/media_build-2020-05-22.tar.bz2
#tar jxvf media_build-2020-05-22.tar.bz2
#cd media_build
#./install.sh
#reboot
best regards!

CvH, 2020-06-08:
> sorry, It's hard to do that

Not really, just start to mainline the drivers to the linux kernel to achieve proper linux support without hacks like media_build.