tbsdtv/linux_media #207: Cannot install tbs drivers on Raspberry Pi 3 with Raspbian and latest kernel 4.19.118-v8+

matej86, 2020-06-16:
I followed your instructions in order to install tbs drivers on Raspberry Pi 3 (Raspbian) with latest kernel 4.19.118-v8+

git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
make -j4
sudo make install
sudo reboot

I get stuck at "make allyesconfig" and it doesn't matter if I use sudo in front or not. Here is the log:

After sudo make allyesconfig:

`pi@tvserver:~/media_build $ sudo make allyesconfig
make -C /home/pi/media_build/v4l allyesconfig
make[1]: Entering directory '/home/pi/media_build/v4l'
No version yet, using 4.19.118-v8+
make[2]: Entering directory '/home/pi/media_build/linux'
Syncing with dir ../../media
Applying patches for kernel 4.19.118-v8+
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/v5.5_alsa_pcm_api_updates.patch
patch -s -f -N -p1 -i ../backports/v5.6_pin_user_pages.patch
patch -s -f -N -p1 -i ../backports/v5.6_memtype_h.patch
patch -s -f -N -p1 -i ../backports/v5.6_const_fb_ops.patch
patch -s -f -N -p1 -i ../backports/v5.4_revert_spi_transfer.patch
patch -s -f -N -p1 -i ../backports/v5.1_vm_map_pages.patch
patch -s -f -N -p1 -i ../backports/v5.1_devm_i2c_new_dummy_device.patch
patch -s -f -N -p1 -i ../backports/v5.0_ipu3-cio2.patch
patch -s -f -N -p1 -i ../backports/v5.0_time32.patch
patch -s -f -N -p1 -i ../backports/v4.20_access_ok.patch
Patched drivers/media/dvb-core/dvbdev.c
Patched drivers/media/v4l2-core/v4l2-dev.c
Patched drivers/media/rc/rc-main.c
Syncing with dir ../../media
make[2]: Leaving directory '/home/pi/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/4.19.118-v8+/build /lib/modules/4.19.118-v8+/build 1
Preparing to compile for kernel version 4.19.118
File not found: /lib/modules/4.19.118-v8+/build/.config at ./scripts/make_kconfig.pl line 33, <IN> line 4.
make[1]: *** [Makefile:388: allyesconfig] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2`

Without sudo in front

`pi@tvserver:~/media_build $ make allyesconfig
make -C /home/pi/media_build/v4l allyesconfig
make[1]: Entering directory '/home/pi/media_build/v4l'
make[2]: Entering directory '/home/pi/media_build/linux'
Syncing with dir ../../media
Patches for 4.19.118-v8+ already applied.
make[2]: Leaving directory '/home/pi/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/4.19.118-v8+/build /lib/modules/4.19.118-v8+/build 1
Preparing to compile for kernel version 4.19.118
File not found: /lib/modules/4.19.118-v8+/build/.config at ./scripts/make_kconfig.pl line 33, <IN> line 4.
make[1]: *** [Makefile:388: allyesconfig] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2`

How can I solve the problem? Am I missing something. I installed all dipendencies

sudo apt-get install libproc-processtable-perl patchutils libelf-dev git make libdigest-sha-perl build-essential raspberrypi-kernel-headers

matej86, 2020-06-16:
One more thing: I can install these drivers with the same commands on a Ubuntu 20.04 system without a problem

zhang111chao, 2020-06-16:
Please follow the steps below to install driver for tbs card .

#apt-get install raspberrypi-kernel-headers
#apt-get install patchutils
#apt-get install libproc-processtable-perl
# wget http://www.tbsdtv.com/download/document/linux/media_build-rpi-2020-01-02.tar.bz2
# sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/
#tar jxvf media_build-rpi-2020-01-02.tar.bz2
#cd media_build 
#./install.sh 
#reboot

zhang111chao, 2020-06-16:
Raspberry Pi 3 has  kernel 4.19.118-v8,  but Ubuntu 20.04 has another . 
Driver installation is related to kernel version 。

matej86, 2020-06-16:
I followed your instruction and the problem remains. 

Here is the log:

`pi@tvserver:~/media_build $ sudo ./install.sh
1 out of 1 hunk FAILED -- saving rejects to file include/media/v4l2-clk.h.rej
Applying patches for kernel 4.19.118-v8+
4 19 118
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/ivtv/ivtv-udma.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/ivtv/ivtv-yuv.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/v4l2-core/videobuf-dma-sg.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/videobuf2/videobuf2-memops.c.rej
dvb-demod-avl6882.fw
dvb-demod-drxk-pctv.fw
dvb-demod-gx1503B.fw
dvb-demod-mn88472-02.fw
dvb-demod-mn88473-01.fw
dvb-demod-si2168-01.fw
dvb-demod-si2168-02.fw
dvb-demod-si2168-a20-01.fw
dvb-demod-si2168-a30-01.fw
dvb-demod-si2168-b40-01.fw
dvb-demod-si2183-b60-01.fw
dvb-fe-bcm3510-01.fw
dvb-fe-cx24116.fw
dvb-fe-cx24117.fw
dvb-fe-drxj-mc-1.0.8.fw
dvb-fe-drxj-mc-vsb-1.0.8.fw
dvb-fe-drxj-mc-vsb-qam-1.0.8.fw
dvb-fe-ds3000.fw
dvb-fe-ds300x.fw
dvb-fe-ds3103.fw
dvb-fe-mxl5xx.fw
dvb-fe-or51132-qam.fw
dvb-fe-or51132-vsb.fw
dvb-fe-or51211.fw
dvb-fe-rs6000.fw
dvb-fe-tda10045.fw
dvb-fe-tda10046.fw
dvb-fe-tda10048-1.0.fw
dvb-fe-tda10071.fw
dvb-fe-xc4000-1.4.1.fw
dvb-fe-xc5000-1.6.114.fw
dvb-fe-xc5000c-4.1.30.7.fw
dvb-tuner-si2158-a20-01.fw
dvb-usb-avertv-a800-02.fw
dvb-usb-bluebird-01.fw
dvb-usb-dib0700-1.20.fw
dvb-usb-dibusb-5.0.0.11.fw
dvb-usb-dibusb-6.0.0.8.fw
dvb-usb-dtt200u-01.fw
dvb-usb-dw2101.fw
dvb-usb-dw2102.fw
dvb-usb-dw2104.fw
dvb-usb-dw3101.fw
dvb-usb-id5301.fw
dvb-usb-id5520se.fw
dvb-usb-id5580.fw
dvb-usb-id5590.fw
dvb-usb-it9135-01.fw
dvb-usb-it9135-02.fw
dvb-usb-p1100.fw
dvb-usb-p1100-old.fw
dvb-usb-p7500.fw
dvb-usb-pctv-400e-01.fw
dvb-usb-pctv-450e-01.fw
dvb-usb-pctv-452e-01.fw
dvb-usb-s620.fw
dvb-usb-s630.fw
dvb-usb-s660.fw
[... 93 lines cut ...]
Can't find ../../media/Makefile at ./use_dir.pl line 287.
Syncing with dir ../../media
Can't find ../../media/Makefile at ./use_dir.pl line 287.
make[2]: *** [Makefile:132: apply_patches] Error 2
make[2]: Leaving directory '/home/pi/media_build/linux'
make[2]: *** [Makefile:132: apply_patches] Error 2
make[2]: Leaving directory '/home/pi/media_build/linux'
Preparing to compile for kernel version 4.19.118
File not found: /lib/modules/4.19.118-v8+/build/.config at ./scripts/make_kconfig.pl line 33, <IN> line 4.
make[1]: *** No rule to make target '.config', needed by '.myconfig'.  Stop.
make[1]: Leaving directory '/home/pi/media_build/v4l'
make: *** [Makefile:26: all] Error 2
V4L drivers installing...
make -C /home/pi/media_build/v4l install
make[1]: Entering directory '/home/pi/media_build/v4l'
Installing kernel modules under /lib/modules/4.19.118-v8+/kernel/drivers/media/:
/sbin/depmod -a 4.19.118-v8+ 
make -C firmware install
make[2]: Entering directory '/home/pi/media_build/v4l/firmware'
Installing firmwares at /lib/firmware: 
make[2]: Leaving directory '/home/pi/media_build/v4l/firmware'
make[1]: Leaving directory '/home/pi/media_build/v4l'
V4L drivers installation done
You need to reboot...
pi@tvserver:~/media_build $ `

zhang111chao, 2020-06-16:
please add skype jack_2431 .

matej86, 2020-06-16:
why skype?

matej86, 2020-06-16:
note i installed latest raspbian today, so the system is clean

zhang111chao, 2020-06-16:
i need a remote  control to check the issue . ssh connection or anydesk .

matej86, 2020-06-16:
Sorry, I cannot do it due to IT security policy.

If do you want me to enter some commands I can do it and show you the logs

matej86, 2020-06-17:
I suppose the problem is that dirver are not supported for the latest kernel 4.19.118-v7+?

`pi@servertv:~/media_build $ sudo make -j4
make -C /home/pi/media_build/v4l 
make[1]: Entering directory '/home/pi/media_build/v4l'
**No version yet, using 4.19.118-v7+**
scripts/make_makefile.pl
Updating/Creating .config
make[2]: Entering directory '/home/pi/media_build/linux'
make[2]: Entering directory '/home/pi/media_build/linux'
Syncing with dir ../../media
Syncing with dir ../../media
Applying patches for kernel 4.19.118-v7+
Applying patches for kernel 4.19.118-v7+
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/api_version.patch
mv: cannot stat '.patches_applied': No such file or directory
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/v5.5_alsa_pcm_api_updates.patch
patch -s -f -N -p1 -i ../backports/debug.patch
1 out of 1 hunk FAILED
make[2]: *** [Makefile:133: apply_patches] Error 1
make[2]: Leaving directory '/home/pi/media_build/linux'
patch -s -f -N -p1 -i ../backports/v5.6_pin_user_pages.patch
patch -s -f -N -p1 -i ../backports/v5.6_memtype_h.patch
patch -s -f -N -p1 -i ../backports/v5.6_const_fb_ops.patch
patch -s -f -N -p1 -i ../backports/v5.4_revert_spi_transfer.patch
patch -s -f -N -p1 -i ../backports/v5.1_vm_map_pages.patch
patch -s -f -N -p1 -i ../backports/v5.1_devm_i2c_new_dummy_device.patch
patch -s -f -N -p1 -i ../backports/v5.0_ipu3-cio2.patch
patch -s -f -N -p1 -i ../backports/v5.0_time32.patch
patch -s -f -N -p1 -i ../backports/v4.20_access_ok.patch
Patched drivers/media/dvb-core/dvbdev.c
Patched drivers/media/v4l2-core/v4l2-dev.c
Patched drivers/media/rc/rc-main.c
Syncing with dir ../../media
make[2]: Leaving directory '/home/pi/media_build/linux'`

4l3x2k, 2020-06-17:
> I followed your instruction and the problem remains.
>
> Here is the log:
>
> `pi@tvserver:~/media_build $ sudo ./install.sh 1 out of 1 hunk FAILED -- saving rejects to file include/media/v4l2-clk.h.rej [...]

Your problem here is:
- Preparing to compile for kernel version 4.19.118 File not found: /lib/modules/4.19.118-v8+/build/.config at ./scripts/make_kconfig.pl line 33, <IN> line 4.

The solution to this would be installing raspberrypi-kernel-headers like @zhang111chao mentioned. But these headers do not match the installed kernel most of the time. That is a big miss by raspbian, but I think they are working on it.

The no version yet comes from script that applies required patches and states that it has no version detected so far. Should not be fatal.

Try to setup a clean environment - at least do an make distclean - and do what @zhang111chao said. Try to use the provided archive and not the branch from git.

4l3x2k, 2020-06-17:
And please provide line breaks as your output is difficult to read. @matej86

caveman99, 2020-06-17:
I can confirm the driver works on x64 with way newer versions of kernel (i am using it with kernel-ml from elrepo on centos7) The raspberry specific problem is indeed the mismatch betwen headers and actual installed kernel. Many embedded platforms suffer from this, cause the boot process is completely different from a desktop pc. At the moment i am trying to find the headers for ubuntu core xenial on "Linux NanoPi-R1 4.14.111 #1 SMP Sat Dec 28 01:30:07 CST 2019 armv7l armv7l armv7l GNU/Linux"

matej86, 2020-06-17:
Dear @4l3x2k and @caveman99 thank you for your advices. I already tried all the @zhang111chao solutions, but they didn't solve the problem :(

Searching in google I found the following Raspbian Kernel Headers: https://packages.debian.org/stable/kernel/linux-headers-4.19.0-9-all-armhf

These seems to be kernel headers for 4.19.118 version too

I also found the specific Raspberry version     http://tardis.tiny-vps.com/aarm/packages/l/linux-raspberrypi-headers/linux-raspberrypi-headers-4.19.118-1-armv7h.pkg.tar.xz

Should I install one of them?

4l3x2k, 2020-06-17:
I would ignore the packages from debian or raspbian because they are most of the times too old.

You could try the second one but that is a .pkg and no .deb. But I bet that these do not match the running kernel.

The most clean solution is building the entire kernel yourself.

matej86, 2020-06-17:
@4l3x2k  The most clean solution is building the entire kernel yourself.   I am really not able to do it :(

4l3x2k, 2020-06-17:
@matej86
Most people are not able to do it or at least do not want to do.
There is a howto on the raspberrypi.org site.

matej86, 2020-06-17:
Maybe I will try to do it, but in the meanwhile I entered into the /usr/src/ folder and found the Raspberry Kernel Headers folders which seem to match the kernel versions... really strange??

[image]

I don't understand :(

4l3x2k, 2020-06-17:
If you installed raspberrypi-kernel-headers it is possible they are available. That's what the package is inteded for. Check whether .config is in /usr/src/linux-headers-$(uname -r) and the link /lib/modules/$(uname -r)/build points to that directory.

matej86, 2020-06-22:
Dear @All 

I retied again this morning and now raspberry kernel headers has been updated and now the installation of the drivers completed successfully.

Thank you for your help!

dawiepoolman, 2020-07-19:
Hi guys

I am trying to install on a VM running Ubuntu 20.04 (kernel 5.4.0-40-generic) on RPi4

I get these (last few lines):

/home/cypher/media_build/v4l/coda-jpeg.c:1218:2: error: implicit declaration of function 'trace_coda_jpeg_done' [-Werror=implicit-function-declaration]
 1218 |  trace_coda_jpeg_done(ctx, dst_buf);
      |  ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:273: /home/cypher/media_build/v4l/coda-jpeg.o] Error 1
make[2]: *** [Makefile:1731: /home/cypher/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-40-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/cypher/media_build/v4l'
make: *** [Makefile:26: all] Error 2

I have not installed the from the shared raspberry pi files in this thread, I have used the default wiki because I presume the VM OS to be the determining factor?

Thx for the help