tbsdtv/linux_media #208: UEFI compatible drivers

Saentist, 2020-07-07:
Is there a way TBS team to write **TBS.EFI** module, with will guarantee loading of module when UEFI mode used.
I understand that this will can be a paint for compilation because signing is needed to modules.

Else is it possible just to make some TBS public certificate for signing modules and using **mokutil** to add signature to UEFI during compilation.

Soon all servers will support only UEFI, not shure with or without option to disable **SECURE BOOT** and curent driver will stay useless.

crazycat69, 2020-11-04:
Modern linux distros have signed EFI loader (Microsoft first stage loader and distro vendor for second stage) and not check module signatures. Confirmed for *buntu 20.04 - work also with secure boot enabled.

davidjwbbc, 2020-11-12:
I've done this before:

Follow the instructions in [https://www.kernel.org/doc/html/v4.15/admin-guide/module-signing.html](https://www.kernel.org/doc/html/v4.15/admin-guide/module-signing.html) to create your own module signing key and import the public key using openssl to convert to DER and mokutil to import it. Then sign your compiled module with the $KERNEL_SRC/scripts/sign-file.

I tweaked the v4l/scripts/make_makefile.pl to check for a signing key and sign the module on install using this patch:
------------------------------------------------------------------------------------------------
diff --git a/v4l/scripts/make_makefile.pl b/v4l/scripts/make_makefile.pl
index 6f51209..60a2bd7 100755
--- a/v4l/scripts/make_makefile.pl
+++ b/v4l/scripts/make_makefile.pl
@@ -255,6 +255,9 @@ while (my ($dir, $files) = each %instdir) {
        print OUT "install -m 644 -c \$\$i \$(DESTDIR)\$(KDIR26)/$dir; fi; done; ";
        print OUT "if [  \$\$n -ne 0 ]; then echo; ";
        print OUT "strip --strip-debug \$(DESTDIR)\$(KDIR26)/$dir/*.ko; ";
+        print OUT "if [ -e /lib/modules/\$(KERNELRELEASE)/build/certs/signing_key.pem ]; then ";
+        print OUT "for i in \$(DESTDIR)\$(KDIR26)/$dir/*.ko; do /lib/modules/\$(KERNELRELEASE)/build/scripts/sign-file sha256 /lib/modules/\$(KERNELRELEASE)/build/certs/signing_key.pem /lib/modules/\$(KERNELRELEASE)/build/certs/signing_key.pem \"\$\$i\"; done; ";
+       print OUT "fi; ";
        print OUT "fi;\n\n";
 }
 print OUT "\t@echo\n";
------------------------------------------------------------------------------------------------

That way it automatically signs when I rebuild for a new kernel.

Saentist, 2020-11-18:
Only negative of this that if it's server need IPMI/KVM to accept inserting in UEFI key store.

davidjwbbc, 2020-11-19:
No, you only have to insert the key once into UEFI and can reuse the same key for multiple builds. What I forgot to mention above was another /etc/kernel/postinst script which copies the key files from the existing kernel source directory to the new kernel source directory when the new kernel package is installed.

Saentist, 2020-11-19:
Server in data centre need to import key in UEFI, yes once but...