tbsdtv/linux_media #209: [TBS 5520SE][voidlinux] disagrees about version of symbol rc_register_device

lschmelting, 2020-07-25:
I could not use newer kernel versions, because i would get this error:
```
ERROR: "__devm_regmap_init_sccb" [/home/user/TBS/media_build/v4l/ov9650.ko] undefined!
```

So I used these instructions to install the driver:
```
# wget http://www.tbsdtv.com/download/document/linux/media_build-2020-05-22.tar.bz2
# sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/
#tar jxvf media_build-2020-05-22.tar.bz2
#cd media_build
#./install.sh
#reboot 
```
But i get the following output:

**Output:**
USB2.0 port:

```
[   39.348639] usb 1-1.2: new high-speed USB device number 5 using ehci-pci
[   39.996293] dvb_usb: disagrees about version of symbol rc_register_device
[   39.996295] dvb_usb: Unknown symbol rc_register_device (err -22)
[   39.996356] dvb_usb: disagrees about version of symbol rc_free_device
[   39.996358] dvb_usb: Unknown symbol rc_free_device (err -22)
[   39.996392] dvb_usb: disagrees about version of symbol rc_allocate_device
[   39.996393] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[   39.996460] dvb_usb: disagrees about version of symbol rc_unregister_device
[   39.996461] dvb_usb: Unknown symbol rc_unregister_device (err -22)
```
USB3.0 port:
```
[   95.285408] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[   95.956360] dvb_usb: disagrees about version of symbol rc_register_device
[   95.956377] dvb_usb: Unknown symbol rc_register_device (err -22)
[   95.956446] dvb_usb: disagrees about version of symbol rc_free_device
[   95.956447] dvb_usb: Unknown symbol rc_free_device (err -22)
[   95.956483] dvb_usb: disagrees about version of symbol rc_allocate_device
[   95.956484] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[   95.956553] dvb_usb: disagrees about version of symbol rc_unregister_device
[   95.956553] dvb_usb: Unknown symbol rc_unregister_device (err -22)
```
So i searched the forums for a solution, and it was recommended to execute:
```
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/media
sudo make install
```
-> this statement is already present in the install.sh file, which i used (see first code block). If i only run the two commands above, it does not make a difference.

What can i do next?

zhang111chao, 2020-07-27:
please add skype jack_2431 ,

pkorpine, 2021-02-16:
I solved this by disabling all CONFIG items that had "RC" (i.e. remote control). Probably better option would be to use `make menuconfig` to select only the needed options. Running Void with kernel 5.10.16_1.

```sh
# Clone repositories
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media

cd media_build
make dir DIR=../media

# Disable certain modules that do not compile
sed -i '/VIDEO_OV9650/d' ./v4l/versions.txt && sed -i '/9.255.255/a VIDEO_OV9650' ./v4l/versions.txt
sed -i '/VIDEO_OV772X/d' ./v4l/versions.txt && sed -i '/9.255.255/a VIDEO_OV772X' ./v4l/versions.txt

make allyesconfig

# Disable all *_RC* config items
# (without this I got symbol errors for rc_register_device etc)
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config

make -j12

sudo make install
```

gadom92, 2021-06-05:
@pkorpine without your script i gry RC error after installing, but when I use your script i get error 
ir-mce kbd-decoder.ko undefined

JonathanMegevand, 2021-07-25:
> @pkorpine without your script i gry RC error after installing, but when I use your script i get error
> "lirc_scancode_event" ir-mce kbd-decoder.ko undefined

 sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config

solved your issue 
but i after i got 
[  +0.000715] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[  +0.000109] usb 1-1.3: Direct firmware load for dvb-usb-id5520se.fw failed with error -2
[  +0.000015] dvb-usb: did not find the firmware file 'dvb-usb-id5520se.fw' (status -2). You can use <kernel_dir>/scripts/get_dvb_firmware to get the firmware

JonathanMegevand, 2021-07-25:
Just missing one step wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
sudo tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

JonathanMegevand, 2021-07-25:
# What i do to have the tuner working is all below following @pkorpine message

# Clone repositories
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media

cd media_build
make dir DIR=../media

# Disable certain modules that do not compile
sed -i '/VIDEO_OV9650/d' ./v4l/versions.txt && sed -i '/9.255.255/a VIDEO_OV9650' ./v4l/versions.txt
sed -i '/VIDEO_OV772X/d' ./v4l/versions.txt && sed -i '/9.255.255/a VIDEO_OV772X' ./v4l/versions.txt

make allyesconfig

# Disable all *_RC* config items
# (without this I got symbol errors for rc_register_device etc)
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.=)./\1n/g' v4l/.config

make -j12

sudo make install

wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
sudo tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

sudo reboot

xanfia, 2022-10-20:
@JonathanMegevand be warned that Markup Language is being interpreted in your comments and there are two * missing per line:
`sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config`
`sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config`