tbsdtv/linux_media #211: Compilation failed with linux 5.8 (ArchLinux)

LinLo, 2020-08-10:
Hello,

I don't know if this linux media is synced with Linux 5.8 yet.

Compilation failed (latest branch) with : 
```
/home/user/tbs-linux_media-git/src/media_build/v4l/tbsecp3-dvb.c: In function 'tbsecp3_frontend_attach':
/home/user/tbs-linux_media-git/src/media_build/v4l/tbsecp3-dvb.c:1077:18: error: implicit declaration of function 'i2c_new_device'; did you mean 'first_net_device'? [-Werror=implicit-function-declaration]
 1077 |   client_demod = i2c_new_device(i2c, &info);
      |                  ^~~~~~~~~~~~~~
      |                  first_net_device

```

Kernel ver 5.8.0-arch1-2

Thank you for your help

ich777, 2020-08-10:
The same also happens on 5.7.8

@LinLo Can you change the title to something like 'Compilation failed on Linux Kernel version 5.7.8 and up'?

```
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c: In function 'tbs6302se_read_mac':
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:164:22: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
  164 |   if((tmpbuf[0]&0x4) == 1) // bit2==1 mcu busy
      |                      ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c: In function 'tbs6304_read_mac':
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:232:22: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
  232 |   if((tmpbuf[0]&0x4) == 1) // bit2==1 mcu busy
      |                      ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c: In function 'tbs6308_read_mac_ext':
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:338:22: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
  338 |   if((tmpbuf[0]&0x4) == 1) // bit2==1 mcu busy
      |                      ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:308:22: warning: unused variable 'dev' [-Wunused-variable]
  308 |  struct tbsecp3_dev *dev = adap->dev;
      |                      ^~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c: In function 'tbs6301_read_mac':
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:401:22: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
  401 |   if((tmpbuf[0]&0x4) == 1) // bit2==1 mcu busy
      |                      ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c: In function 'tbsecp3_frontend_attach':
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:1196:3: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
 1196 |   if (adapter->fe == NULL)
      |   ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:1198:7: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
 1198 |       if(adapter->nr <4)
      |       ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:1238:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 1238 |   int regdata;
      |   ^~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c: In function 'start_dma_transfer':
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c:1359:6: warning: unused variable 'bitrate' [-Wunused-variable]
 1359 |  u32 bitrate;
      |      ^~~~~~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c:1358:6: warning: unused variable 'delay' [-Wunused-variable]
 1358 |  u32 delay;
      |      ^~~~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c: In function 'tbsmod_open':
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c:1403:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 1403 |  u8 buff[4] = {0,0,0,0};
      |  ^~
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c: In function 'tbsmod_write':
At top level:
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:808:12: warning: 'max_send_burst' defined but not used [-Wunused-function]
  808 | static int max_send_burst(struct dvb_frontend *fe, enum fe_sec_mini_cmd burst)
      |            ^~~~~~~~~~~~~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsecp3-dvb.c:803:12: warning: 'max_send_master_cmd' defined but not used [-Wunused-function]
  803 | static int max_send_master_cmd(struct dvb_frontend *fe, struct dvb_diseqc_master_cmd *cmd)
      |            ^~~~~~~~~~~~~~~~~~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c:1480:21: warning: format '%d' expects argument of type 'int', but argument 3 has type 'size_t' {aka 'long unsigned int'} [-Wformat=]
 1480 |    printk("%s size:%d  %d\n", __func__, size, copied);
      |                    ~^                   ~~~~
      |                     |                   |
      |                     int                 size_t {aka long unsigned int}
      |                    %ld
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c: In function 'spi_write':
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c:1491:16: warning: unused variable 'tmpbuf' [-Wunused-variable]
 1491 |  unsigned char tmpbuf[4];
      |                ^~~~~~
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c: In function 'tbsmod_ioctl':
/usr/src/TBS-OpenSource/media_build/v4l/tbsmod.c:1705:16: warning: passing argument 1 of 'copy_to_user' makes pointer from integer without a cast [-Wint-conversion]
 1705 |   copy_to_user((unsigned long)arg, &finfo, sizeof(struct dvb_frontend_info));
      |                ^~~~~~~~~~~~~~~~~~
      |                |
      |                long unsigned int
In file included from ./include/linux/sched/task.h:11,
[... 76 lines cut ...]
/usr/src/TBS-OpenSource/media_build/v4l/tbsci.c:127:21: warning: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'size_t' {aka 'long unsigned int'} [-Wformat=]
  127 |    printk("%s size:%u  %u\n", __func__, size, copied);
      |                    ~^                   ~~~~
      |                     |                   |
      |                     unsigned int        size_t {aka long unsigned int}
      |                    %lu
/usr/src/TBS-OpenSource/media_build/v4l/tbsci.c: In function 'ts_read':
/usr/src/TBS-OpenSource/media_build/v4l/tbsci.c:168:21: warning: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'size_t' {aka 'long unsigned int'} [-Wformat=]
[... cut ...]
/usr/src/TBS-OpenSource/media_build/v4l/tbsci.c:227:6: warning: unused variable 'clk_freq' [-Wunused-variable]
  227 |  u32 clk_freq;
      |      ^~~~~~~~
  CC [M]  /usr/src/TBS-OpenSource/media_build/v4l/ivtv-ioctl.o
/usr/src/TBS-OpenSource/media_build/v4l/tbsci.c:233:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
  233 |   if (props.num == 1)
      |      ^
/usr/src/TBS-OpenSource/media_build/v4l/tbsci.c:279:2: note: here
  279 |  case FE_ECP3FW_READ:
      |  ^~~~
```

piotrasd, 2020-08-11:
5.7.8 failing also

crazycat69, 2020-08-12:
Please check again
13f046347f561426d6b327c0e7808fd410d0c82c

LinLo, 2020-08-13:
Hello,

Compilation now successfully pass with 5.8.0, thanks a lot !
Archlinux have just released Linux kernel 5.8.1 and unfortunately there is another failure:

```
make -C /lib/modules/5.8.1-arch1-1/build M=/home/user/tbs-linux_media-git/src/media_build/v4l  modules
make[1]: Entering directory '/usr/lib/modules/5.8.1-arch1-1/build'
  CC [M]  /home/user/tbs-linux_media-git/src/media_build/v4l/msp3400-driver.o
In file included from <command-line>:
/home/user/tbs-linux_media-git/src/media_build/v4l/compat.h:1594:19: error: redefinition of 'prandom_u32_max'
 1594 | static inline u32 prandom_u32_max(u32 ep_ro)
      |                   ^~~~~~~~~~~~~~~
In file included from ./include/linux/random.h:118,
                 from /home/user/tbs-linux_media-git/src/media_build/v4l/compat.h:1589,
                 from <command-line>:
./include/linux/prandom.h:44:19: note: previous definition of 'prandom_u32_max' was here
   44 | static inline u32 prandom_u32_max(u32 ep_ro)
      |                   ^~~~~~~~~~~~~~~
In file included from <command-line>:
/home/user/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'prandom_u32_max':
/home/user/tbs-linux_media-git/src/media_build/v4l/compat.h:1591:21: error: implicit declaration of function 'random32'; did you mean 'prandom_u32'? [-Werror=implicit-function-declaration]
 1591 | #define prandom_u32 random32
      |                     ^~~~~~~~
/home/user/tbs-linux_media-git/src/media_build/v4l/compat.h:1596:22: note: in expansion of macro 'prandom_u32'
 1596 |  return (u32)(((u64) prandom_u32() * ep_ro) >> 32);
      |                      ^~~~~~~~~~~
In file included from <command-line>:
/home/user/tbs-linux_media-git/src/media_build/v4l/compat.h: At top level:
/home/user/tbs-linux_media-git/src/media_build/v4l/compat.h:2438:19: error: redefinition of 'next_pseudo_random32'
 2438 | static inline u32 next_pseudo_random32(u32 seed)
      |                   ^~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/random.h:118,
                 from /home/user/tbs-linux_media-git/src/media_build/v4l/compat.h:1589,
                 from <command-line>:
./include/linux/prandom.h:73:19: note: previous definition of 'next_pseudo_random32' was here
   73 | static inline u32 next_pseudo_random32(u32 seed)
      |                   ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/user/tbs-linux_media-git/src/media_build/v4l/msp3400-driver.o] Error 1
make[1]: *** [Makefile:1756: /home/user/tbs-linux_media-git/src/media_build/v4l] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.8.1-arch1-1/build'
make: *** [Makefile:53: default] Error 2
```

Kernel version: 5.8.1-arch1-1

Thanks so much for maintaining this project.

caveman99, 2020-08-13:
for a temporary fix create a file 'config-mycompat.h' in media_build/v4l that contains:

```
#undef NEED_PRANDOM_U32_MAX
#undef NEED_PRANDOM_U32
#undef NEED_NEXT_PSEUDO_RANDOM32
```

caveman99, 2020-08-13:
and a heads up - Kernel 5.8 needs at least gcc 9.x, older distros may be out of luck.

LinLo, 2020-08-13:
Ok I will try the temporary fix you provided for 5.8.1.
Archlinux gcc version:
$ gcc --version
gcc (GCC) 10.1.0

LinLo, 2020-08-13:
> 
> 
> for a temporary fix create a file 'config-mycompat.h' in media_build/v4l that contains:
> 
> ```
> #undef NEED_PRANDOM_U32_MAX
> #undef NEED_PRANDOM_U32
> #undef NEED_NEXT_PSEUDO_RANDOM32
> ```

Compilation succeed for 5.8.1 with this temporary fix.
Thanks a lot.

caveman99, 2020-08-13:
This needs to be addressed thoroughly in the compat layer of the build
system itself. the file you just created will be removed with distclean
(!). The mycompat file is meant for working around backporting issued by
distros, e.g. changes from newer kernels ported to older kernels and
collisions occurring from that. In this case 5.8.1 defines some functions
that are also in the compat layer and the kernel is too new for the video
subsystem used here.

kumy, 2020-10-10:
Same here on `Ubuntu 16.04.6 LTS`
```
$ uname -a
Linux nas 4.15.0-118-generic #119~16.04.1-Ubuntu SMP Tue Sep 8 14:54:40 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
The provided workaround seems to let it compile now.
PS: my build tree is still on commit 94caad9