tbsdtv/linux_media #212: tbs6909x unicable

jurek4321, 2020-08-24:
Is it possible to connect tbs6909x to unicable using only one cable? In user guide of tbs6909 is written to connect unicable to port 3 to use unicable, but in user guide of tbs6909x i can't any information about it. Does anybody know how to do it on tbs6909x?

crazycat69, 2020-08-24:
Similar mode option for module stid135 instead mxl58x
Example:
**options stid135 mode=0** - quattro/quad LNB connection, default mode
**options stid135 mode=1** - direct connection for each frontend pairs
**options stid135 mode=1 rfsource=1-4** - direct connection to single input 1-4
**options stid135 mode=2** - direct connection to single input 3 (equivalent mode=1 rfsource=3).

jurek4321, 2020-08-25:
Ok, thanks. So when i connect unicable to input 3 i can record 8 channels from different transponders simultaneously?