tbsdtv/linux_media #213: Compilation failed with 5.4.48 (Gentoo) and gcc 9.3

gnah33, 2020-08-29:
gcc version is 9.3.0-r1.
Any trick to help ?

```
  LD [M]  /home/user/tbsdriver/media_build/v4l/vimc.o
  LD [M]  /home/user/tbsdriver/media_build/v4l/vivid.o
  LD [M]  /home/user/tbsdriver/media_build/v4l/vicodec.o
  LD [M]  /home/user/tbsdriver/media_build/v4l/shark2.o
  LD [M]  /home/user/tbsdriver/media_build/v4l/altera-stapl.o
  Building modules, stage 2.
  MODPOST 719 modules
ERROR: "__devm_regmap_init_sccb" [/home/user/tbsdriver/media_build/v4l/ov9650.ko] undefined!
make[3]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[2]: *** [Makefile:1619: modules] Error 2
make[2]: Leaving directory '/usr/src/linux-5.4.48-gentoo'
make[1]: *** [Makefile:53: default] Error 2
make[1] : on quitte le répertoire « /home/user/tbsdriver/media_build/v4l »
make: *** [Makefile:26: all] Error 2
```  

How not to build the ov9650 module I don't need it anyway ?

Thank you

zhang111chao, 2020-08-31:
please try this steps :
1. #wget http://www.tbsdtv.com/download/document/linux/media_build-2020-07-07.tar.bz2
2. #tar jxvf media_build-2020-07-07.tar.bz2
3. #cd media_build
4 #./install.sh
wait some time...
5 #reboot

gnah33, 2020-09-15:
Thanks a lot, it works now !

closed 2020-09-15 by gnah33