tbsdtv/linux_media #214: 5520SE green lock LED doesn't work

smp79, 2020-09-16:
The green LED works only in Windows. With current Linux driver it doesn't seem to work. Is it a known issue?

Davin622, 2020-09-18:
Hi 
please try this patch

--- a/drivers/media/dvb-frontends/si2183.c
+++ b/drivers/media/dvb-frontends/si2183.c
@@ -1088,6 +1088,13 @@ static int si2183_init(struct dvb_frontend *fe)
                return ret;
        }
 
+       // set pins
+       memcpy(cmd.args, "\x12\x8\x0", 3);
+       cmd.wlen = 3;
+       cmd.rlen = 3;
+       si2183_cmd_execute(client, &cmd);
+
+
        dev->fw_loaded = true;
 warm:
        dev->active_fe |= (1 << fe->id);

smp79, 2020-09-18:
It works, thank you!!

closed 2020-09-18 by smp79

kyvaith, 2022-01-28:
is this path is merged? I have an issue on ubuntu - channels stops working afer few minutes. There is no green LED at all. On Windows works ok