tbsdtv/linux_media #215: 6904 Signal Level Problem

semihkirgel, 2020-09-24:
Hello,
------
root@server:/usr/src/media# uname -a
Linux server 4.4.0-171-generic #200-Ubuntu SMP Tue Dec 3 11:04:55 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
--------
root@server:/usr/src/media# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial
---------
Now Iistalled version

root@server:/usr/src/media# git log
commit bdcf535011834b31568d221375419195715e9811
Author: CrazyCat <crazycat69@narod.ru>
Date:   Mon May 18 03:55:22 2020 +0300

    media: dvb-frontends/stid135: Improved signal statistic performance.
--------

6x TBS6904 installed on server. I'm using Cesbo.
I've tried many different linux media versions but the signal levels are low.
I use a 3 meter dish and pro series multiswitch. 
I'm checking the multiswitch output with the signal meter. For many frequencies, I get 100% Signal Level, over 15 db. I get similar results when testing with the Gigablue Linux Receiver. (see images for a few examples)
[image]
[image]
[image]
[image]
[image]
[image]

But the TBS6904 signal results are very bad.

[image]

Comparison: 

TP         -  GigaBlue -  TBS6904
12209H -  100         -  44
12685H -  87           -  40
12015H -  100         -  51
12103H -  96           -  41
11054V -   98          -  51
12329H -  87           -  39

Other than that, some channels give CC: 1 error. While other channels work normally on the same transponder, CC: 1 error can be seen on only one channel.

Sep 24 16:52:30[NTV HD i/1] Bitrate:5916Kbit/s CC:1
Sep 24 16:52:30[STAR TV HD i/1] Bitrate:4822Kbit/s CC:1
Sep 24 16:52:34[KANAL D HD i/1] Bitrate:4454Kbit/s CC:1
Sep 24 16:52:34[T.A.Y TV i/1] Bitrate:3630Kbit/s CC:1
Sep 24 16:52:34[teve2 HD i/1] Bitrate:4364Kbit/s CC:1
Sep 24 16:52:34[TV35 i/1] Bitrate:1836Kbit/s CC:1
Sep 24 16:52:35[CNN TÜRK HD i/1] Bitrate:3648Kbit/s CC:1
Sep 24 16:52:36[FOX TV HD i/1] Bitrate:12739Kbit/s CC:1
Sep 24 16:52:37[MELTEM TV i/1] Bitrate:1574Kbit/s CC:1
Sep 24 16:52:37[teve2 HD i/1] Bitrate:3453Kbit/s CC:1
Sep 24 16:52:37[TV35 i/1] Bitrate:1833Kbit/s CC:1
Sep 24 16:52:38[CNN TÜRK HD i/1] Bitrate:3439Kbit/s CC:1
Sep 24 16:52:38[KANAL D HD i/1] Bitrate:3559Kbit/s CC:1
Sep 24 16:52:38[T.A.Y TV i/1] Bitrate:3148Kbit/s CC:1
Sep 24 16:52:43[BLOOMBERGHT HD i/1] Bitrate:4219Kbit/s CC:1
Sep 24 16:52:43[HABERTURK HD i/1] Bitrate:7784Kbit/s CC:1
Sep 24 16:52:43[SHOW HD i/1] Bitrate:9309Kbit/s CC:1
Sep 24 16:52:47[teve2 HD i/1] Bitrate:4882Kbit/s CC:1
Sep 24 16:52:47[TV35 i/1] Bitrate:1891Kbit/s CC:1
Sep 24 16:52:48[CNN TÜRK HD i/1] Bitrate:7009Kbit/s CC:1
Sep 24 16:52:48[KANAL D HD i/1] Bitrate:4144Kbit/s CC:1
Sep 24 16:52:48[T.A.Y TV i/1] Bitrate:4324Kbit/s CC:1
Sep 24 16:52:51[teve2 HD i/1] Bitrate:2188Kbit/s CC:1
Sep 24 16:52:51[TV35 i/1] Bitrate:1842Kbit/s CC:1
Sep 24 16:52:52[CNN TÜRK HD i/1] Bitrate:7321Kbit/s CC:1
Sep 24 16:52:52[KANAL D HD i/1] Bitrate:3560Kbit/s CC:1
Sep 24 16:52:52[T.A.Y TV i/1] Bitrate:3369Kbit/s CC:1
Sep 24 16:52:52[NTV HD i/1] Bitrate:3315Kbit/s CC:1
Sep 24 16:52:52[STAR TV HD i/1] Bitrate:3889Kbit/s CC:1
Sep 24 16:52:55[MELTEM TV i/1] Bitrate:1558Kbit/s CC:2
Sep 24 16:52:58[teve2 HD i/1] Bitrate:2827Kbit/s CC:1
Sep 24 16:52:58[TV35 i/1] Bitrate:1436Kbit/s CC:1
Sep 24 16:52:59[CNN TÜRK HD i/1] Bitrate:6694Kbit/s CC:1
Sep 24 16:52:59[KANAL D HD i/1] Bitrate:3412Kbit/s CC:1
Sep 24 16:52:59[T.A.Y TV i/1] Bitrate:4846Kbit/s CC:1
Sep 24 16:53:05[KANAL D HD i/1] Bitrate:4933Kbit/s CC:1
Sep 24 16:53:05[T.A.Y TV i/1] Bitrate:5198Kbit/s CC:1
Sep 24 16:53:05[BLOOMBERGHT HD i/1] Bitrate:4219Kbit/s CC:1
Sep 24 16:53:05[HABERTURK HD i/1] Bitrate:7784Kbit/s CC:1
Sep 24 16:53:05[SHOW HD i/1] Bitrate:9310Kbit/s CC:1
Sep 24 16:53:05[teve2 HD i/1] Bitrate:2704Kbit/s CC:1
Sep 24 16:53:05[TV35 i/1] Bitrate:1723Kbit/s CC:1
Sep 24 16:53:06[CNN TÜRK HD i/1] Bitrate:7604Kbit/s CC:1

Thank you for help.
Kind Regards

Davin622, 2020-09-25:
Hi 
because the chip sample relatively less for signals , so the signal features is not accurately. we will try the new algorihm.
about the lose packets, is it a continuous loss? or just occasionally ?

Best Regards

semihkirgel, 2020-09-25:
Hi,

Continuously with intervals of 3-10 seconds for TP - 12245H. However, it does this on 4 out of 6 channels.
Continuously with intervals of 21-22 seconds for TP - 12209H. CC: 1 error on all channels for this frequency
Continuously with intervals of 21-22 seconds for TP -  12015H. CC: 1 error on all channels for this frequency
Continuously with intervals of 41-42 seconds for TP -  12329H. CC: 1 error on all channels for this frequency

Best Regards

semihkirgel, 2020-09-25:
Hi,

I used Cesbo Analysis now. There is no problem in Video and Audio data. CC: 1 errors come in the AIT table. You can see it in the images. 
[image]