tbsdtv/linux_media #216: TBS 6981, Archlinux 5.8.13 and frequency limitation

rajil, 2020-10-04:
I am using the driver shipped with 5.8.13 and tried to do a blind scan with w_scan2.

The log below throws up a warning "unsupported by driver: min=0.95, max=2.15". Do i need the Closed source drivers to scan this frequency range?

```
# w_scan2 -c IN -fs -s S83E0 -G
w_scan2 -c IN -fs -s S83E0 -G 
w_scan2 version 1.0.9 (compiled for DVB API 5.11)
using settings for 83.0 east Insat 2E/4A
scan type SATELLITE, channellist 37
output format gstreamer
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
/dev/dvb/adapter0/frontend0 -> SATELLITE "Conexant CX24117/CX24132": very good :-))

Using SATELLITE frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.11
frontend 'Conexant CX24117/CX24132' supports
INVERSION_AUTO
DVB-S
DVB-S2
FREQ (0.95GHz ... 2.15GHz)
SRATE (1.000MSym/s ... 45.000MSym/s)
using LNB "UNIVERSAL"
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
trying 'S f = 3521 kHz V SR = 8640 3/4 0,35 QPSK (0:0:0)'
(time: 00:00.093) skipped: (freq 6.23 unsupported by driver: min=0.95, max=2.15)

initial_tune:3327: Setting frontend failed S f = 3521 kHz V SR = 8640 3/4 0,35 QPSK (0:0:0)
trying 'S f = 3537 kHz V SR = 10800 3/4 0,35 QPSK (0:0:0)'
(time: 00:01.233) skipped: (freq 6.21 unsupported by driver: min=0.95, max=2.15)
```

crazycat69, 2020-10-04:
Driver accept intermediate frequency 950-2150 Mhz, you try scan C-band satellite with Ku universal LNB params - so wrong IF passwd to driver. So set rigth LNB params for C band and try again :)

rajil, 2020-10-04:
Thanks for the tip. Is this the right way to do it?

Unfortunately, it doesnt detect any channels.

```
# w_scan2 -c IN -fs -s S83E0 -l C-BAND
w_scan2 -c IN -fs -s S83E0 -l C-BAND 
w_scan2 version 1.0.9 (compiled for DVB API 5.11)
using settings for 83.0 east Insat 2E/4A
scan type SATELLITE, channellist 37
output format vdr-2.0
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> SATELLITE "Conexant CX24117/CX24132": very good :-))

Using SATELLITE frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_ 
Using DVB API 5.11
frontend 'Conexant CX24117/CX24132' supports
INVERSION_AUTO
DVB-S
DVB-S2
FREQ (0.95GHz ... 2.15GHz)
SRATE (1.000MSym/s ... 45.000MSym/s)
using LNB "C-BAND"
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_ 
trying 'S  f = 3521 kHz V SR =  8640  3/4 0,35  QPSK  (0:0:0)'
(time: 00:00.033) 
trying 'S  f = 3537 kHz V SR = 10800  3/4 0,35  QPSK  (0:0:0)'
(time: 00:04.496) 
trying 'S  f = 3725 kHz H SR = 26665  3/4 0,35  QPSK  (0:0:0)'
(time: 00:08.963) 
trying 'S  f = 3756 kHz H SR = 13333  7/8 0,35  QPSK  (0:0:0)'
(time: 00:13.423) 
trying 'S  f = 3767 kHz H SR =  3000  3/4 0,35  QPSK  (0:0:0)'
(time: 00:17.883) 
trying 'S  f = 3777 kHz H SR = 10900  3/4 0,35  QPSK  (0:0:0)'
(time: 00:22.352) 
trying 'S  f = 3805 kHz H SR = 28500  7/8 0,35  QPSK  (0:0:0)'
(time: 00:26.822) 
trying 'S  f = 3841 kHz H SR =  6920  3/4 0,35  QPSK  (0:0:0)'
(time: 00:31.272) 
trying 'S  f = 3847 kHz H SR =  3333  3/4 0,35  QPSK  (0:0:0)'
(time: 00:35.729) 
trying 'S  f = 3860 kHz H SR =  6920  3/4 0,35  QPSK  (0:0:0)'
(time: 00:40.192) 
trying 'S  f = 3868 kHz H SR =  3000  3/4 0,35  QPSK  (0:0:0)'
(time: 00:44.645) 
trying 'S  f = 3874 kHz H SR =  3403  3/4 0,35  QPSK  (0:0:0)'
(time: 00:49.099) 
trying 'S  f = 3884 kHz H SR =  4440  3/4 0,35  QPSK  (0:0:0)'
(time: 00:53.552) 
trying 'S  f = 3892 kHz H SR =  3300  3/4 0,35  QPSK  (0:0:0)'
(time: 00:58.015) 
trying 'S  f = 3898 kHz H SR =  6800  3/4 0,35  QPSK  (0:0:0)'
(time: 01:02.472) 
trying 'S  f = 3909 kHz H SR =  3600  7/8 0,35  QPSK  (0:0:0)'
(time: 01:06.935) 
trying 'S  f = 3919 kHz H SR = 13000  3/4 0,35  QPSK  (0:0:0)'
(time: 01:11.388) 
trying 'S  f = 3936 kHz H SR = 10100  7/8 0,35  QPSK  (0:0:0)'
(time: 01:15.848) 
trying 'S2 f = 3949 kHz H SR =  3673  3/4 0,AUTO  8PSK  (0:0:0)'
(time: 01:20.318) 
trying 'S  f = 3958 kHz H SR =  9500  7/8 0,35  QPSK  (0:0:0)'
(time: 01:27.468) 
trying 'S  f = 3976 kHz H SR =  3200  3/4 0,35  QPSK  (0:0:0)'
(time: 01:31.931) 
trying 'S  f = 4004 kHz H SR = 22220  5/6 0,35  QPSK  (0:0:0)'
[... 58 lines cut ...]
initial_tune:3327: Setting frontend failed S  f = 11470 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)
trying 'S  f = 11510 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:53.137)        skipped: (freq 6.36 unsupported by driver: min=0.95, max=2.15)

initial_tune:3327: Setting frontend failed S  f = 11510 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)
trying 'S  f = 11550 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:53.137)        skipped: (freq 6.40 unsupported by driver: min=0.95, max=2.15)

initial_tune:3327: Setting frontend failed S  f = 11550 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)
trying 'S  f = 11590 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:53.137)        skipped: (freq 6.44 unsupported by driver: min=0.95, max=2.15)

initial_tune:3327: Setting frontend failed S  f = 11590 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)
trying 'S  f = 11630 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:53.137)        skipped: (freq 6.48 unsupported by driver: min=0.95, max=2.15)

initial_tune:3327: Setting frontend failed S  f = 11630 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)
trying 'S  f = 11670 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:53.137)        skipped: (freq 6.52 unsupported by driver: min=0.95, max=2.15)

initial_tune:3327: Setting frontend failed S  f = 11670 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)

ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!
```

rajil, 2020-10-05:
The LNB is of type Inverto Unicable II. Is there anything special which needs to be passed by the scanning utility?

rajil, 2020-10-05:
I changed the LNB to a simple dual port LNB.

I am able to scan with TBSviewer in Windows but both w_scan2/dvbv5-scan are failing.

In Windows, 762 channels on found
[image]

Under Linux 
```

# w_scan2 -c IN -fs -s S83E0 -l ENHANCED                                                                                                                                                                               
w_scan2 -c IN -fs -s S83E0 -l ENHANCED                                                                                                                                                                                                 
w_scan2 version 1.0.9 (compiled for DVB API 5.11)                                                                                                                                                                                      
using settings for 83.0 east Insat 2E/4A                                                                                                                                                                                               
scan type SATELLITE, channellist 37                                                                                                                                                                                                    
output format vdr-2.0                                                                                                                                                                                                                  
WARNING: could not guess your codepage. Falling back to 'UTF-8'                                                                                                                                                                        
output charset 'UTF-8', use -C <charset> to override                                                                                                                                                                                   
Info: using DVB adapter auto detection.                                                                                                                                                                                                

Using SATELLITE frontend (adapter /dev/dvb/adapter0/frontend0)                                                                                                                                                                         
-_-_-_-_ Getting frontend capabilities-_-_-_-_                                                                                                                                                                                         
Using DVB API 5.11                                                                                                                                                                                                                     
frontend 'Conexant CX24117/CX24132' supports                                                                                                                                                                                           
INVERSION_AUTO                                                                                                                                                                                                                         
DVB-S                                                                                                                                                                                                                                  
DVB-S2                                                                                                                                                                                                                                 
FREQ (0.95GHz ... 2.15GHz)                                                                                                                                                                                                             
SRATE (1.000MSym/s ... 45.000MSym/s)                                                                                                                                                                                                   
using LNB "ENHANCED"
initial_tune:3327: Setting frontend failed S  f = 4180 kHz H SR =  3233  3/4 0,35  QPSK  (0:0:0)
trying 'S2 f = 10970 kHz H SR = 28500  3/4 0,AUTO  8PSK  (0:0:0)'
(time: 00:00.040) 
trying 'S  f = 11010 kHz H SR = 27500  3/4 0,35  QPSK  (0:0:0)'
(time: 00:09.340) 
trying 'S  f = 11050 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:14.460) 
trying 'S  f = 11090 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:19.610) 
trying 'S  f = 11130 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:24.700) 
trying 'S  f = 11170 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:29.780) 
trying 'S  f = 11470 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:34.870) 
trying 'S  f = 11510 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:39.990) 
trying 'S  f = 11550 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:45.130) 
trying 'S  f = 11590 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:50.240) 
trying 'S  f = 11630 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 00:55.360) 
trying 'S  f = 11670 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:00.480) 

[... cut ...]
Scanning frequency #47 11510000
RF     (0x01) Signal= 73.43% C/N= 0.00% UCB= 0 postBER= 0
Scanning frequency #48 11550000
RF     (0x01) Signal= 73.28% C/N= 0.00% UCB= 0 postBER= 0
Scanning frequency #49 11590000
RF     (0x01) Signal= 73.28% C/N= 0.00% UCB= 0 postBER= 0
Scanning frequency #50 11630000
RF     (0x01) Signal= 73.13% C/N= 0.00% UCB= 0 postBER= 0
Scanning frequency #51 11670000
RF     (0x01) Signal= 72.97% C/N= 0.00% UCB= 0 postBER= 0
```

Any idea why scan returns result in Windows but not in Linux?

crazycat69, 2020-10-05:
what is **-l ENHANCED** ?

rajil, 2020-10-05:
```
# w_scan2  -l?
w_scan2 -l? 
UNIVERSAL
        Europe
        10800 to 11800 MHz and 11600 to 12700 Mhz
        Dual LO, loband 9750, hiband 10600 MHz
DBS
        Expressvu, North America
        12200 to 12700 MHz
        Single LO, 11250 MHz
STANDARD
        10945 to 11450 Mhz
        Single LO, 10000 Mhz
ENHANCED
        Astra
        10700 to 11700 MHz
        Single LO, 9750 MHz
C-BAND
        Big Dish - Monopoint LNBf
        3700 to 4200 MHz
        Single LO, 5150 Mhz
C-MULTI
        Big Dish - Multipoint LNBf
        3700 to 4200 MHz
        Dual LO, 5150/5750 Mhz
AUSTRALIA
        Australia Single LO LNB
        11700 to 12750 MHz
        Single LO, 10700 MHz
```

crazycat69, 2020-10-05:
try UNIVERSAL

rajil, 2020-10-05:
Where do i get the transponder list for 83E?

```
# w_scan2 -c IN -fs -s S83E0 -l UNIVERSAL
w_scan2 -c IN -fs -s S83E0 -l UNIVERSAL 
w_scan2 version 1.0.9 (compiled for DVB API 5.11)
using settings for 83.0 east Insat 2E/4A
scan type SATELLITE, channellist 37
output format vdr-2.0
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> SATELLITE "Conexant CX24117/CX24132": very good :-))

Using SATELLITE frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_ 
Using DVB API 5.11
frontend 'Conexant CX24117/CX24132' supports
INVERSION_AUTO
DVB-S
DVB-S2
FREQ (0.95GHz ... 2.15GHz)
SRATE (1.000MSym/s ... 45.000MSym/s)
using LNB "UNIVERSAL"
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_ 
trying 'S2 f = 10970 kHz H SR = 28500  3/4 0,AUTO  8PSK  (0:0:0)'
(time: 00:50.909) 
trying 'S  f = 11010 kHz H SR = 27500  3/4 0,35  QPSK  (0:0:0)'
(time: 01:01.948) 
trying 'S  f = 11050 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:08.548) 
trying 'S  f = 11090 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:15.248) 
trying 'S  f = 11130 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:21.848) 
trying 'S  f = 11170 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:28.518) 
trying 'S  f = 11470 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:35.198) 
trying 'S  f = 11510 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:41.728) 
trying 'S  f = 11550 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:48.328) 
trying 'S  f = 11590 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 01:54.887) 
trying 'S  f = 11630 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:01.427) 
trying 'S  f = 11670 kHz H SR = 30000  7/8 0,35  QPSK  (0:0:0)'
(time: 02:08.047) 

ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!
```

crazycat69, 2020-10-05:
https://www.flysat.com/gsat10-30.php

rajil, 2020-10-05:
I compared my /usr/share/dvb/dvb-s/Insat-4A-83.0E with https://www.lyngsat.com/Insat-4A-and-G-Sat-10.html, and they mostly match. This is the file i used to scan with dvbv5-scan [above](https://github.com/tbsdtv/linux_media/issues/216#issuecomment-703641902).

By the way these are the contents of /usr/share/dvb/dvb-s/Insat-4A-83.0E which have PSK/8

[code]
# Channel table for Insat 4A at 83.0°E
# Source: http://www.lyngsat.com/Insat-4A.html

[Indiasign]
	DELIVERY_SYSTEM = DVBS
	FREQUENCY = 3756000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 13333000
	INNER_FEC = AUTO
	INVERSION = AUTO

[SVBC]
	DELIVERY_SYSTEM = DVBS2
	FREQUENCY = 3767000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 3000000
	MODULATION = QPSK
	INNER_FEC = AUTO
	INVERSION = AUTO

[channel 3776000]
	DELIVERY_SYSTEM = DVBS2
	FREQUENCY = 3776000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 11250000
	MODULATION = PSK/8
	INNER_FEC = AUTO
	INVERSION = AUTO

[Planetcast]
	DELIVERY_SYSTEM = DVBS2
	FREQUENCY = 3805000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 28500000
	MODULATION = PSK/8
	INNER_FEC = AUTO
	INVERSION = AUTO

[channel 3840000]
	DELIVERY_SYSTEM = DVBS
	FREQUENCY = 3840000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 6920000
	INNER_FEC = AUTO
	INVERSION = AUTO

[Kasthuri TV]
	DELIVERY_SYSTEM = DVBS
	FREQUENCY = 3858000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 6920000
	INNER_FEC = AUTO
	INVERSION = AUTO

[Mega TV Network]
	DELIVERY_SYSTEM = DVBS2
	FREQUENCY = 3870000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 5000000
	MODULATION = PSK/8
	INNER_FEC = AUTO
	INVERSION = AUTO

[channel 3880000]
	DELIVERY_SYSTEM = DVBS
[... 351 lines cut ...]
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 32720000
	MODULATION = PSK/8
	INNER_FEC = AUTO
	INVERSION = AUTO

[Tata Sky]
	DELIVERY_SYSTEM = DVBS2
	FREQUENCY = 11630000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 32720000
	MODULATION = PSK/8
	INNER_FEC = AUTO
	INVERSION = AUTO

[Tata Sky]
	DELIVERY_SYSTEM = DVBS2
	FREQUENCY = 11670000
	POLARIZATION = HORIZONTAL
	SYMBOL_RATE = 32720000
	MODULATION = PSK/8
	INNER_FEC = AUTO
	INVERSION = AUTO

[/code]

rajil, 2020-10-06:
I was able to scan with w_scan2 after modifying the satellites.dat file as mentioned [here](https://github.com/stefantalpalaru/w_scan2/issues/27).

Once w_scan2 was able to lock the tuner, dvbv5-scan also started working.

```
dvbv5-scan /usr/share/dvb/dvb-s/Insat-4A-83.0E -l UNIVERSAL
Using LNBf UNIVERSAL
        Old European Universal. Nowadays mostly replaced by Astra 19.2E
        Freqs     : 10800 to 11800 MHz, LO: 9750 MHz
        Freqs     : 11600 to 12700 MHz, LO: 10600 MHz
Scanning frequency #41 11010000
Lock   (0x1f) Signal= 74.35% C/N= 0.19% UCB= 124 postBER= 0
```

closed 2020-10-06 by rajil