tbsdtv/linux_media #218: SEC_VOLTAGE_OFF is not handled in TBS 6909

malakudi, 2020-10-21:
TBS609 mxl58x driver does not handle SEC_VOLTAGE_OFF correctly. When tuning stops and adapter is released, votlage stays at 13V and doesn't drop to 0V. Also if you try to set SEC_VOLTAGE_OFF, it doesn't work. Issue is inside mxl58x.c in funtion set_voltage
```
if (voltage == SEC_VOLTAGE_18)
      state->rf_in &= ~2;
    else
      state->rf_in |= 2;
  }
```
It only checks if voltage = 18V, else does what it has to be done for 13V. 0V is never handled.

This issue affects re-tuning in some Unicable 2 switches like Technirouter 9/4x16 by Technisat and some Jultec models as well.

crazycat69, 2020-10-22:
This fragment related to mode 0 - select rf input depend from polarizaton. For direct connection mode possible set  voltage off.

malakudi, 2020-10-23:
I think I have also tested with mode=1 and mode=2 and voltage never goes to 0V. Tested with multimeter on the R/F output. If you are sure it works now I will retest today and report back.

Davin622, 2020-12-10:
Hi malakudi,
  In  mode 1 , because two adapter use the same  input,  if one adapter shut down the LNB power, it will make the another adapter unlocked. so we do not handle it.
the same status with mode 2. 

Best Regards

deeptho, 2020-12-30:
> Hi malakudi,
In mode 1 , because two adapter use the same input, if one adapter shut down the LNB power, it will make the another adapter unlocked. so we do not handle it.
the same status with mode 2.

Related to these problems (but not 100% the same):
See issue #173 and the associated patch.