tbsdtv/linux_media #221: TBS6281SE: endless firmware messages in dmesg

jmwislez, 2020-12-11:
I am running a Ubuntu 18.04 system with 3 TBS boards (TBS6910, TBS6281SE and TBD6903) and latest open source TBS drivers.  The boards work, but every 5 seconds or so the following entries appear in dmesg:

[ 1144.474332] si2168 12-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 1146.453395] si2168 12-0064: firmware version: B 4.0.25
[ 1146.474249] si2157 15-0060: found a 'Silicon Labs Si2157-A30'
[ 1146.502643] si2157 15-0060: firmware version: 3.0.5

This significantly clutters the dmesg, making it unusable for investigating issues.

Davin622, 2021-03-17:
Hi
You can disable the debug info.
the file on drivers/media/dvb-frontends/si2168.c
line 583: 	//dev_info(&client->dev, "downloading firmware from file '%s'\n",
		//	dev->firmware_name);
line 633: //	dev_info(&client->dev, "firmware version: %c %d.%d.%d\n",
		// dev->version >> 24 & 0xff, dev->version >> 16 & 0xff,
		// dev->version >> 8 & 0xff, dev->version >> 0 & 0xff);
another file on drivers/media/tuners/si2157.c
line 178 :  //	dev_info(&client->dev, "found a 'Silicon Labs Si21%d-%c%c%c'\n",
		//	cmd.args[2], cmd.args[1], cmd.args[3], cmd.args[4]);

line 200:	//dev_info(&client->dev, "downloading firmware from file '%s'\n",
		//	fw_name);

line 240: //	dev_info(&client->dev, "firmware version: %c.%c.%d\n",
		//	cmd.args[6], cmd.args[7], cmd.args[8]);

save them ,then  make &&make install
the debug info will be gone.

jmwislez, 2021-05-19:
Thanks, but what is the use of such debug info by default?  I would suggest to suppress such messages by default, to enhance the usefulness of the logging.