tbsdtv/linux_media #223: Compilation fails on Kernel v5.10.1

ich777, 2020-12-19:
Tried to compile the drivers on Kernel v5.10.1 and get this error:

```
/usr/src/TBS_OS/media_build/v4l/tbs5301.c: In function 'reg_i2cread':
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:186:10: error: 'return' with a value, in function returning void [-Werror=return-type]
  186 |   return -ENODEV;
      |          ^
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:180:13: note: declared here
  180 | static void reg_i2cread(struct i2c_adapter *i2c,u8 chip_addr,u8 reg, u8 num, u8 *buf)
      |             ^~~~~~~~~~~
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:188:10: error: 'return' with a value, in function returning void [-Werror=return-type]
  188 |   return -EAGAIN;
      |          ^
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:180:13: note: declared here
  180 | static void reg_i2cread(struct i2c_adapter *i2c,u8 chip_addr,u8 reg, u8 num, u8 *buf)
      |             ^~~~~~~~~~~
/usr/src/TBS_OS/media_build/v4l/tbs5301.c: In function 'reg_i2cwrite':
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:213:10: error: 'return' with a value, in function returning void [-Werror=return-type]
  213 |   return -ENODEV;
      |          ^
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:206:13: note: declared here
  206 | static void reg_i2cwrite(struct i2c_adapter *i2c,u8 chip_addr,u8 reg, u8 num, u8 *buf)
      |             ^~~~~~~~~~~~
/usr/src/TBS_OS/media_build/v4l/tbs5301.c:215:10: error: 'return' with a value, in function returning void [-Werror=return-type]
  215 |   return -EAGAIN;
```

crazycat69, 2020-12-19:
7bed1616b8dad27848010c672b0633366a8ab3a2

ich777, 2020-12-19:
@crazycat69 Thank you for the quick fix!

closed 2020-12-19 by ich777