tbsdtv/linux_media #224: How to add new MyGica T230A?

kuncy7, 2020-12-22:
Hi,

I bought the MyGica T230A tuner. I didn't know there would be a problem with the letter A.

When I connected it to rpi with libreelec, dmesg showed something like this:

```
[ 262.006051] usb 1-1.1: new high-speed USB device number 4 using xhci_hcd

[ 262.097416] usb 1-1.1: New USB device found, idVendor=0572, idProduct=689a, bcdDevice= 8.00

[ 262.097433] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 262.097448] usb 1-1.1: Product: USB Stick

[ 262.097462] usb 1-1.1: Manufacturer: Gen

[ 262.097475] usb 1-1.1: SerialNumber: 2017.12.09
```

This tuner has entries in the "inf" file of the windows driver identical to the T230C v2 :

```
%TC68A_DVB.DeviceDesc% = TC68A_DVB, USB\VID_0572&PID_C68A

%TC68A_DVB.DeviceDesc% = TC68A_DVB, USB\VID_0572&PID_689A
```

I patched the files: dvb-usb-ids.h, dvbsky.c and built a new image libreelec.

After running, dmesg shows now:

```
[ 262.006051] usb 1-1.1: new high-speed USB device number 4 using xhci_hcd

[ 262.097416] usb 1-1.1: New USB device found, idVendor=0572, idProduct=689a, bcdDevice= 8.00

[ 262.097433] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 262.097448] usb 1-1.1: Product: USB Stick

[ 262.097462] usb 1-1.1: Manufacturer: Gen

[ 262.097475] usb 1-1.1: SerialNumber: 2017.12.09

[ 262.402693] usb 1-1.1: dvb_usb_v2: found a 'MyGica Mini DVB-(T/T2/C) USB Stick T230A' in warm state

[ 262.402961] usb 1-1.1: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer

[ 262.403000] dvbdev: DVB: registering new adapter (MyGica Mini DVB-(T/T2/C) USB Stick T230A)

[ 262.403012] usb 1-1.1: media controller created

[ 262.404228] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.

[ 262.479360] si2168 13-0064: probe failed = -121

[ 262.479383] si2168: probe of 13-0064 failed with error -121

[ 262.481522] usbcore: registered new interface driver dvb_usb_dvbsky
```

What could that mean?

Is there a problem with the firmware?

Can it be fixed somehow?

Tested on the devel version:

```
LibreELEC (community): devel-20201214210405-8a6ca28 (RPi4.arm)

LibreELEC:~ # uname -a

Linux LibreELEC 5.9.10 #1 SMP Mon Dec 14 21:20:14 UTC 2020 armv7l GNU/Linux

LibreELEC:~ #
```

erbasandrade, 2021-01-04:
@crazycat
We have similar problems, awaiting some definitive solution.
Thank you if you can help!

crazycat69, 2021-01-04:
34770c639f8c961f8d548d739b6d13205ea86a4c

kuncy7, 2021-01-04:
I don't understand, my tuner TC230A is like this:
```
[ 262.097416] usb 1-1.1: New USB device found, idVendor=0572, idProduct=689a, bcdDevice= 8.00
```
MyGica Mini DVB-T2 USB Stick T230C v2 has idVendor=0572, idProduct=0xc68a .
That's for sure, it's the correct entry?
```
#define USB_PID_MYGICA_T230C2_LITE			0xc69a
```

crazycat69, 2021-01-04:
I not have 0xc68a, 0xc69a sticks - so don't know how this marked :)

erbasandrade, 2021-01-04:
> Eu não tenho 0xc68a, 0xc69a varas - então não sei como isso marcou :) Principal diffrence do clássico T230C (0xc689, 0xc699) algum outro controlador de clone Cypress FX2 com clock de barramento TS diferente.

erbasandrade, 2021-01-04:
@crazycat69 
Does this help in any way?
Or are we unsupported?
https://www.mediafire.com/file/0i23ff7y42gx2hu/T230A_software%2526driver%2526maunal.zip/file

crazycat69, 2021-01-04:
Hmm, look like 689a is some new. This product from AliExpress or another source?

erbasandrade, 2021-01-04:
> Hmm, look like 689a is some new. This product from AliExpress or another source?

Yes friend happy new year! I hope your family is well.
New product said to be the replacement for classic T230C.
It was purchased on aliexpress in the official store.
https://en.aliexpress.com/item/1883597971.html?spm=a2g0o.cart.0.0.8b543c000HSh7U&mp=1
Its website is marked as a legitimate product.
https://www.mygica.com/support/

kuncy7, 2021-01-04:
I wanted to test and I have the same problem as in #200
```
  Building modules, stage 2.
  MODPOST 669 modules
ERROR: "saa7146_pgtable_free" [/home/pi/media_build/v4l/saa7146_vv.ko] undefined!
ERROR: "saa7146_pgtable_alloc" [/home/pi/media_build/v4l/saa7146_vv.ko] undefined!
ERROR: "saa7146_pgtable_build_single" [/home/pi/media_build/v4l/saa7146_vv.ko] undefined!
ERROR: "__aeabi_ldivmod" [/home/pi/media_build/v4l/mn88436.ko] undefined!
ERROR: "__aeabi_uldivmod" [/home/pi/media_build/v4l/mn88436.ko] undefined!
ERROR: "__devm_regmap_init_sccb" [/home/pi/media_build/v4l/ov9650.ko] undefined!
make[3]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[2]: *** [Makefile:1645: modules] Error 2
make[2]: Leaving directory '/home/pi/linux-9797f1a4938c20139b00a25de93cc99efb5c291b'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
make: *** [Makefile:26: all] Error 2
pi@raspberrypi:~/media_build $ 
```

crazycat69, 2021-01-06:
This POWER UP & RESET command + clock freq selection (4-6 bits in cmd[6], 24MHz in this case, but possible 16 and 27 MHz )

kuncy7, 2021-01-06:
It's a fragment si2186.c :

```
<------>/* Initialize */
<------>cmd_init(&cmd, "\xc0\x12\x00\x0c\x00\x0d\x16\x00\x00\x00\x00\x00\x00",
<------><------> 13, 0);
<------>ret = si2168_cmd_execute(client, &cmd);
<------>if (ret)
<------><------>goto err_kfree;

<------>/* Power up */
<------>cmd_init(&cmd, "\xc0\x06\x01\x0f\x00\x20\x20\x01", 8, 1);
<------>ret = si2168_cmd_execute(client, &cmd);
<------>if (ret)
<------><------>goto err_kfree;

<------>/* Query chip revision */
<------>cmd_init(&cmd, "\x02", 1, 13);
<------>ret = si2168_cmd_execute(client, &cmd);
<------>if (ret)
<------><------>goto err_kfree;
```
"Initialize" is OK, bat "Power up"  it generates an error -110, or -121
Can I do anything to get additional information?

crazycat69, 2021-01-07:
need sniff usb traffic from windows drivers

pego149, 2021-01-14:
But in my case with this stick t230a (0572:689a) these 3 procedures pass with 0 ret code. My stick stops in switch where is choosing chip_id and drops into default. It shows me:
`si2168 11-0064: unknown chip version Si2159-\x88\x90`
or next time when I connect:
`si2168 11-0064: unknown chip version Si2159-\x88\xd0X`
 I tried to sniff USB traffic with usbpcap but I don't know how to do it.

fabio70mi, 2021-01-14:
> I tried to sniff USB traffic with usbpcap but I don't know how to do it.
I bought also T230A and investigating how to sniff device initialization with wireshark+USBPcap on Windows

pego149, 2021-01-14:
I found good software which can show URB transfer requests: USBlyzer. In attachment is my dump from t230a.
[t230a.zip](https://github.com/tbsdtv/linux_media/files/5815337/t230a.zip)

fabio70mi, 2021-01-14:
I'm trying usblyzer and sending my dump, but I don't know yet how to inspect it.

[t230a-fabio.zip](https://github.com/tbsdtv/linux_media/files/5815793/t230a-fabio.zip)

pego149, 2021-01-14:
I think that it's same board... So I tried to force these use firmwares and stick now looks fine buut it not find any channels. I don't know much about programing drivers so maybe is somewhere error with reading of output from stick. Here is my output: 
```
[  503.036214] usb 1-1.4: new high-speed USB device number 5 using xhci_hcd
[  503.167558] usb 1-1.4: New USB device found, idVendor=0572, idProduct=689a, bcdDevice= 8.00
[  503.167578] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  503.167595] usb 1-1.4: Product: USB Stick
[  503.167611] usb 1-1.4: Manufacturer: Gen
[  503.167626] usb 1-1.4: SerialNumber: 2017.12.09
[  503.396322] usb 1-1.4: dvb_usb_v2: found a 'MyGica Mini DVB-T2 USB Stick T230A' in warm state
[  503.396693] usb 1-1.4: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[  503.396733] dvbdev: DVB: registering new adapter (MyGica Mini DVB-T2 USB Stick T230A)
[  503.396750] usb 1-1.4: media controller created
[  503.399516] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[  503.401413] usb 1-1.4: dvbsky_mygica_t230c_attach: tsclkman
[  503.421898] si2168 11-0064: unknown chip version Si2159-\x88\x90
[  503.422285] i2c i2c-11: Added multiplexed i2c bus 12
[  503.422304] si2168 11-0064: Silicon Labs Si2168-\xff255232 successfully identified
[  503.422320] si2168 11-0064: firmware version: \xff 255.232.33
[  503.435936] si2157 12-0060: Silicon Labs Si2141 successfully attached
[  503.436032] usb 1-1.4: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[  503.436062] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
[  503.437882] usb 1-1.4: dvb_usb_v2: 'MyGica Mini DVB-T2 USB Stick T230A' successfully initialized and connected
```

kuncy7, 2021-01-14:
These files are from windows where everything works fine and supports my theory it's a clone t230cv2. In my opinion, there is a problem with this piece of driver:
```
/* execute firmware command */
static int si2168_cmd_execute(struct i2c_client *client, struct si2168_cmd *cmd)
{
	struct si2168_dev *dev = i2c_get_clientdata(client);
	int ret;
	unsigned long timeout;

	mutex_lock(&dev->i2c_mutex);

	if (cmd->wlen) {
		/* write cmd and args for firmware */
		ret = i2c_master_send(client, cmd->args, cmd->wlen);
		if (ret < 0) {
			goto err_mutex_unlock;
		} else if (ret != cmd->wlen) {
			ret = -EREMOTEIO;
			goto err_mutex_unlock;
		}
	}

	if (cmd->rlen) {
		/* wait cmd execution terminate */
		#define TIMEOUT 70
		timeout = jiffies + msecs_to_jiffies(TIMEOUT);
		while (!time_after(jiffies, timeout)) {
			ret = i2c_master_recv(client, cmd->args, cmd->rlen);
			if (ret < 0) {
				goto err_mutex_unlock;
			} else if (ret != cmd->rlen) {
				ret = -EREMOTEIO;
				goto err_mutex_unlock;
			}

			/* firmware ready? */
			if ((cmd->args[0] >> 7) & 0x01)
				break;
		}

		dev_dbg(&client->dev, "cmd execution took %d ms\n",
				jiffies_to_msecs(jiffies) -
				(jiffies_to_msecs(timeout) - TIMEOUT));

		/* error bit set? */
		if ((cmd->args[0] >> 6) & 0x01) {
			ret = -EREMOTEIO;
			goto err_mutex_unlock;
		}

		if (!((cmd->args[0] >> 7) & 0x01)) {
			ret = -ETIMEDOUT;
			goto err_mutex_unlock;
		}
	}

	mutex_unlock(&dev->i2c_mutex);
	return 0;
err_mutex_unlock:
	mutex_unlock(&dev->i2c_mutex);
	dev_dbg(&client->dev, "failed=%d\n", ret);
	return ret;
}
```
This definition is interesting:   #define TIMEOUT 70

Why exactly 70? Not 65 or 75 ?

pego149, 2021-01-16:
Finally MyGica T230A working on linux (Raspberry Pi 4). I just compared usb dumps from Windows and Linux as I post before and changed GPIO_WRITE commands in dvbsky.c. So here is edited function in dvbsky.c which makes gpio write:
```
static int dvbsky_identify_state(struct dvb_usb_device *d, const char **name)
{
	dvbsky_gpio_ctrl(d, 0x87, 0);
	msleep(20);
	dvbsky_gpio_ctrl(d, 0x86, 1);
	dvbsky_gpio_ctrl(d, 0x80, 0);
	msleep(100);
	dvbsky_gpio_ctrl(d, 0x80, 1);
	msleep(50);

	return WARM;
}
```
Also I made like here:  34770c639f8c961f8d548d739b6d13205ea86a4c. And here is log in dmesg:
```
[  154.573489] usb 1-1.3: new high-speed USB device number 9 using xhci_hcd
[  154.704617] usb 1-1.3: New USB device found, idVendor=0572, idProduct=689a, bcdDevice= 8.00
[  154.704633] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  154.704646] usb 1-1.3: Product: USB Stick
[  154.704658] usb 1-1.3: Manufacturer: Gen
[  154.704670] usb 1-1.3: SerialNumber: 2017.12.09
[  154.933495] usb 1-1.3: dvb_usb_v2: found a 'MyGica Mini DVB-(T/T2/C) USB Stick T230A' in warm state
[  154.933917] usb 1-1.3: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[  154.933962] dvbdev: DVB: registering new adapter (MyGica Mini DVB-(T/T2/C) USB Stick T230A)
[  154.933983] usb 1-1.3: media controller created
[  154.936756] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[  154.956630] i2c i2c-11: Added multiplexed i2c bus 12
[  154.956651] si2168 11-0064: Silicon Labs Si2168-D60 successfully identified
[  154.956668] si2168 11-0064: firmware version: D 6.0.1
[  154.970484] si2157 12-0060: Silicon Labs Si2141 successfully attached
[  154.970628] usb 1-1.3: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[  154.970660] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
[  154.972280] usb 1-1.3: dvb_usb_v2: 'MyGica Mini DVB-(T/T2/C) USB Stick T230A' successfully initialized and connected
```
Also I tried tuner in TvHeadend and it working well, but I am not expert in drivers. So please developers test and if works add this patch to drivers and also give here feedback if it works or not for you people.

kuncy7, 2021-01-16:
Congratulations, good job! I confirm, the tuner is working. More thorough tests as I compile libreelec.

pego149, 2021-01-16:
But I have now another problem. I compared this tuner with realtek 2832u (aliexpress cheap RTL-SDR) and from MyGica T230A is signal not so good as realtek. It can't get one multiplex which realtek can, but Realtek can't receive DVB-T2. Here is comparison carrier to noise of one multiplex via dvbv5-scan.
From T230A
```
Scanning frequency #2 474000000
Lock   (0x1f) Signal= -21,00dBm C/N= 20,75dB UCB= 0 postBER= 1,00
```
From RTL2832U:
```
Scanning frequency #2 474000000
Lock   (0x1f) Signal= 92,55% C/N= 32,18dB postBER= 65,5x10^-3
```
I don't know if LNA or AGC is supported or how to enable it.

pinkflozd, 2021-01-17:
Could you guys post a full patch?

pego149, 2021-01-17:
Okay so I made pull request.

erbasandrade, 2021-01-18:
> Parabéns, bom trabalho! Confirmo, o sintonizador está funcionando. Testes mais completos durante a compilação do libreelec.

> Congratulations, good job! I confirm, the tuner is working. More thorough tests as I compile libreelec.

Have you tested the same with libreelec? Could I make the used patch available?
Here we are in trouble! it would be of great help if I managed to solve it.
https://forum.libreelec.tv/thread/23142-mygica-t230a/?postID=149219#post149219

erbasandrade, 2021-01-18:
I can confirm that it is working with LibreELEC.
Thanks to @smp Patch: https://www.dropbox.com/s/j1y51whewq0ld71/linux-mygica-t230a.patch?dl=1

erbasandrade, 2021-01-18:
> ESTÁ BEM. Confirmo que está funcionando, obtive boa qualidade / intensidade do sinal no DVB-C, igual ao T230C.
> Meu sistema: Arch Linux + 5.4.89 LTS + Tvheadend
> Aqui está o patch que estou usando:
> https://github.com/pinkflozd/linux_media/commit/8ec77749b81a70c8616e39be4a0e2e0a5a9eb165.patch

[image]

erbasandrade, 2021-01-18:
> OK. I confirm it's working, I Got good quality/signal strength on DVB-C, Same as T230C.
> My system: Arch Linux + 5.4.89 LTS + Tvheadend
> Here's the patch I'm using:
> https://github.com/pinkflozd/linux_media/commit/8ec77749b81a70c8616e39be4a0e2e0a5a9eb165.patch

[image]
I tried your patch, I have an error in the construction of LibreELEC.

pego149, 2021-01-18:
So it's can be problem like mine.. can you test if in Windows you have better results? Because I have in Linux same results like in Windows. But on both OSes reception is very bad and I can't receive multiplex which I can get with Realtek stick. Tommorow I try to debug this stick again and try to enable LNA or AGC. Or have someone any suggestions what can be issue?

kuncy7, 2021-01-18:
First, a new firmware is needed.
```
[   13.312097] si2168 11-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[   13.639898] si2168 11-0064: firmware version: D 6.0.13
[   13.644624] si2157 12-0060: found a 'Silicon Labs Si2141-A10'
[   13.645827] si2157 12-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[   13.810937] si2157 12-0060: firmware version: 1.1.11
```
Secondly, the @pego149 is right, something is wrong with the amplifier LNA or AGC.

On Windows 10, one mux works on this stick from the supplier.

pinkflozd, 2021-01-19:
yes, something is funky also on DVB-C some muxes work perfectly, some need 1 retune, others can't get a lock

pego149, 2021-01-22:
I tried to dump firmware which loads in Windows and its same file as dvb-tuner-si2141-a10-01.fw​ but in case Si2168 there loads only small part of file but when I tried to load this file it's shows me error when downloading firmware EINVAL 22 /* Invalid argument */ ​or EREMOTEIO 121 /* Remote I/O error */​​​ so is any way how to get proper firmware? I tried to add another hex commands to set chip flags like in Windows but result are same and nothing changed.

inogueira82, 2021-02-15:
Hi!

Do you think that this T230a will work in a Synology DS218+ NAS?

i've put fw files in /lib/firmware/ and gave permissions, Reboot, but dmesg only shows this when I connect it to usb port... 

usb 1-2: new high-speed USB device number 6 using xhci_hcd

I'm totally noob with this (drivers).

Do you know what must I do?

closed 2021-02-15 by crazycat69

antonioag95, 2021-05-01:
I have a very strange issue. Dongle is correctly recognized, firmwares are correctly loaded, when it scans it crashes my system when starting to scan a frequency where services should be found. I need to force reset my system to make it work again.

antonioag95, 2021-05-01:
I confirm that this bug is only reproducible on Raspberry Pi Zero W (armv6), on Raspberry Pi 3 it works flawlessly. Any clue? On Pi Zero it just hangs, no kernel panic (so no logs), the cursor in the screen stops blinking and it closes every SSH connection.

AapoTahkola, 2023-07-13:
I still have issues with T230A running 6.4.3 kernel. I have to connect the stick to virtualbox windows after that I get picture with kaffeine without this I only get EPG texts. I did took a capture with wireshark on linux side and replay with usbrply but strangely enough this does not work. Have not examined what could be the issue with replay. Anyway this is not a very big issue for me because the state of T230A survives suspending pc connected to it so I hardly ever have to do the trip to connect virtualbox to get it working. I have only drivers for windows no other software.