tbsdtv/linux_media #225: TBSQBOX2-CI (5980): Not sending diseqc motor commands.

bluzee, 2020-12-29:
I cannot get the TBS5980 to send any motor control commands.  It will operate a diseqc 4 port switch OK.  Software TVHeadend,  tune-s2.  I have another stv090x PCI device that uses CX88 and it works perfectly for GOTOX or bump motor east/west, TVH debug shows  correct diseqc commands being sent by software.

bluzee, 2020-12-31:
I tracked it down to a splitter in the line.  The diseqc codes from the PCI card will pass through the splitter,  only E0 10 XX XX codes from the TBS device will pass through the splitter.  E0 31 XX XX codes do not get through.  Unfortunately the splitter in the line is necessary since the loop out connector on the V-Box broke off the first time I tried to connect the unit.  Only way to get signal to both V-Box and connect to LNBs now is with splitter.  Does anyone know of a splitter type or brand that will work with TBS device to pass all diseqc codes?

crazycat69, 2020-12-31:
Splitter is just diode-blocking circuit. So 18V and 22Khz tone always priority and block slave inputs.

bluzee, 2021-01-01:
Thanks crazycat.  Best wishes to you to.   I guess I'll just have to experiment with a few things. I don't understand why diseqc from one device will pass through but not for another.  Only one tuner is ever connected at a time.  Maybe have to get brave take V-Box apart and try to fix.

closed 2021-01-01 by bluzee