tbsdtv/linux_media #228: TBS6205 with 5.10.4 Identified, but nothing works. TBS6285 is perfectly fine in the same system.

ahmhoseini, 2021-01-11:
I've compiled latest drivers from here with kernel 5.10.4 in ubuntu 20.04.
I have a TBS6285 and TBS6205.
When I use TBS6285 in my system everything is OK.
...
[    8.707195] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    8.723574] i2c i2c-8: Added multiplexed i2c bus 9
[    8.723577] si2168 8-0064: Silicon Labs Si2168-A20 successfully identified
[    8.723578] si2168 8-0064: firmware version: A 2.0.3
[    8.744120] si2157 9-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    8.744662] SAA716x Budget 0000:01:00.0: TurboSight TBS 6285  MAC[0]=ff:ff:ff:ff:ff:ff
[    8.744665] SAA716x Budget 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
....
[  228.498772] si2168 8-0064: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
[  229.938109] si2168 8-0064: firmware version: \x81 2.0.35
[  229.942927] si2157 9-0062: found a 'Silicon Labs Si2158-A20'
[  229.956962] si2157 9-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[  230.223072] si2157 9-0062: firmware version: 2.1.9

And I can use dvbv5-scan, w_scan2 and dvblast without anyproblem.

But when I install TBS6205 in the same system:
...
[    8.689495] TBSECP3 driver 0000:01:00.0: TurboSight TBS 6205 DVB-T/T2/C
[    8.690474] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    8.816897] TBSECP3 driver 0000:01:00.0: MAC address 00:22:ab:b1:aa:ac
[    8.833147] i2c i2c-7: Added multiplexed i2c bus 11
[    8.833151] si2168 7-0064: Silicon Labs Si2168-B40 successfully identified
[    8.833152] si2168 7-0064: firmware version: B 4.0.2
[    8.846085] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    8.846118] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[    8.846260] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    8.966722] TBSECP3 driver 0000:01:00.0: MAC address 00:22:ab:b1:aa:ad
[    8.974652] i2c i2c-8: Added multiplexed i2c bus 12
[    8.974655] si2168 8-0064: Silicon Labs Si2168-B40 successfully identified
[    8.974656] si2168 8-0064: firmware version: B 4.0.2
[    8.981322] si2157 12-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
...
[  560.932420] si2168 7-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  562.870014] si2168 7-0064: firmware version: B 4.0.25
[  562.890103] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[  562.917799] si2157 11-0060: firmware version: 3.0.5

But nothing works. dvbv5-scan does not lock on DVBT2 frequencies, and in DVBT frequencies it times out and returns nothing. same is for w_scan2 and dvblast.

Any suggestions?

crazycat69, 2021-01-11:
Maybe wrong TS  bus config.
Try add 
**si2168_config.ts_clock_inv = 1;** in tbsecp3-dvb.c @ https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/pci/tbsecp3/tbsecp3-dvb.c#L1422

ahmhoseini, 2021-01-13:
Added si2168_config.ts_clock_inv = 1;  that to media/drivers/media/pci/tbsecp3/tbsecp3-dvb.c,
(si2168_config.ts_clock_inv=0;//zc2016/07/20 for TBSECP3_BOARD_TBS6290SE in the same file, any relations?)
(There is also a tbsecp3-dvb.c in media_build/linux/drivers/media/pci/tbsecp3 but I didn't modify it)

Recompiled, Deleted old modules and installed new ones.
No Success!
P.S:
How could I get sure that this new ts_clock_inv = 1 is applied in module compile? Anywhere to check?

sudo dvbv5-scan -a6 ir-Tehran
Scanning frequency #1 738000000
       (0x00) Signal= -83.00dBm
Scanning frequency #2 554000000
       (0x00) Signal= -88.00dBm
Scanning frequency #3 602000000
Lock   (0x1f) Signal= -87.00dBm C/N= 18.75dB UCB= 57433 postBER= 552x10^-3
	  Layer A: C/N= 37.54%
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #4 650000000
       (0x00) Signal= -85.00dBm

w_scan2 -a6
650000: (time: 01:44.603)         signal ok:	QAM_AUTO f = 650000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 650000 kHz I999B8C999D999T999G999Y999 (0:0:0))

w_scan2 -a6 -F -t3
538000: (time: 00:31.837)         signal ok:	QAM_AUTO f = 538000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        Info: no data from PAT after 6 seconds
        deleting (QAM_AUTO f = 538000 kHz I999B8C999D999T999G999Y999 (0:0:0))

crazycat69, 2021-01-13:
Look like you have very bad signal, so you not have any TS stream.

ahmhoseini, 2021-01-13:
As I mentioned before, TBS6285 works fine with the same wire in the same system, and scans all frequencies with success, so it doesn't seems signal problem to me.

ahmhoseini, 2021-01-13:
Do anybody here have 6205 to test with latest driver to confirm it works with this driver?

takatyan14, 2021-01-17:
I have a TBS6205, and have been using it on CentOS 8 (kernel: 4.18.0-80.11.2.el8_0.x86_64) since December 2019. It have worked well since then (It receives signal well, and the video and the audio has no noise.)
Some days ago, I suddenly made up my mind to upgrade OS and driver, I installed Fedora CoreOS (kernel: 5.9.13-200.fc33.x86_64) to my machine, cloned this latest driver repository again, and built it and installed the driver.
At first, It looks like it's working well, but when one tuner is being used, then I open an another tuner, and the first one loses the signal for a moment, the video and the audio have some noise. With that CentOS 8, there was no such thing.

I tried various things and I tried some revisions of the driver, so I had deleted the cloned local repository with that CentOS 8, I couldn't find out the cause. I searched similar issue here, but there aren't anything. Because it can't be helped, I gave up , I restored that CentOS 8 from my backup, and I use it.
Today I found this issue, then I tried this post.

ahmhoseini, 2021-01-18:
After working on reception side, and recompiling tbsdtv drivers, I could confirm TBS6205 works normally, and there is no need to change si2168_config.ts_clock_inv for this card.

The problem was, it seems my TBS6285 has a much better sensitivity, so it received all channels, but TBS6205 failed to receive in the same configuration. After working over signal reception, now both cards work well.

Sorry for disruption.

closed 2021-01-18 by ahmhoseini