tbsdtv/linux_media #229: TBS 5520SE can only use 1 tuner with Tvheadend

smp79, 2021-01-12:
I opened an [issue report](https://tvheadend.org/issues/5990) on Tvheadend site. Can anyone help to fix this?

The issue is that both tuners fail to tune to anything until one of them is disabled. I'm not sure if it is a TVH or a driver issue.

crazycat69, 2021-01-12:
you must set frontends as linked in TVH (so if one used, second not usedby TVH)

smp79, 2021-01-12:
I tried that, it doesn't fix the issue.
I did more testing and it can fail even with 1 tuner enabled. 

В VDR всё идеально работает. А в этом дурацком TVH сначала вроде бы работает, но после перезагрузки "no free adapters available" в Коди. Если в настройках TVH выключить и включить 1 тюнер, то может опять заработать.

smp79, 2021-01-14:
У меня подозрение, что tvheadend-у не нравится, что у 5520SE 2 тюнера подцеплены к 1 демоду. Что подкрутить в драйвере, чтобы был только один FE на котором только DVB-S/S2?

crazycat69, 2021-01-15:
Вообще-то там два фронтєнда на адаптер, что нужно так как физически два тюнера подключены к демодулятору. Очередность регистрации в tbs5520se_frontend_attach()
https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/usb/dvb-usb/tbs5520se.c#L254

smp79, 2021-01-15:
> Вообще-то там два фронтенда на адаптер, что нужно так как физически два тюнера подключены к демодулятору.

Это понятно, просто хочу добраться до корня проблемы с tvh.  Хочу собрать тестовый драйвер, чтобы tvh видел только DVB-S тюнер. Если это решит проблему, по крайней мере будет видно, в какую сторону копать.

closed 2021-01-20 by smp79