tbsdtv/linux_media #232: 6909x, 6903x v2 quality of signal is bad

JianSun1, 2021-01-27:
I ordered 6909xv2 and 6903xv2. I also have 6903x and 6909x first version. It seems like 6903xV1 is the best about receiving signal. The receiving quality is listed below
1- 6903x V1 (is the best) 
2- 6909x V1 (is good)
3- 6909x V2 
4- 6903x V2 (the worst)

The problem happens in every locked channels, not to specific one. I randomly locked into a signal and take screenshots of 6909 v2 vs v1. I wont put screenshots of 6903xv2; it is the worst. The snr value is about 4 db worse than the first version of 6903x.

This is 6909xV2
[image]

This is 6909xV1
[image]

I am thinking of the problem and the issue can be the vglna. I have tried to setfield and getfield of vglna register [in here](https://github.com/tbsdtv/linux_media/commit/808e62eaff848b35690b4175b0bb877617357538#diff-50704091cdddc93579e767bb95bbcf8f1308d8fbd0e007ba676d4e67a59c085eR11343) . If i set field to 1 and try to getfield and print the value of set register, it stays at 0. I think there is something wrong at setting the registers. A little worked and searched on the internet and found how to set the registers, [tried this source code](https://github.com/myorangedragon/longmynd/blob/master/stvvglna.c) and again results are not good.

The 6909xv2 vglnamode is set to 1 in [here](https://github.com/tbsdtv/linux_media/commit/808e62eaff848b35690b4175b0bb877617357538#diff-a328b5417e349f548e2b579ea8dfcbdd3d9fac58eb1089f07d4f931748def03eR940)

The 6903xv2 vglnamode is set to 2 in [here](https://github.com/tbsdtv/linux_media/commit/808e62eaff848b35690b4175b0bb877617357538#diff-a328b5417e349f548e2b579ea8dfcbdd3d9fac58eb1089f07d4f931748def03eR928)

There is no vglna init option for 6903xV2 [this code](https://github.com/tbsdtv/linux_media/commit/808e62eaff848b35690b4175b0bb877617357538#diff-56f01a981acca49296f94029e7489654ca818800b22bdf6c8f764fcd508d728bR223)

This issue may seems its not a big problem (2-4 db less SNR values in new manifactured cards) but when watching tv streams, the glitches happens on tv stream because of low SNR values. 
Is it possible to bypass the vglna chip in new Version2 cards and use these cards as version 1?

The version1 is not selling anymore so it is impossible to buy old versions. At least both versions could be sold. Will be any update?

crazycat69, 2021-01-27:
You can compare signal strength in dBm ?

crazycat69, 2021-01-28:
Try enable VGLNA on init - **stvvglna_set_standby(vglna_handle,0);**
https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/dvb-frontends/stid135/stid135-fe.c#L239

JianSun1, 2021-01-29:
> You can compare signal strength in dBm ?
I am unable to use crazyscan tool, i should try to install windows7 32 bit again and check them later.

> Try enable VGLNA on init - **stvvglna_set_standby(vglna_handle,0);**
> https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/dvb-frontends/stid135/stid135-fe.c#L239

I tried but it didnt work. Also, i have checked the source code again, I think something wrong in chip initialization. [Current Chipmode](https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/dvb-frontends/stid135/stid135_drv.c#L11464) should be changed to [this mode](https://github.com/canalplus/r7oss/blob/fb30ac72c9e541f0eb8ef24a2899cd17e306733b/G6/src/frontend_engine-modules-15.2.1.87.10.12.4/linux/driver/demod/vglna/stvvglna.c#L80).

My knowledge on 'i2c' is poor, so i dont want to mess so much. if i change the chipmode, i can see that the register values are changing. Verifying the registers by using the chipGetOneRegister function.

crazycat69, 2021-02-01:
Look like your issue not related to LNA. Some changes in stid135 driver until latest 2 years give some degradation (i catch it with 6903x,6909x V1also).

Can you test driver from my [media_build](https://github.com/crazycat69/media_build) + [linux-media](https://github.com/crazycat69/linux_media) ? 6903x, 6909x V2 also supported (without VGLNA init, look like work with default settings).

deeptho, 2021-02-03:
@JianSun1 ,

I had similar problems to yours. For me the soluttion was

> Try enable VGLNA on init - stvvglna_set_standby(vglna_handle,0);

as suggested by crazycat69.

I have been testing the code heavily on tbs6909xV2 and I have good results. SNR
is the same (within 0.5dB) as on tbs6909X V1.

You can double check correct registers in  my blindscan code 
https://github.com/deeptho/linux_media
which has diverged quite a bit from the official source.
It also has some debug code to download the registers from the chips as 
a binary file from /proc/hchip_soc      and /proc/hchip_stid135
 
There should be also a /proc/hchip_stvvglna, but currently it seems to be 
missing on my own computer. Perhaps I disabled the code. Note that the implementation
of the debug code is "quick and dirty" (not professional code)

Not so long ago I also tested with the tbsdtv drivers from this site (version ), and with the changed value 
for stvvglna_set_standby. Those also worked. I think the version I used as
 7bed1616b

crazycat69, 2021-02-16:
e621b3dac857b3f84f1c1405d4b52394b04ac8be

mustang011, 2021-04-11:
> Look like your issue not related to VGLNA. Some changes in stid135 driver until latest 2 years give some degradation (i catch it with 6903x,6909x V1also).
> 
> Can you test driver from my [media_build](https://github.com/crazycat69/media_build) + [linux-media](https://github.com/crazycat69/linux_media) ? 6903x, 6909x V2 also supported (without VGLNA init, look like work with default settings).

I have TBS 6205 (DVB-T) and TBS 6909X-V1 (DVB-S2).
I am using Linux Gentoo and VDR 2.4.1-r3 with vnsiserver 1.8.0 plugin as server and RPI4 with LibreELEC 9.2.6 (Kodi 18.9) as client. I tested old github crazycat69 drivers. On old drivers, the channels loaded 3 seconds faster. Unfortunately, from time to time the program crashed (error no channel data) the DVB-T or DVB-S2 channel did not load. Can drivers for TBS 6909-X be compiled without the "CONFIG_FRAME_VECTOR" service, as I write it here - https://github.com/crazycat69/media_build/commit/209bb8ff9a99a3f4cabd0190d7a9849699655d6f ? I install the drivers according to the instructions - https://www.tbsiptv.com/download/tbs6909x/tbs6909x_user_guide.pdf .
I also tested a newer version of VDR 2.4.6-r1, but programs were loading slower and I went back to 2.4.1-r3.

closed 2021-04-11 by crazycat69

owaidh123, 2022-12-01:
I want a link to download the signal test program, how to apply it