tbsdtv/linux_media #235: Build errors armbian / kernel 5.10.12

Lamarqe, 2021-02-10:
I am running into the following compile error during compilation with kernel 5.10.12:

/home/scritch/tbsdtv/media_build/v4l/m88rs6060.c: In function 'm88rs6060_probe':
/home/scritch/tbsdtv/media_build/v4l/m88rs6060.c:1827:6: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 1827 |      i2c_new_dummy(client->adapter, dev->config.tuner_adr);
      |      ^~~~~~~~~~~~~
/home/scritch/tbsdtv/media_build/v4l/m88rs6060.c:1826:20: warning: assignment to 'struct i2c_client *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 1826 |  dev->tuner_client =
      |                    ^
cc1: some warnings being treated as errors

This is very similar to this issue: https://github.com/tbsdtv/media_build/issues/32
The build error can be fixed using the same pattern as used here: https://github.com/tbsdtv/linux_media/commit/e29db5d8e23a3bb95d88669a6109ab956d02064e lines 1378-1380

A 2nd build error is related with include hierarchy changes:

In file included from /home/scritch/tbsdtv/media_build/v4l/rkisp1-capture.c:21:
/home/scritch/tbsdtv/media_build/v4l/rkisp1-common.h:16:10: fatal error: linux/rkisp1-config.h: No such file or directory
   16 | #include <linux/rkisp1-config.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

The include file is available here:
/home/scritch/tbsdtv/media/include/uapi/linux/rkisp1-config.h

This folder is apparently not included in the make include folders. As a workaround, I manually copied the file to a folder which is included:

cp media/include/uapi/linux/rkisp1-config.h media_build/linux/include/uapi/linux

crazycat69, 2021-02-11:
c66f7e863fbc7a58299a4e1b890369115f0df060

Lamarqe, 2021-02-11:
I have bad news. Even though I was able to compile everything correctly after having fixed the 2 issues above, I am not able to use the build. Reason: Using it, there are many distrubances, (stutter, green color effects). I have two DVB-S2 receivers connected: TBS 5922 (QBOX2) and DvbSky S960. Both receivers are affected by this problem.

As soon as I revert to the original kernel build, the problem disappears (Dvbsky only, TBS remains unusable as the TBS drivers are not mainlined). 
The problem did not exist with my previous build (still using Kernel 5.8 and TBS media build from ~August 2020).

crazycat69, 2021-02-11:
What usb host ? EHCI or XHCI ?

Lamarqe, 2021-02-11:
Odroid C2, so I believe EHCI:

```
dmesg | grep -i 'ehci\|xhci'
[    1.786419] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.786452] ehci-pci: EHCI PCI platform driver
[    1.786491] ehci-platform: EHCI generic platform driver
```

crazycat69, 2021-02-12:
Maybe some 5.10 specific issues (usb hostdriver or dvb-usb susbsustem ). I test only with LTS kernels (now 5.4 with ubuntu 20.04)

Lamarqe, 2021-02-12:
5.10 is an LTS kernel. This is why I updated. Raspberry PI OS switched to 5.10 a few weeks ago, too.

When is it planned that TBS will support 5.10 kernel?

If you need debug output or testing, let me know. I will try to support.

crazycat69, 2021-02-12:
5.10 is stable kernel - https://www.kernel.org
5,4 is latest LTS

5.10 generic kernels suppoted in current media_build (at least on x86).
Your specific issues maybe related to SoC specific kernel. I plan test 5.10 after ubuntu 21.04 release (or with HWE update for 20.04)

Lamarqe, 2021-02-12:
We could have an argument about the LTS topic (https://www.kernel.org/category/releases.html), but not worth or necessary to discuss here.

Thank you for your explanations. 
Mainlining of Odroidc2 is complete. So we talk about a vanilla kernel here. 

If it helps, I could give it a try on a raspberry, armhf and maybe even arm64.

For the time being, I will switch of my TBS receiver.
Still, thanks alot for the prompt responses and taking the time.

zhang111chao, 2021-02-21:
i know how to install tbs card driver in rpi V5.4,  if you want to install in v5.10 , please add skype jack_2431 and setup teamviewer , i will check .

Lamarqe, 2021-02-21:
Installation on 5.10 works fine for me. Thanks for offering the help!

It is not about compile issues or any problems related to the installation.
The issue is that the installed build causes problems (many distrubances, stutter, green color effects). 

Do you want me to provide specific logs so you can analyze the problems?

closed 2024-08-05 by Lamarqe