tbsdtv/linux_media #236: compile errors ubuntu 18.04

hamishegg, 2021-02-17:
trying to install latest and having compile errors

root@i7:/opt/linux_media-latest# cd ..
root@i7:/opt# git clone https://github.com/tbsdtv/media_build.git
Cloning into 'media_build'...
remote: Enumerating objects: 619, done.
remote: Counting objects: 100% (619/619), done.
remote: Compressing objects: 100% (259/259), done.
remote: Total 6566 (delta 413), reused 548 (delta 360), pack-reused 5947
Receiving objects: 100% (6566/6566), 1.31 MiB | 1.32 MiB/s, done.
Resolving deltas: 100% (4742/4742), done.
root@i7:/opt# git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
Cloning into './media'...
remote: Enumerating objects: 75928, done.
remote: Counting objects: 100% (75928/75928), done.
remote: Compressing objects: 100% (72495/72495), done.
remote: Total 75928 (delta 6272), reused 13479 (delta 2646), pack-reused 0
Receiving objects: 100% (75928/75928), 202.91 MiB | 6.42 MiB/s, done.
Resolving deltas: 100% (6272/6272), done.
Checking out files: 100% (71532/71532), done.
root@i7:/opt# cd media_build
root@i7:/opt/media_build# make dir DIR=../media
make -C linux/ dir DIR="../../media"
make[1]: Entering directory '/opt/media_build/linux'
rm -rf drivers include sound mm .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
Searching in ../../media/Makefile for kernel version.
./use_dir.pl ../../media
sync file: include/uapi/linux/cec.h
sync file: include/uapi/linux/ivtvfb.h
sync file: include/uapi/linux/cec-funcs.h
sync file: include/uapi/linux/uvcvideo.h
sync file: include/linux/kconfig.h
sync file: include/uapi/linux/media-bus-format.h
sync file: include/linux/pci_ids.h
sync file: include/uapi/linux/meye.h
sync file: include/uapi/linux/v4l2-mediabus.h
sync file: include/trace/events/v4l2.h
sync file: include/uapi/linux/v4l2-controls.h
sync file: include/uapi/linux/videodev2.h
sync file: include/uapi/linux/v4l2-subdev.h
sync file: include/linux/ti_wilink_st.h
sync file: include/uapi/linux/v4l2-dv-timings.h
sync file: include/linux/hdmi.h
sync file: include/linux/via-core.h
sync file: include/uapi/linux/smiapp.h
sync file: include/linux/overflow.h
sync file: include/uapi/linux/lirc.h
sync file: include/trace/events/vb2.h
sync file: include/uapi/linux/media.h
sync file: include/sound/aci.h
sync file: include/uapi/linux/xilinx-v4l2-controls.h
sync file: include/linux/fixp-arith.h
sync file: include/uapi/linux/ivtv.h
sync file: sound/pci/bt87x.c
sync file: include/uapi/linux/ccs.h
sync file: include/uapi/linux/max2175.h
sync file: include/uapi/linux/v4l2-common.h
sync file: include/linux/of_graph.h
sync file: include/uapi/linux/usb/video.h
sync file: include/linux/mmc/sdio_ids.h
sync file: include/linux/dma/xilinx_dma.h
sync file: include/misc/altera.h
sync file: include/trace/events/pwc.h
sync dir: include/dt-bindings/media/
sync dir: include/uapi/linux/dvb/
sync dir: include/media/
sync dir: include/linux/platform_data/media/
sync dir: drivers/misc/altera-stapl/
sync dir: drivers/media/
sync dir: drivers/staging/media/
[... 693 lines cut ...]
make[3]: *** Waiting for unfinished jobs....
In file included from ./include/linux/printk.h:332:0,
                 from ./include/linux/kernel.h:15,
                 from ./include/asm-generic/bug.h:19,
                 from ./arch/x86/include/asm/bug.h:83,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /opt/media_build/v4l/config-compat.h:12,
                 from /opt/media_build/v4l/compat.h:10,
                 from <command-line>:0:
/opt/media_build/v4l/si2183.c: In function 'si2183_probe':
./include/linux/dynamic_debug.h:157:25: warning: 'ret' may be used uninitialized in this function [-Wmaybe-uninitialized]
  _dynamic_func_call(fmt,__dynamic_dev_dbg,   \
                         ^~~~~~~~~~~~~~~~~
/opt/media_build/v4l/si2183.c:1488:6: note: 'ret' was declared here
  int ret;
      ^~~
Makefile:1652: recipe for target '/opt/media_build/v4l' failed
make[2]: *** [/opt/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.4.1-050401-generic'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/opt/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

0x2A, 2021-02-18:
in file /opt/media_build/v4l/mxl58x.c line 600 replace `for(i=0;` with `for(int i=0;` 
and in line 616 the same

crazycat69, 2021-02-18:
0bc8cfa5b40283ec1f075d690c94c192c375b051

closed 2021-02-18 by crazycat69