tbsdtv/linux_media #239: Compilation errors on Debian with backported Kernel 5.10

bistory, 2021-04-08:
Hi !

I'm using Debian for a while and compiling TBS drivers was usually fine but since 5.10 is released, I'm not able to compile it successfully...
Here are the errors I get : 

```
ERROR: modpost: "__devm_regmap_init_sccb" [/usr/src/media_build/v4l/ov772x.ko] undefined!
make[4]: *** [/usr/src/linux-headers-5.10.0-0.bpo.4-common/scripts/Makefile.modpost:111: /usr/src/media_build/v4l/Module.symvers] Error 1
make[4]: *** Deleting file '/usr/src/media_build/v4l/Module.symvers'
make[3]: *** [/usr/src/linux-headers-5.10.0-0.bpo.4-common/Makefile:1717: modules] Error 2
make[2]: *** [/usr/src/linux-headers-5.10.0-0.bpo.4-common/Makefile:185: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.10.0-0.bpo.4-amd64'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/usr/src/media_build/v4l'
make: *** [Makefile:26: all] Error 2
```

Kernel version : 5.10.0-0.bpo.4-amd64
Debian 10 (Buster) with backports enabled

Thanks for your help !

bistory, 2021-04-26:
Tested with latest commits but it still fails with the same error...

bistory, 2021-05-03:
Ok I found a solution, in your media_build folder, after doing `make allyesconfig`, edit the file "v4l/.config"

Find these lines and disable the modules (by putting =n instead of =m)

```
CONFIG_VIDEO_RDACM20=m
CONFIG_VIDEO_OV9650=m
CONFIG_VIDEO_OV772X=m
```

crazycat69, 2021-05-03:
you can use menu-based console setup **make menuconfig**

bistory, 2021-05-03:
@crazycat69 unfortunately, I tried menuconfig but it doesn't work, ncurses is installed so I don't know why

crazycat69, 2021-05-03:
some errors while parsing Kconfigs ?

bistory, 2021-05-05:
@crazycat69 here is the error I get 

```
# make menuconfig
make -C /usr/src/media_build/v4l menuconfig
make[1]: Entering directory '/usr/src/media_build/v4l'
make -C /lib/modules/5.10.0-0.bpo.5-amd64/build -f /usr/src/media_build/v4l/Makefile.kernel config-targets=1 mixed-targets=0 dot-config=0 SRCDIR=/lib/modules/5.10.0-0.bpo.5-amd64/source v4l-mconf
make[2]: Entering directory '/usr/src/linux-headers-5.10.0-0.bpo.5-amd64'
/usr/src/media_build/v4l/Makefile.kernel:39: warning: overriding recipe for target 'v4l-mconf'
/lib/modules/5.10.0-0.bpo.5-amd64/source/Makefile:181: warning: ignoring old recipe for target 'v4l-mconf'
make[2]: *** No rule to make target 'scripts_basic', needed by 'v4l-mconf'.  Stop.
make[2]: Leaving directory '/usr/src/linux-headers-5.10.0-0.bpo.5-amd64'
make[1]: *** [Makefile:387: /lib/modules/5.10.0-0.bpo.5-amd64/build/scripts/kconfig/mconf] Error 2
make[1]: Leaving directory '/usr/src/media_build/v4l'
make: *** [Makefile:26: menuconfig] Error 2
```

crazycat69, 2021-05-05:
Look like some Debian specific issue. Ubuntu work ok.