tbsdtv/linux_media #240: ast driver fails to load on Ubuntu 20.04

dabb1er, 2021-04-13:
After installation of latest git pull on Ubuntu 20.04 the ast driver fails to load (which leaves IPMI display on Supermicro server stuck at 640x480). This has always been the case with Ubuntu 20.04 (not just a recent thing). New kernel upgrades prior to install of tbsdtv software allows ast module to load correctly (so its down to this software).

Running "modprobe ast" produces the following errors in  syslog:

[45326.666262] drm_kms_helper: disagrees about version of symbol cec_delete_adapter
[45326.666264] drm_kms_helper: Unknown symbol cec_delete_adapter (err -22)
[45326.666585] drm_kms_helper: disagrees about version of symbol cec_transmit_attempt_done_ts
[45326.666586] drm_kms_helper: Unknown symbol cec_transmit_attempt_done_ts (err -22)
[45326.667233] drm_kms_helper: disagrees about version of symbol cec_s_phys_addr
[45326.667234] drm_kms_helper: Unknown symbol cec_s_phys_addr (err -22)
[45326.667273] drm_kms_helper: disagrees about version of symbol cec_s_conn_info
[45326.667274] drm_kms_helper: Unknown symbol cec_s_conn_info (err -22)
[45326.667344] drm_kms_helper: disagrees about version of symbol cec_s_phys_addr_from_edid
[45326.667345] drm_kms_helper: Unknown symbol cec_s_phys_addr_from_edid (err -22)
[45326.668903] drm_kms_helper: disagrees about version of symbol cec_unregister_adapter
[45326.668904] drm_kms_helper: Unknown symbol cec_unregister_adapter (err -22)
[45326.669380] drm_kms_helper: disagrees about version of symbol cec_allocate_adapter
[45326.669381] drm_kms_helper: Unknown symbol cec_allocate_adapter (err -22)
[45326.669975] drm_kms_helper: disagrees about version of symbol cec_fill_conn_info_from_drm
[45326.669976] drm_kms_helper: Unknown symbol cec_fill_conn_info_from_drm (err -22)
[45326.670100] drm_kms_helper: disagrees about version of symbol cec_received_msg_ts
[45326.670101] drm_kms_helper: Unknown symbol cec_received_msg_ts (err -22)
[45326.670428] drm_kms_helper: disagrees about version of symbol cec_register_adapter
[45326.670429] drm_kms_helper: Unknown symbol cec_register_adapter (err -22)

crazycat69, 2021-04-13:
remove kernel media modules before install (see[ media_build\install.sh](https://github.com/tbsdtv/media_build/blob/master/install.sh) script)

dabb1er, 2021-04-13:
Installed software using install.sh script (so it removed the kernel modules).

Anything else to try?

crazycat69, 2021-04-13:
Run this before **make install**
```
rm -r -f /lib/modules/$(uname -r)/kernel/drivers/media
rm -r -f /lib/modules/$(uname -r)/kernel/drivers/staging/media
```

kalashnikov2, 2021-04-18:
hi
I am compiling / installing on Ubuntu 20.04.2, fresh install  , no media drivers installed before
after installing and rebooting
ubuntu does not load nvidia video driver *

and the video resolution is forced/ stuck at 640x480

I consider that it is related to this problem, 
otherwise I must create a new isues?

bastard79, 2021-04-23:
i have the same issues with ubuntu 21.04 with kernel 5.11. my amdgpu and acl887 stops working after compile drivers. Drivers TBS working, but system is unusable without gpu modules :(

project looka like abandoned, 3 month without commits, need to back to ubuntu 20.10

crazycat69, 2021-04-23:
same issues with [upstream media_build](https://git.linuxtv.org/media_build.git)

bastard79, 2021-04-24:
the problem is kernel, 5.11 is not supported, so if you use ubuntu 21.04 must install kernel 5.8 from repo ubuntu 20.10 and working like a charm. idk about kernel 5.9+ in other distros

crazycat69, 2021-04-24:
media_build and linux_media updated.

dabb1er, 2021-04-25:
Still same problem after media_build and linux_media updated.

Mnyromyr, 2021-04-26:
I see the same issue as dabb1er and kalashnikov2:
- many unknown symbols cec_*
- nvidia-driver doesn't load (460.73.01 here)
- display setting stuck at 1024x768
- media_build freshly cloned & built
- Linux  5.8.0-50-generic #56-Ubuntu SMP Mon Apr 12 17:18:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Mnyromyr, 2021-04-27:
JFTR (and to state the obvious): reinstalling the respective kernel module package
                       sudo aptitude reinstall linux-modules-5.8.0-50-generic
restores screen resolution and nvidia-settings, but of course my DVB-S doesn't work anymore … ^_^

crazycat69, 2021-04-27:
Just install 20.04 HWE kernel (5.8.0-50) and media_build - nvidia 460 driver work ok.
drm_kms_helper module loaded and depend from cec module (builded with media_build).
ast module also loaded without error.

dabb1er, 2021-04-27:
crazycat69 - can you give clear instructions on how to do this. I normally run the following:

cd media_build
make dir DIR=../media
make allyesconfig
make -j4

But this does not work if I do not clone media folder (or its empty). Note I already have HWE kernel installed.

crazycat69, 2021-04-27:
i just run build script (this download latest media sources snapshot and firmware from https://github.com/tbsdtv/media_build/releases/tag/latest). And modules install with install.sh script.

crazycat69, 2021-04-27:
But this not critical, bedause cec module recompiled anyway. Now i enable Secure boot, but nvidia 460 still work and dvb devices also. drm_kms_helper module loaded ok.

dabb1er, 2021-04-27:
Tried build script to pull down code and install.sh to install but still getting cec errors when trying to install ast module.

dabb1er, 2021-05-03:
Just confirmed and it is the change between kernel 5.8.0-44-generic and 5.8.0-45-generic when the issue occurs.

If I regress back to kernel 5.8.0-44-generic then both the saa716x_tbs_* and ast modules load:

xxxx@xxxx:~$ lsmod | grep saa
saa716x_tbs_dvb        65536  0
cx24117                36864  1 saa716x_tbs_dvb
saa716x_core           86016  1 saa716x_tbs_dvb
tas2101                24576  1 saa716x_tbs_dvb
dvb_core              147456  3 cxd2820r,saa716x_tbs_dvb,saa716x_core
i2c_algo_bit           16384  4 igb,ivtv,saa716x_tbs_dvb,ast
xxxx@xxxx:~$ lsmod | grep ast
ast                    53248  1
drm_vram_helper        24576  1 ast
drm_kms_helper        217088  4 drm_vram_helper,ast
drm                   552960  7 drm_kms_helper,drm_vram_helper,ast,drm_ttm_helper,ttm
i2c_algo_bit           16384  4 igb,ivtv,saa716x_tbs_dvb,ast

xxxx@xxxx:~$ uname -a
Linux masterbackend2 5.8.0-44-generic #50~20.04.1-Ubuntu SMP Wed Feb 10 21:07:30 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

crazycat69, 2021-05-05:
work well with latest 20.04 hwe kernel 5.8.0-50

bastard79, 2021-05-15:
drivers are still broken :(
on ubuntu 21.04 after compile without errors with kernel 5.11, gpu is unusable :(
so i install kernel 5.8 from ubuntu 20.10, and compile again without error, but after some time of watching tv pc hangs and dmesg say

`[  192.681880] gmc_v9_0_process_interrupt: 23 callbacks suppressed
[  192.681889] amdgpu 0000:06:00.0: amdgpu: [gfxhub0] retry page fault (src_id:0                                                                                                                      ring:0 vmid:2 pasid:32774, for process kodi.bin pid 1193 thread kodi.bin:cs0 pi                                                                                                                     d 1311)
[  192.681896] amdgpu 0000:06:00.0: amdgpu:   in page starting at address 0x0000                                                                                                                     800101c22000 from client 27
[  192.681899] amdgpu 0000:06:00.0: amdgpu: VM_L2_PROTECTION_FAULT_STATUS:0x0020                                                                                                                     1031
[  192.681902] amdgpu 0000:06:00.0: amdgpu:      Faulty UTCL2 client ID: 0x8
[  192.681904] amdgpu 0000:06:00.0: amdgpu:      MORE_FAULTS: 0x1
[  192.681907] amdgpu 0000:06:00.0: amdgpu:      WALKER_ERROR: 0x0
[  192.681909] amdgpu 0000:06:00.0: amdgpu:      PERMISSION_FAULTS: 0x3
[  192.681911] amdgpu 0000:06:00.0: amdgpu:      MAPPING_ERROR: 0x0
[  192.681913] amdgpu 0000:06:00.0: amdgpu:      RW: 0x0
`
maybe new mesa broke this? i have no other compiling stuff, only tbs drivers, of course without tbs drivers on fresh ANY kernel issue dont exist.

any idea whats wrong?

crazycat69, 2021-05-15:
My NV GPU now also not work after initrd update (with recompiled cec and rc-code modules).
Temporary solution reinstal kernel-modules and kernel-extra-modules packages + compile media_build withot CEC and RC_CORE support + install as extra modules (so kernel cec and rc-core modules untouched and pre-loaded from initrd)
https://github.com/crazycat69/media_build/commit/a45df85b99f6f2844c8170ffc1e2b374a88ef080

bastard79, 2021-05-16:
with your patch on kernel 5.11 amdgpu looks good, but sound still dead
 
 3.186354] snd_usb_audio: disagrees about version of symbol media_devnode_remove
[    3.186358] snd_usb_audio: Unknown symbol media_devnode_remove (err -22)
[    3.186451] snd_usb_audio: disagrees about version of symbol __media_device_register
[    3.186452] snd_usb_audio: Unknown symbol __media_device_register (err -22)
[    3.186491] snd_usb_audio: disagrees about version of symbol media_create_intf_link
[    3.186492] snd_usb_audio: Unknown symbol media_create_intf_link (err -22)
[    3.186499] snd_usb_audio: disagrees about version of symbol media_create_pad_link
[    3.186500] snd_usb_audio: Unknown symbol media_create_pad_link (err -22)
[    3.186542] snd_usb_audio: disagrees about version of symbol media_device_register_entity
[    3.186543] snd_usb_audio: Unknown symbol media_device_register_entity (err -22)
[    3.186602] snd_usb_audio: disagrees about version of symbol media_entity_pads_init
[    3.186603] snd_usb_audio: Unknown symbol media_entity_pads_init (err -22)
[    3.186618] snd_usb_audio: disagrees about version of symbol media_remove_intf_link
[    3.186619] snd_usb_audio: Unknown symbol media_remove_intf_link (err -22)
[    3.186654] snd_usb_audio: disagrees about version of symbol media_device_delete
[    3.186655] snd_usb_audio: Unknown symbol media_device_delete (err -22)
[    3.186669] snd_usb_audio: disagrees about version of symbol media_devnode_create
[    3.186670] snd_usb_audio: Unknown symbol media_devnode_create (err -22)
[    3.186684] snd_usb_audio: disagrees about version of symbol media_device_unregister_entity
[    3.186684] snd_usb_audio: Unknown symbol media_device_unregister_entity (err -22)
[    3.186697] snd_usb_audio: disagrees about version of symbol media_device_usb_allocate
[    3.186698] snd_usb_audio: Unknown symbol media_device_usb_allocate (err -22)

crazycat69, 2021-05-16:
snd_usb_audio depend from mc module (media controller core). i will try disable it also (video camera support also).

bastard79, 2021-05-22:
@crazycat69 is there any patch on the way???

or maybe how to revert last patches from drivers ? i was compiled drivers in september/november for kernel 5.8 without any issues, now compiling is without error, but screen freeze ang hangs on and gpu problems so i think problem is with last driver patches? dkms ?

dabb1er, 2021-05-22:
There was a change in the 5.8 kernel between 5.8.0-44-generic and 5.8.0-45-generic which caused issued in loading other modules (for me ast for screen driver). The latest git pull works perfectly for me on kernel 5.8.0-44-generic but if I upgrade to any later kernel version then ast driver fails with "disagrees about version of symbol" errors. I have now re-installed kernel 5.8.0-44-generic and locked system from kernel upgrades (not ideal from patching point of view but it works).

bastard79, 2021-05-22:
i was using ubuntu 20.10 and latest kernel works perfect, now im upgrade to ubuntu 21.04, and kernel 5.11 dont work at all after compile tbs (gpu, sound as mentioned before) so im instaled kernel from 20.10 5.8.0-25 -generic, and compile drivers without errors, but after 15-30 min screen freeze, kodi hangs on, only restart helps.

so latest drivers are not compatible with something from ubuntu 21.04, maybe mesa? idk. but im sure something is broken and i dont know to fix this. on fresh any kernel system works without issues of gpu or sound.

crazycat69, 2021-05-22:
strange, look like problems with your GPU/APU (maybe gpu driver)

bastard79, 2021-05-22:
yes this is problem with driver, but on fresh kernel there is no issues, problems starts after compile tbs driver ;(

so now without tbs driver everything  works perfect but i cant watch tv, with tbs driver i can watch tv but only 15-30min and need restart

crazycat69, 2021-05-22:
what dvb card ? maybe general problems with insterrupt and DMA handling on your platform

bastard79, 2021-05-22:
i have TBS6902 and clone TBS6922, last two years working without problems, problems start after upgrade to ubuntu 21.04 and compile tbs drivers, on default kernel 5.11 tbs drivers dont work at all (problem gpu and sound)

idk, i dont have enough knowledge to diagnose problem, maybe if tbs works on kernel 5.11 everything will be ok.

crazycat69, 2021-05-22:
Beter use LTS distro like ubuntu 18.04 or 20.04 with 5.4 kernel

papakai, 2021-07-19:
I bought a TBS 6904 about a half year ago and I can just forward how disappointed I am regarding this decision. The better choice would have been a Digital Devices Device. I had such a trouble getting started, different installation guides in www as you guys provide here. In addition I had the driver again and again stopping to work after few days without having the system touched. I finally started again using my old DD Cine S2 and put my TBS one in our HTPC in basement that is just rately used. My support request addressed to the team at tbs-technology.de still unanswered. My last order at TBS, sad to say that.

tbsiptv, 2021-07-20:
> I bought a TBS 6904 about a half year ago and I can just forward how disappointed I am regarding this decision. The better choice would have been a Digital Devices Device. I had such a trouble getting started, different installation guides in www as you guys provide here. In addition I had the driver again and again stopping to work after few days without having the system touched. I finally started again using my old DD Cine S2 and put my TBS one in our HTPC in basement that is just rately used. My support request addressed to the team at tbs-technology.de still unanswered. My last order at TBS, sad to say that. I am just writing it down here as I ended up in this helpful thread. Still it did not lower my frustration level.

Hi Sir,

Please direct send email to support@tbsdtv.com we will handle this , you can setup SSH remote let us check the driver 

Thanks

Kind Regards

papakai, 2021-07-20:
@tbsiptv I got the card last week working under Ubuntu 18, Kernel 5.3, NVidia 365 (the setup as far as I remember). Still, it took really a long while until I came there - started with 20.04, where it initially somehow worked until I noticed that after some time the driver stops working. So, I've worked on an installation bash script and I executed it every time to get that fixed while being always faced with my wife: "Oh hell, it again doesn't work - get it please back up and running". I upgraded then to Ubuntu 21.04 to see if this improves the situation, but wtf - it just was the worst decision (you even suggest LTS versions, but I haven't noticed this in any of the public installation guides). So, it was about reinstalling from scratch, losing configurations settings of the old os and installed apps - but hey, now it works. Great to hear that for you, I guess. Improving your public installation & setup guides is one important topic, I can address. Still, you notice my emotions regarding your product.  If you really want to improve in direction of end-users, get your driver directly shipped with the kernel. This comes probably with more costs, but tbh worth it to gain and KEEP customers and market share. I gave your product a chance as I've been interested in a S2 Quad Tuner and decided by price, a decision I regret today. Feels like 150 bugs thrown out of the window.  I'll eventually re-decide when you came there. Until then, I'll continue with Digital Devices, because especially with Linux Distros they do just better with their Cine S2 / Octopus cards that are out-of-the-box compatible with Linux - but thanks to the pandemic, discontinued - replacement to be expected. I'll continue until then with my old setup. Still thanks for the offer @tbsiptv, just to late in my case.