tbsdtv/linux_media #241: firmware download failed -11

pk1057, 2021-04-27:
Hello,
i often see
[64119.286849] si2157 5-0060: found a 'Silicon Labs Si2158-A20'
[64119.286903] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[64120.129449] si2157 5-0060: firmware download failed -11
for my tbs 6205 card. This occurs on startup of my TVHeadend docker container.
When tvh is started and i disable and enable the tuner inside TVH download works mostly.

OS is  Linux version 5.10.21-Unraid (root@faca68f1c795) (gcc (Debian 10.2.1-6) and newer.

Any possible steps to hunt down this problem ?

Hawkeye1909, 2021-08-18:
Same here with a 6522 on LibreELEC 10RC1.

tydaikho, 2021-12-28:
Same... But, i saw that. USB Stick still works with that message. dvb-tuner-si2141-a10-01.fw . Of course, i get that problem after installing TBS linux (Kernel 5.x )

crazycat69, 2021-12-31:
Just test with ubuntu 20.04 LTS 5.11 kernel and T230. With kernel media_driver same issue:

> 31.12.2021 22:10	usb 1-1.3.3	new high-speed USB device number 15 using ehci-pci
> 31.12.2021 22:10	usb 1-1.3.3	New USB device found, idVendor=0572, idProduct=c688, bcdDevice= 8.00
> 31.12.2021 22:10	usb 1-1.3.3	New USB device strings: Mfr=1, Product=2, SerialNumber=3
> 31.12.2021 22:10	usb 1-1.3.3	Product: EyeTV Stick
> 31.12.2021 22:10	usb 1-1.3.3	Manufacturer: Geniatech
> 31.12.2021 22:10	usb 1-1.3.3	SerialNumber: 160421
> 31.12.2021 22:10	usb 1-1.3.3	dvb_usb_v2: found a 'MyGica Mini DVB-(T/T2/C) USB Stick T230' in warm state
> 31.12.2021 22:10	usb 1-1.3.3	dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
> 31.12.2021 22:10	dvbdev	DVB: registering new adapter (MyGica Mini DVB-(T/T2/C) USB Stick T230)
> 31.12.2021 22:10	usb 1-1.3.3	media controller created
> 31.12.2021 22:10	dvbdev	dvb_create_media_entity: media entity 'dvb-demux' registered.
> 31.12.2021 22:10	i2c i2c-9	Added multiplexed i2c bus 10
> 31.12.2021 22:10	si2168 9-0064	Silicon Labs Si2168-B40 successfully identified
> 31.12.2021 22:10	si2168 9-0064	firmware version: B 4.0.2
> 31.12.2021 22:10	si2157 10-0060	Silicon Labs Si2147/2148/2157/2158 successfully attached
> 31.12.2021 22:10	usb 1-1.3.3	DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
> 31.12.2021 22:10	dvbdev	dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
> 31.12.2021 22:10		Registered IR keymap rc-total-media-in-hand-02
> 31.12.2021 22:10	rc rc0	MyGica Mini DVB-(T/T2/C) USB Stick T230 as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.3/rc/rc0
> 31.12.2021 22:10	rc rc0	lirc_dev: driver dvb_usb_dvbsky registered at minor = 0, scancode receiver, no transmitter
> 31.12.2021 22:10	input	MyGica Mini DVB-(T/T2/C) USB Stick T230 as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.3/rc/rc0/input17
> 31.12.2021 22:10	usb 1-1.3.3	dvb_usb_v2: schedule remote query interval to 300 msecs
> 31.12.2021 22:10	usb 1-1.3.3	dvb_usb_v2: 'MyGica Mini DVB-(T/T2/C) USB Stick T230' successfully initialized and connected
> 31.12.2021 22:10	si2168 9-0064	downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
> 31.12.2021 22:10	si2168 9-0064	firmware version: B 4.0.25
> 31.12.2021 22:10	si2157 10-0060	found a 'Silicon Labs Si2148-A20'
> 31.12.2021 22:10	si2157 10-0060	downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
> 31.12.2021 22:10	si2157 10-0060	downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
> 31.12.2021 22:10	si2157 10-0060	firmware version: 2.1.9
> 31.12.2021 22:10	usb 1-1.3.3	DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
> 31.12.2021 22:10	si2157 10-0060	found a 'Silicon Labs Si2148-A20'
> 31.12.2021 22:10	si2157 10-0060	downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
> 31.12.2021 22:10	si2157 10-0060	firmware download failed -11
>

crazycat69, 2021-12-31:
5.4 is ok, tuner firmware loaded first time. So look like bug introduced in media_tree after 5.10 kernel release.

crazycat69, 2022-01-01:
I make some revert a8dd93c1ab76074476086a174898d474e9967c53
Please try.
I still have issue with Geniatech T230, but TBS 5220 with same chips is ok:

> 01.01.2022 23:37	usb 1-1.3.3	new high-speed USB device number 8 using ehci-pci
> 01.01.2022 23:37	usb 1-1.3.3	New USB device found, idVendor=734c, idProduct=5220, bcdDevice= 0.01
> 01.01.2022 23:37	usb 1-1.3.3	New USB device strings: Mfr=1, Product=2, SerialNumber=0
> 01.01.2022 23:37	usb 1-1.3.3	Product: TBS 5220
> 01.01.2022 23:37	usb 1-1.3.3	Manufacturer: TBS-Tech
> 01.01.2022 23:37	dvb-usb	found a 'TurboSight TBS 5220 DVB-T/T2/C' in cold state, will try to load a firmware
> 01.01.2022 23:37	dvb-usb	downloading firmware from file 'dvb-usb-tbsqbox-id5220.fw'
> 01.01.2022 23:37	tbs5220	start downloading TBS5220 firmware
> 01.01.2022 23:37	dvb-usb	found a 'TurboSight TBS 5220 DVB-T/T2/C' in warm state.
> 01.01.2022 23:37	dvb-usb	will pass the complete MPEG2 transport stream to the software demuxer.
> 01.01.2022 23:37	dvbdev	DVB: registering new adapter (TurboSight TBS 5220 DVB-T/T2/C)
> 01.01.2022 23:37	dvb-usb	MAC address: 22:ab:12:34:56:78
> 01.01.2022 23:37	i2c i2c-9	Added multiplexed i2c bus 10
> 01.01.2022 23:37	si2168 9-0064	Silicon Labs Si2168-B40 successfully identified
> 01.01.2022 23:37	si2168 9-0064	firmware version: B 4.0.2
> 01.01.2022 23:37	si2157 10-0060	Silicon Labs Si2147/2148/2157/2158 successfully attached
> 01.01.2022 23:37	usb 1-1.3.3	DVB: registering adapter 5 frontend 0 (TurboSight TBS 5220 DVB-T/T2/C)...
> 01.01.2022 23:37		Registered IR keymap rc-tbs-nec
> 01.01.2022 23:37	rc rc0	TurboSight TBS 5220 DVB-T/T2/C as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.3/rc/rc0
> 01.01.2022 23:37	rc rc0	lirc_dev: driver dvb_usb_tbs5220 registered at minor = 0, scancode receiver, no transmitter
> 01.01.2022 23:37	input	TurboSight TBS 5220 DVB-T/T2/C as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3.3/rc/rc0/input17
> 01.01.2022 23:37	dvb-usb	schedule remote query interval to 150 msecs.
> 01.01.2022 23:37	dvb-usb	TurboSight TBS 5220 DVB-T/T2/C successfully initialized and connected.
> 01.01.2022 23:37	si2168 9-0064	downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
> 01.01.2022 23:37	si2168 9-0064	firmware version: B 4.0.25
> 01.01.2022 23:37	si2157 10-0060	found a 'Silicon Labs Si2158-A20'
> 01.01.2022 23:37	si2157 10-0060	downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
> 01.01.2022 23:37	si2157 10-0060	firmware version: 2.1.9
> 01.01.2022 23:37	usb 1-1.3.3	DVB: adapter 5 frontend 0 frequency 0 out of range (48000000..870000000)
> 01.01.2022 23:38	si2157 10-0060	found a 'Silicon Labs Si2158-A20'
> 01.01.2022 23:38	si2157 10-0060	downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
> 01.01.2022 23:38	si2157 10-0060	firmware version: 2.1.9
>

tydaikho, 2022-01-02:
I saw this thing, my T230C (dvb-tuner-si2141-a10-01.fw) work well with standard linux kernel 5.1x, i get  "Fail -11" error after install  TBS but DVB stick still works. Other, "downloading firmware" message repeats so many times in dmesg both cases standard and TBS installed. Below message is standard linux kernel, TBS case will be same but including "Fail -11" message. 
..........................................................
> [17828.531509] si2157 5-0060: firmware version: 1.1.11
[18313.270892] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[18313.270898] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[18314.465259] si2168 3-0064: firmware version: D 6.0.13
[18314.484857] si2157 5-0060: found a 'Silicon Labs Si2141-A10'
[18314.484914] si2157 5-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
[18314.484919] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[18315.095433] si2157 5-0060: firmware version: 1.1.11
[18323.800549] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[18323.800556] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[18325.002121] si2168 3-0064: firmware version: D 6.0.13
[18325.021120] si2157 5-0060: found a 'Silicon Labs Si2141-A10'
[18325.021179] si2157 5-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
[18325.021183] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[18325.622623] si2157 5-0060: firmware version: 1.1.11
[18331.799747] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[18331.799753] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[18332.995455] si2168 3-0064: firmware version: D 6.0.13
[18333.014473] si2157 5-0060: found a 'Silicon Labs Si2141-A10'
[18333.014518] si2157 5-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
[18333.014522] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[18333.621403] si2157 5-0060: firmware version: 1.1.11
[18338.912529] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[18338.912536] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[18340.112223] si2168 3-0064: firmware version: D 6.0.13
[18340.131374] si2157 5-0060: found a 'Silicon Labs Si2141-A10'
[18340.131418] si2157 5-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
[18340.131422] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[18340.741476] si2157 5-0060: firmware version: 1.1.11
[18361.397512] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[18361.397518] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[18362.594630] si2168 3-0064: firmware version: D 6.0.13
[18362.613799] si2157 5-0060: found a 'Silicon Labs Si2141-A10'
[18362.613844] si2157 5-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
[18362.613848] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[18363.224410] si2157 5-0060: firmware version: 1.1.11
[18369.397351] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[18369.397357] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[18370.604343] si2168 3-0064: firmware version: D 6.0.13
...................

crazycat69, 2022-01-02:
Fixed 118caba2ee903a0664d4133ceee0ea4229f5cfc4

closed 2022-01-02 by crazycat69