tbsdtv/linux_media #242: tbs6205: tbsecp3 load failed

pk1057, 2021-04-28:
Hello,
with current master build i have the following err:
[  907.130953] videobuf2_vmalloc: Unknown symbol media_frame_vector_to_pages (err -2)
[  907.131023] videobuf2_vmalloc: Unknown symbol media_frame_vector_to_pfns (err -2)

Due to this loading of the drvb driver fails.

crazycat69, 2021-04-28:
kernel version ?

pk1057, 2021-04-28:
Linux 5.10.28 for Unraid (fails)
Build with 5.10.21 works

crazycat69, 2021-04-28:
kernel media modules deleted before media_build install ?

pk1057, 2021-04-29:
I don't think so.
The modules are injected during boot of unraid from a tarfile extracted over the existing ones.

ich777, 2021-05-01:
Hi, @crazycat69 I'm the one who compiles the TBS-OS drivers for Unraid.
Yes the TBS-OS build is installed in a new directory without any modules preinstalled.

crazycat69, 2021-05-01:
media_frame_vector_to_pages() present in videobuf2-common module (part of media_build). So maybe you have installed old videobuf2 modules ? Check please:
**modinfo videobuf2_vmalloc
modinfo videobuf2_common**

ich777, 2021-05-01:
**modinfo videobu2_vmalloc:**
```
filename:       /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-vmalloc.ko
license:        GPL
author:         Pawel Osciak <pawel@osciak.com>
description:    vmalloc memory handling routines for videobuf2
depends:        videobuf2-common,videobuf2-memops
retpoline:      Y
name:           videobuf2_vmalloc
vermagic:       5.10.28-Unraid SMP mod_unload
```

**modinfo videobuf2_common:**
```
filename:       /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-common.ko.xz
license:        GPL
author:         Pawel Osciak <pawel@osciak.com>, Marek Szyprowski
description:    Media buffer core framework
depends:        mc,videodev
retpoline:      Y
intree:         Y
name:           videobuf2_common
vermagic:       5.10.28-Unraid SMP mod_unload 
parm:           debug:int
```

Is this looking ok?

I also checked on a vanilla Unraid installation if the modules are there and the 2nd exists and looks the same but the first one is not there.

crazycat69, 2021-05-01:
look like compressed videobuf2-common.ko.xz from kernel.
check all files in /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/
anyway need delete **/lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-common.ko.xz** and run **depmod -a**

ich777, 2021-05-01:
Does this seem now right:
```
filename:       /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-common.ko
license:        GPL
author:         Pawel Osciak <pawel@osciak.com>, Marek Szyprowski
description:    Media buffer core framework
depends:        mc,videodev
retpoline:      Y
name:           videobuf2_common
vermagic:       5.10.28-Unraid SMP mod_unload
parm:           debug:int
```

Anyways, this is really tough, I think if I compress the Kernel modules I wouldn't have that problem because all modules with that name would be replaced in the first place since the Kernel modules are not compressed within media build.

The DVB Plugin for Unraid always installs all the modules and then runs `depmod -a` but I simply can't detect if a compressed Kernel module is there and remove it.

This is the output from the videobuf2 directory:
```
videobuf2-common.ko         videobuf2-memops.ko
videobuf2-common.ko.xz      videobuf2-memops.ko.xz
videobuf2-dma-contig.ko     videobuf2-v4l2.ko
videobuf2-dma-contig.ko.xz  videobuf2-v4l2.ko.xz
videobuf2-dma-sg.ko         videobuf2-vmalloc.ko
videobuf2-dvb.ko
```

crazycat69, 2021-05-01:
need delete kernel media modules tree before install (see install.sh script) or install as /kernel/extra or  /kernel/dkms

ich777, 2021-05-02:
@crazycat69 thanks for the suggestion but in the case for Unraid it's a little more difficulty...
But I think I found a solution to this, anyways thank you for the speedy responses and your help. :)

@pk1057 if you got time please try the new package/plugin for Unraid 6.9.2 should work now fine without the symbols error. ;)

pk1057, 2021-05-03:
Problem is solved on the unraid side. Seems the compressed kernel modules where the problem.
Thank you @crazycat69 and @ich777 

peter

closed 2021-05-03 by pk1057