tbsdtv/linux_media #243: driver install ubuntu 5.8.0-50-generic

lucentsquid, 2021-04-29:
OS: Ubuntu 20.04, 64-bit
Kernel: 5.8.0-50-generic

Driver install according to https://github.com/tbsdtv/linux_media/wiki worked for me several times before until update from kernel 5.8.0.44 to kernel 5.8.0.45, which also crashed my nvidia drivers until I updated to kernel 5.8.0.50 and nvidia driver update to version 450.119.03, which brought back nvidia, but tbs driver install now doesn't work any more – no chance.

On the other side my TBS 8922 is still working with kernel 5.8.0.44 but nvidia 450 doesn't, so it is no solution for me anymore to simply use an old kernel.

Partial output of dmesg using kernel 5.8.0.50:

...
[   17.026484] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	4a3d911da787b1f7544a70cd85eab6c73be97de0 Fix some system compatibility
[   17.135440] cx88xx: disagrees about version of symbol ir_raw_event_handle
[   17.135441] cx88xx: Unknown symbol ir_raw_event_handle (err -22)
[   17.135486] cx88xx: disagrees about version of symbol rc_keyup
[   17.135487] cx88xx: Unknown symbol rc_keyup (err -22)
[   17.135500] cx88xx: disagrees about version of symbol rc_keydown_notimeout
[   17.135501] cx88xx: Unknown symbol rc_keydown_notimeout (err -22)
[   17.135512] cx88xx: disagrees about version of symbol rc_register_device
[   17.135512] cx88xx: Unknown symbol rc_register_device (err -22)
[   17.135524] cx88xx: disagrees about version of symbol rc_keydown
[   17.135524] cx88xx: Unknown symbol rc_keydown (err -22)
[   17.135542] cx88xx: disagrees about version of symbol rc_free_device
[   17.135542] cx88xx: Unknown symbol rc_free_device (err -22)
[   17.135556] cx88xx: disagrees about version of symbol rc_allocate_device
[   17.135556] cx88xx: Unknown symbol rc_allocate_device (err -22)
[   17.135583] cx88xx: disagrees about version of symbol ir_raw_event_store_with_filter
[   17.135583] cx88xx: Unknown symbol ir_raw_event_store_with_filter (err -22)
[   17.135596] cx88xx: disagrees about version of symbol rc_unregister_device
[   17.135597] cx88xx: Unknown symbol rc_unregister_device (err -22)
...

According to https://www.linuxtv.org/wiki/index.php/TBS_driver_installation I tried out the following:

...
Troubleshooting

If you find module load errors like "module has wrong symbol version" means that there still are old modules from your previous media tree installation (usually duplicated modules in two different places). The brute-force approach is to simply do:

sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/*

Other workarounds may be offered on the aforementioned authors' wiki pages on Github, and on TBS' forum, depending on specific problems reported by users.
...

I did this after first install to kernel 5.8.0.50 without success (besides the fact, that these files reside in /usr/lib/modules and /lib/modules is just a link to that in Ubuntu 20.04).

I also tried to:

1. remove any previously installed modules of tbs driver entering according to proposals in crazycat69's script:

sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/media
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/staging/media
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/misc/altera-stapl
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/linux
sudo rm -r -f /lib/modules/$(uname -r)/extra

and then downloading the whole thing again by entering:

git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build

then (as a last try) I executed:

media_build/build

by:

$ perl build

This is perl 5, version 30, subversion 0 (v5.30.0) built for x86_64-linux-gnu-thread-multi
(with 50 registered patches, see perl -V for more detail)

Copyright 1987-2019, Larry Wall

and:

media_build/install.sh

by: 

$ bash install.sh

GNU bash, Version 5.0.17(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2019 Free Software Foundation, Inc.
Lizenz GPLv3+: GNU GPL Version 3 oder jünger <http://gnu.org/licenses/gpl.html>

This did something as I now can see in dmesg (after beforehand deleting any remnant of any tbs-module, that something (else) is again going completely wrong as I get the whole symbol error garbage again and again.... So if the whole driver is now out of function, i'd like to buy me another still working device of another hardware vendor or so.

Any help to get it to work again is highly appreciated. Thanks a lot in advance, keep shining!

closed 2021-05-01 by lucentsquid

jadsonlourenco, 2021-05-11:
I'm getting the same issue, how to solve? You closed but don't share the fix. TY!

lucentsquid, 2021-05-13:
I closed the issue because I was too frustrated.

Meanwhile I think, that this is some kind of complex multiple Linux Kernel 5.8.0, Nvidia plus TBS incompatiblity.

First of all: None of any repairings of Kernel 5.8.0.5x worked for me. I stopped testing any combinations of 5.8.0.5x kernel, Nvida whatsoever and TBS drivers including completely fresh Ubuntu 20.04 installs because none of them worked.

But I was lucky: Some time ago I saved the whole harddrive with free of charge Clonezilla (very recommendable H/SDD cloning tool) some months ago. At that time I used Kernel 5.4.0-48-generic with Nvidia drivers (390 or so) AND TBS-drivers installed as discribed in TBS-wiki, which worked absolutely fine for me.

What I did after anything else failed after some days of painfull reinstalling and repairing:

1. doing some backup of user-data
2. restoring Clonezilla-backup (using Clonezilla bootable image from USB), which took some time
3. apt-updating any software but avoiding (!) any kernel update (!) (as it always killed either Nvidia or TBS drivers or both before)

2.1 I'm on multiboot but TBS-driver for Microbrain W10 never worked for me anyway.

3.1. Huge update, of course. Updating Nvidia to 460.73.01 didn't kill TBS again so I think, kernel ... can't cope with combination of Nvidia and/or TBS drivers (or maybe drivers for other tv-devices as well).

This is not a so called 'solution' of the formerly described problem but a strategy, which somehow works at the moment. Now I can use TBS + VLC commandline for recording, ffmpeg + nvenc for cutting and encoding promo-contaminated material again and it is also of advantage to get something else than 1024*768 mole-resolution on my cracked video kernel.

Avoiding kernel updates is probably a bad idea in general, so I will backup the whole thing again with Clonezilla and try out newer kernels as long as this HDD (or any other brave storage medium or me) has reached end of lifetime.

If any newer kernel works, I'll tell you next year in another thread. If no kernel or respectively driver works, I'll keep that one, until my TBS 8922 or the mainboard was charcoaled.

jadsonlourenco, 2021-05-13:
Cool! I've solved my issue replacing the signal splitter, was not driver or kernel, but yes I did get this issue with another kernel version. 

Anyway thank you for your share, and I will do the same, I'm using Docker because I like that, you create an "image" of your system/app and it will run the same on every computer that run this "container" image, so, we can have versions os a system/service that works,

lucentsquid, 2021-05-13:
It is allways a good idea to use anything else than those Dockers. If something goes wrong we rather use Clonezilla to keep our brains but if so called Docker makes fun, why not? The principal thing is to keep it going.

lucentsquid, 2021-05-13:
My signal splitter obviously works fine because I get absolutely clean results with kernel 5.4.0-48-generic but 5.5 is true horror nightmare.

lucentsquid, 2021-05-13:
I closed this thread once before, because I didn't got any answer to any question after a few days but linux driver support for TBS-hardware is not as bad as insect support in abandoned  rain forests.