tbsdtv/linux_media #246: MyGica T230A doesn't work with DVB-C

andrasbiro, 2021-04-30:
I managed to build the modules on a slightly modified version of this package:
https://aur.archlinux.org/packages/t230c2-git

And it does work using DVB-T (can't test DVB-T2 or ATSC). 

However, when I plug in my cable, and run [w_scan2](https://github.com/stefantalpalaru/w_scan2) -fc -c HU, I have the following on many channels:
```  
    signal ok:    QAM_AUTO f = 474000 kHz S6875C999  (0:0:0)
    Info: no data from PAT after 2 seconds
    deleting (QAM_AUTO f = 474000 kHz S6875C999  (0:0:0))
```

There's a small chance that the issue is caused by long usb cables - It's not easy to wire cable to my desktop. However, I used the same cable length when tested with DVB-T (DVB-C is QAM256, 6900 srate here)

Kernel is 5.12.0-1-MANJARO. Firmwares:
```
[ 4143.756942] si2157 10-0060: found a 'Silicon Labs Si2141-A10'
[ 4143.756973] si2157 10-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[ 4143.893145] si2157 10-0060: firmware version: 1.1.10
[ 4148.076024] si2157 10-0060: found a 'Silicon Labs Si2141-A10'
[ 4148.076059] si2157 10-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
[ 4148.204304] si2157 10-0060: firmware version: 1.1.10
```

closed 2021-04-30 by crazycat69

andrasbiro, 2021-05-03:
What did I do wrong? If I missed some vital information from the issue, I'm happy to add it.

andimik, 2021-07-22:
TBS is not the manufacturer of your T230A, so @crazycat69 asked you to address your request to MyGica.