tbsdtv/linux_media #247: TBS5580 CI issues

Kitoko86, 2021-05-07:
Hi,

I noticed that the TBS5580 has a strange behavior on my Rasperry Pi 3 with latest drivers.  After a certain amount of time, TVHeadend loses the connection with the tuner or minisatip (several versions tested) crashes.

However, my TBS5520SE owning the same kind of hardware works without an issue.

Kitoko86, 2021-05-17:
Seems the TBS support doesn't care about their buggy drivers... What a pity for a manufacturer like this because the hardware's worth it. If it was a serious company, their products should work out of the box with a minimum of configuration for the end user. Sorry for the complaining but I find that TBS doesn't send a good signal to their customers.

zhang111chao, 2021-05-18:
please write issue to support@tbsdtv.com

Kitoko86, 2021-05-24:
Already made it. I have a conversation with jack from the support team but no answer from him since a week.

Kitoko86, 2021-06-04:
No, same behavior with Ubuntu 18.04.

crazycat69, 2021-06-04:
This common issues for dvb-usb drivers and some xhci USB hosts (usb 3.x)