tbsdtv/linux_media #249: TBS6508 DVB-C si2183 6-0067: read_status fe0 cmd_exec failed=-5

EgidijusL, 2021-05-13:
Use TBS6508 with DVB-C on Cesbo Astra. Now all _msi_ is off (grub, modprobe.d) and same problem.
Random time, random tunners show "_FE_READ_STATUS failed [Input/output error]_". Only restarting the server helps.
[image]
dmesg:
```
[1828178.234719] TBSECP3 driver 0000:06:00.0: i2c nack (7)
[1828178.235680] si2183 6-0064: read_status fe0 cmd_exec failed=-5
[1828178.236183] si2183 6-0064: read_status failed=-5
[1828178.342720] TBSECP3 driver 0000:06:00.0: i2c nack (7)
[1828178.343415] si2183 6-0067: read_status fe0 cmd_exec failed=-5
[1828178.343898] si2183 6-0067: read_status failed=-5
[1828178.438714] TBSECP3 driver 0000:06:00.0: i2c nack (7)
[1828178.439272] si2183 6-0064: read_status fe0 cmd_exec failed=-5
[1828178.439833] si2183 6-0064: read_status failed=-5
[1828178.449419] TBSECP3 driver 0000:06:00.0: i2c nack (7)
[1828178.450195] si2183 6-0067: read_status fe0 cmd_exec failed=-5
[1828178.450702] si2183 6-0067: read_status failed=-5
[1828178.538307] TBSECP3 driver 0000:06:00.0: i2c nack (7)
[1828178.538940] si2183 6-0064: read_status fe0 cmd_exec failed=-5
[1828178.539420] si2183 6-0064: read_status failed=-5
```
Server info:
```
Supermicro  H11SSL-i
AMD EPYC 7251
Debian 10 4.19.0-16-amd64
```

Davin622, 2021-05-14:
Hi
thanks for your feedback, we will do the same test.
BTW. how many cards in your system? 

Best Regards

EgidijusL, 2021-06-01:
Hi. I installed Ubuntu on the server on April 20, 2002 and for a week now there have been no problems with TBS. 
Kernel version: 5.4.0-73-generic

EgidijusL, 2021-11-15:
Ubuntu, 5.4.0-89-generic
The problem is still ..
TBSECP3 driver 0000:06:00.0: i2c nack (7)
si2183 6-0067: read_status fe0 cmd_exec failed=-5
si2183 6-0067: read_status failed=-5

Neolo, 2022-12-28:
Same problem, system froze with i2c nack (FFFFFFF) errors at TBSECP3
ubuntu 20 LTS

wwilde, 2024-04-15:
Solution?

Znuff, 2024-04-18:
We have the same issue, then after a reboot, seems *half* of the tuner died completely:

```
[  +0.000002] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[  +0.122672] TBSECP3 driver 0000:10:00.0: MAC address 00:22:ab:b2:10:d7
[  +0.000644] si2183 6-0067: Silicon Labs Si2183 successfully attached
[  +0.000578] TBSECP3 driver 0000:10:00.0: i2c nack (7)
[  +0.000051] TBSECP3 driver 0000:10:00.0: i2c nack (7)
[  +0.000004] si2157: probe of 17-0060 failed with error -5
[  +0.000002] TBSECP3 driver 0000:10:00.0: TBSECP3 frontend 7 attach failed
[  +0.000002] TBSECP3 driver 0000:10:00.0: frontend attach failed
[  +0.000084] TBSECP3 driver 0000:10:00.0: adapter7 attach failed
[  +0.000003] TBSECP3 driver 0000:10:00.0: Turbosight TBS 6209 (Octa DVB-T/T2/C/C2/ISDB-T): PCI 0000:10:00.0, IRQ 83, MMIO 0xfbe00000
```

Adapters 0-3 on the tuner work, adapters 4-7 are dead and will fail registering.