tbsdtv/linux_media #250: Driver installation breaks OS graphics and sound

citkane, 2021-05-16:
Hardware:
AMD Ryzen 5 3400g (this is the one with graphics integrated in the CPU)
Realtek Audio
TBS5520SE

Software:
Ubuntu 20.04 (Clean install with all updates)

From a clean start, all graphics and sound works perfectly. Below is the output of `xrandr`
```
Screen` 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
HDMI-A-0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 1218mm x 685mm
   1920x1080     60.00*+  50.00    59.94    30.00    24.00    29.97    23.98  
   4096x2160     60.00    50.00    59.94    24.00    23.98  
   3840x2160     60.00    50.00    59.94    30.00    25.00    24.00    29.97    23.98  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     60.02  
   1440x900      60.00  
   1280x800      60.00  
   1152x864      75.00  
   1280x720      60.00    50.00    30.00    59.94    29.97    24.00    23.98  
   1024x768      60.00  
   800x600       60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       60.00    59.94  
DVI-D-0 disconnected (normal left inverted right x axis y axis)
HDMI-A-1 disconnected (normal left inverted right x axis y axis)
HDMI-A-2 disconnected (normal left inverted right x axis y axis)
```
After installation of the drivers from this repo, no error messages, the TBS unit works as expected.
However, the graphics is severely limited to the `xrandr` output below:
```
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 1920 x 1080, current 1920 x 1080, maximum 1920 x 1080
default connected primary 1920x1080+0+0 0mm x 0mm
   1920x1080     77.00*
```
Sound output is pysically non existent, and the OS only sees a "Dummy Device" from the GUI sound settings.

Booting from grub into the previous kernel resolves all the graphics ans sound issues, but then obviously the TBS unit does not work.

Please advise on correct procedure to get the TBS5520SE installed for this scenario.

bastard79, 2021-05-16:
yea, look here, and try kernel the lower kernel of 5.8.0, its working quite good on 5.8.0-25, and dont work on 5.8.0.53.
look here too https://github.com/tbsdtv/linux_media/issues/240 - now im test patch crazycat

citkane, 2021-05-17:
> yea, look here #240, and try the lower kernel of 5.8.0, its working quite good on 5.8.0-25, and dont work on 5.8.0.53.

Ok, that may be a workaround.

The problem is that I am a Tv watching, product buying kind of Linux guy, and not a kernel modding, away from LTS standard kind of Linux superuser... 
The latter is too time consuming for my enthusiasm level.

How is this REPO tied in with TBS, the company as distinct from an open source community?

How soon is this issue likely to be fixed at a consumer level?
Should I return my product for a refund?
Are there alternatives that work on Ubuntu LTS?

crazycat69, 2021-05-17:
Better downgrade to GA Ubuntu 20.04 kernel (5.4) - https://wiki.ubuntu.com/Kernel/LTSEnablementStack#Ubuntu_20.04_LTS_-_Focal_Fossa

LTS kernel 4.4 - 5.4 suported without this issues.

citkane, 2021-05-22:
I can confirm that downgrading to GA Ubuntu 20.04 kernel resolves the issue.

Thanks crazycat69

ben-abolins, 2021-05-23:
Hello,

Thank you for making this driver available. It has worked really well for me on many versions of Ubunbu

I am having issues compiling and using the TBS drivers here with Ubuntu 21.04 / Kernel 5.11. Depending on the order I do things, I can make two of the following work, but not all three:
- TBS Driver for tuner card
- Nvidia/Intel Graphics
- Infrared remote receiver - mceusb module.

Does anyone know when the TBS drivers will be updated to support kernel 5.11, as used in the latest Ubuntu 21.04.

Thanks,
Ben

crazycat69, 2021-05-23:
This general issue related to main-stream [media_build](https://git.linuxtv.org/media_build.git)
Some kernel modules outside media tree depend from some media core modules (cec, rc-core, mc). So after rebuild media tree this modules not load (drm modules for GPU drivers, some usb sound adapters).
Better use 18.04, 20.04 LTS with 5.4 kernel.

Borim7, 2025-09-06:
I had a similar issue with intel graphic drivers. I avoid the problem by skipping building the remote control drivers as done by  https://github.com/tbsdtv/media_build/blob/latest/install.sh

The important part is:
````
# Disable RC/IR support
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
````

I think this is a better solution, than sticking to an older kernel version.