tbsdtv/linux_media #251: Compile Error on Manjaro Linux with Kernel 5.12.1

thxwein, 2021-05-16:
Hi, 
I'm getting a this error message on Manjaro.

/home/zoing/temp/media_build-master/v4l/cafe-driver.c:47:25: error: expected declaration specifiers or '...' before string constant
   47 | MODULE_SUPPORTED_DEVICE("Video");
      |                         ^~~~~~~
make[3]: *** [scripts/Makefile.build:271: /home/zoing/temp/media_build-master/v4l/cafe-driver.o] Fehler 1
make[3]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[2]: *** [Makefile:1851: /home/zoing/temp/media_build-master/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.12.2-1-MANJARO/build'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/home/zoing/temp/media_build-master/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2
build failed at ./build line 554

Thanks for your help!

zhang111chao, 2021-05-18:
please add skype jack_2431.

zhang111chao, 2021-05-18:
please write email to  jack@tbsdtv.com or   add skype jack_2431 .

caveman99, 2021-05-19:
add VIDEO_OV7670 and VIDEO_CAFE_CCIC to the 9.255.255 section of media_build/v4l/versions.txt file before make

LinLo, 2021-05-19:
MODULE_SUPPORTED_DEVICE has been removed after the 5.12rc2 (last merge here I think):
https://lkml.org/lkml/2021/3/17/518
https://github.com/torvalds/linux/commit/6417f03132a6952cd17ddd8eaddbac92b61b17e0

caveman99, 2021-05-19:
the macro was never actually used for anything, so this can be removed safely for _every_ kernel version.

thxwein, 2021-05-19:
thanks caveman99,
your command line did the trick. I got it compiled.

closed 2021-05-20 by thxwein