tbsdtv/linux_media #253: build failure ubuntu

iotic-labs-markwharton, 2021-05-29:
```sudo ./install.sh 
[sudo] password for mark: 
make -C /home/mark/temp/tbs/media_build/v4l stagingconfig
make[1]: Entering directory '/home/mark/temp/tbs/media_build/v4l'
make[2]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[3]: Entering directory '/home/mark/temp/tbs/media_build/linux'
Unapplying patches
make[3]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Applying patches for kernel 5.4.0-73-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
make[1]: *** [Makefile:380: stagingconfig] Error 2
make[1]: Leaving directory '/home/mark/temp/tbs/media_build/v4l'
make: *** [Makefile:26: stagingconfig] Error 2
V4L drivers building...
make -C /home/mark/temp/tbs/media_build/v4l 
make[1]: Entering directory '/home/mark/temp/tbs/media_build/v4l'
make -C firmware prep
creating symbolic links...
make[2]: Entering directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/v4l/firmware'
Kernel build directory is /lib/modules/5.4.0-73-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[3]: Entering directory '/home/mark/temp/tbs/media_build/linux'
Unapplying patches
make[3]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Applying patches for kernel 5.4.0-73-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
make[1]: *** [Makefile:52: default] Error 2
make[1]: Leaving directory '/home/mark/temp/tbs/media_build/v4l'
make: *** [Makefile:26: all] Error 2
V4L drivers installing...
make -C /home/mark/temp/tbs/media_build/v4l install
make[1]: Entering directory '/home/mark/temp/tbs/media_build/v4l'
Installing kernel modules under /lib/modules/5.4.0-73-generic/kernel/drivers/media/:
/sbin/depmod -a 5.4.0-73-generic 
make -C firmware install
make[2]: Entering directory '/home/mark/temp/tbs/media_build/v4l/firmware'
Installing firmwares at /lib/firmware: 
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make[1]: Leaving directory '/home/mark/temp/tbs/media_build/v4l'
V4L drivers installation done
You need to reboot...
```

crazycat69, 2021-05-29:
Updated

iotic-labs-markwharton, 2021-05-29:
pulled, but still failing
```mark@windy:~/temp/tbs/media_build$ sudo ./install.sh 
make -C /home/mark/temp/tbs/media_build/v4l stagingconfig
make[1]: Entering directory '/home/mark/temp/tbs/media_build/v4l'
No version yet, using 5.4.0-73-generic
make[2]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[3]: Entering directory '/home/mark/temp/tbs/media_build/linux'
Unapplying patches
make[3]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Applying patches for kernel 5.4.0-73-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
make[1]: *** [Makefile:380: stagingconfig] Error 2
make[1]: Leaving directory '/home/mark/temp/tbs/media_build/v4l'
make: *** [Makefile:26: stagingconfig] Error 2
V4L drivers building...
make -C /home/mark/temp/tbs/media_build/v4l 
make[1]: Entering directory '/home/mark/temp/tbs/media_build/v4l'
scripts/make_makefile.pl
Updating/Creating .config
make[2]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[2]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[3]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[3]: Entering directory '/home/mark/temp/tbs/media_build/linux'
Unapplying patches
Unapplying patches
make[3]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Applying patches for kernel 5.4.0-73-generic
make[3]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Applying patches for kernel 5.4.0-73-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/api_version.patch
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Preparing to compile for kernel version 5.4.0

***WARNING:*** You do not have the full kernel sources installed.
This does not prevent you from building the v4l-dvb tree if you have the
kernel headers, but the full kernel source may be required in order to use
make menuconfig / xconfig / qconfig.

If you are experiencing problems building the v4l-dvb tree, please try
building against a vanilla kernel before reporting a bug.

Vanilla kernels are available at http://kernel.org.
On most distros, this will compile a newly downloaded kernel:

cp /boot/config-`uname -r` <your kernel dir>/.config
cd <your kernel dir>
make all modules_install install

Please see your distro's web site for instructions to build a new kernel.

WARNING: This is the V4L/DVB backport tree, with experimental drivers
	 backported to run on legacy kernels from the development tree at:
		http://git.linuxtv.org/media-tree.git.
	 It is generally safe to use it for testing a new driver or
	 feature, but its usage on production environments is risky.
	 Don't use it in production. You've been warned.
CEC_CROS_EC: Requires at least kernel 9.255.255
V4L2_H264: Requires at least kernel 9.255.255
VIDEO_IPU3_CIO2: Requires at least kernel 9.255.255
VIDEO_OMAP3: Requires at least kernel 9.255.255
VIDEO_IMX258: Requires at least kernel 9.255.255
VIDEO_IMX274: Requires at least kernel 9.255.255
SND_BT87X: Requires at least kernel 9.255.255
INTEL_ATOMISP: Requires at least kernel 9.255.255
VIDEO_HANTRO: Requires at least kernel 9.255.255
VIDEO_ROCKCHIP_VDEC: Requires at least kernel 9.255.255
VIDEO_IPU3_IMGU: Requires at least kernel 9.255.255
Created default (all yes) .config file
./scripts/make_myconfig.pl
[ ! -f "./config-mycompat.h" ] && echo "/* empty config-mycompat.h */" > "./config-mycompat.h" || true
make -C firmware prep
creating symbolic links...
perl scripts/make_config_compat.pl /lib/modules/5.4.0-73-generic/build ./.myconfig ./config-compat.h
make[2]: Entering directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/v4l/firmware'
Kernel build directory is /lib/modules/5.4.0-73-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/mark/temp/tbs/media_build/linux'
make[3]: Entering directory '/home/mark/temp/tbs/media_build/linux'
Unapplying patches
make[3]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
Applying patches for kernel 5.4.0-73-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/linux'
make[1]: *** [Makefile:52: default] Error 2
make[1]: Leaving directory '/home/mark/temp/tbs/media_build/v4l'
make: *** [Makefile:26: all] Error 2
V4L drivers installing...
make -C /home/mark/temp/tbs/media_build/v4l install
make[1]: Entering directory '/home/mark/temp/tbs/media_build/v4l'
Installing kernel modules under /lib/modules/5.4.0-73-generic/kernel/drivers/media/:
/sbin/depmod -a 5.4.0-73-generic 
make -C firmware install
make[2]: Entering directory '/home/mark/temp/tbs/media_build/v4l/firmware'
Installing firmwares at /lib/firmware: 
make[2]: Leaving directory '/home/mark/temp/tbs/media_build/v4l/firmware'
make[1]: Leaving directory '/home/mark/temp/tbs/media_build/v4l'
V4L drivers installation done
You need to reboot...
mark@windy:~/temp/tbs/m```

crazycat69, 2021-05-30:
Look like you not have cloned&synced linux-media repo and just run **./install.sh** script ?
https://github.com/tbsdtv/linux_media/wiki

iotic-labs-markwharton, 2021-05-30:
Thanks that worked. Apologies for noobishness. I normally pull both directories, run `make distclean` and then run `install.sh`. Yesterday, one of the pulls failed and I got into a git mess.

closed 2021-05-30 by iotic-labs-markwharton