tbsdtv/linux_media #254: No rule to make target '/media_build/v4l/stid135.o' on 5.10.40

ich777, 2021-05-31:
Hi,

I got a compilation error on Kernel version 5.10.40:

```
  CC [M]  /usr/src/TBS_OS/media_build/v4l/sp8870.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/sp887x.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/sr030pc30.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/st-mipid02.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/stb0899.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/stb6000.mod.o
make[4]: *** No rule to make target '/usr/src/TBS_OS/media_build/v4l/stid135.o', needed by '/usr/src/TBS_OS/media_build/v4l/stid135.ko'.  Stop.
make[4]: *** Waiting for unfinished jobs....
```

am I missing something?

caveman99, 2021-05-31:
did you update the tree? try strting over with a fresh checkout first. if that doesnt work, add #include <linux/slab.h> to globaldefs.h:

```
--- media/drivers/media/dvb-frontends/stid135/globaldefs.h.orig     2021-05-31 10:44:32.000000000 +0200
+++ media/drivers/media/dvb-frontends/stid135/globaldefs.h      2021-05-31 10:53:47.067416731 +0200
@@ -45,6 +45,7 @@
        #ifdef HOST_PC
                #include <linux/kernel.h>
                #include <linux/delay.h>
+               #include <linux/slab.h>

                #define WAIT_N_MS(X) msleep(X)
                #define calloc(n,s) kzalloc(n*s,GFP_KERNEL)
```

ich777, 2021-05-31:
Yes, I always download a fresh copy from Github for my compilation.
Anyways, tried it again but still the same.

If I add the patch you submitted it works just fine, thank you.
Can this patch be applied to the master branch?

caveman99, 2021-05-31:
Thing is, i was compiling for 5.12.8 and ran into another issue with the same driver (unknown function declarations which i traced down to linux/slab.h). It was a chance fix for me, so i thought it might help you too. Apart from that i don't know _why_ this breaks with the latest media tree and which kernels require this fix or if this is missing for all kernel series. 5.12.6 compiled for me without the fix, but that was before backporting the media tree mid-may.

Maybe @crazycat69 has a better insight into the global scope.

ich777, 2021-05-31:
@caveman99 Thank you very much for the help! Really much appreciated.
I pre-compile the drivers for a plugin that I created for Unraid that installs all necessary modules and I can tell for sure that it worked on Kernel v5.10.39, with the new test release of Unraid that is on Kernel v5.10.40 it doesn't work without that patch.

I let the issue open eventually @crazycat69 can look into this when he got time.

Viele Grüße und Danke!

ich777, 2021-05-31:
New test version from Unraid released with Kernel v5.10.41 and as expected same behavior:

```
  CC [M]  /usr/src/TBS_OS/media_build/v4l/saa7185.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/saa7706h.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/seco-cec.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/serial_ir.mod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/shark2.mod.o
make[4]: *** No rule to make target '/usr/src/TBS_OS/media_build/v4l/stid135.o', needed by '/usr/src/TBS_OS/media_build/v4l/stid135.ko'.  Stop.
make[4]: *** Waiting for unfinished jobs....
```

If I apply that patch everything is compiling just fine.

crazycat69, 2021-06-03:
9e56a751fbb49d36182a8d53fd42730165d607b7

ich777, 2021-06-03:
Thank you @crazycat69! :)

closed 2021-06-03 by ich777