tbsdtv/linux_media #255: 2x TBS 6902 on 2nd card only 1 tuner seems to work

SpeedProg, 2021-06-12:
OS: Gentoo Linux with 5.10.27 kernel
Driver used: linux_media with allyesconfig (except CONFIG_VIDEO_OV9650=n and CONFIG_VIDEO_OV772X=n because they  use __devm_regmap_init_sccb)
Also used the archlinux out of tree one before and behaviour is the same (patched it with this https://github.com/tbsdtv/linux_media/commit/942af455a1fca71dc15a248dea72a94aae6dd030 ) but then switched to this official one to see if the problem persists and it did.

I decided to add an other TBS 6902 to my TV server since the first one had worked quite well.
Both TBS 6902 are connected via Unicable I to the LNB.
There is an other DVB-S2 usb card that uses a legacy connection.

What I noticed is that on the newly added TBS 6902 only one adapter did work in tvheadend.
I tried to exchange the cables connected to see if that is the problem, but even then the same adapter stayed none functional while the others worked.
I changed the PCIe slots the cards are in and then the one of the adapters of the other (older) card was not working. Turns out it is always one of the adapters of the card in the lower PCIe slot that is not working.

Cards are of different hardware revision it seems, one is 6902:0002 and one 6902:0003. Tried to get a tbsdtv forum account but even week after I sent the mail to support it is still not active :(.

The none working adapter also gives no found transponders when scanning on it with scan_w.

crazycat69, 2021-06-13:
Strange message when loading gx1133 demod driver for 6902:0003
**gx1133: loading out-of-tree module taints kernel.**

SpeedProg, 2021-06-13:
Is it not normal that the kernel marks tainted when these externally compiled modules get loaded?

crazycat69, 2021-06-13:
check loaded modules list with **lsmod**

SpeedProg, 2021-06-13:
lsmod just shows that it is loaded
Also idk if that is of any consquence or if it was already understood, but the "only 1 tuner out of the 2 per card working" happens with both cards (only one at once through) depending on in which order they are in the PCIe slots.

```
Module                  Size  Used by
8021q                  28672  0
algif_skcipher         16384  0
dm_zero                16384  0
dm_round_robin         16384  0
dm_multipath           28672  1 dm_round_robin
virtio_pci             24576  0
virtio_scsi            20480  0
virtio_blk             20480  0
virtio_console         28672  0
virtio_balloon         20480  0
xts                    16384  1
ecb                    16384  1
sha1_generic           16384  0
iscsi_tcp              20480  0
libiscsi_tcp           24576  1 iscsi_tcp
libiscsi               45056  2 libiscsi_tcp,iscsi_tcp
cxgb3                 143360  0
mdio                   16384  1 cxgb3
cxgb4                 393216  0
bonding               114688  0
virtio_net             40960  0
virtio_ring            20480  6 virtio_console,virtio_balloon,virtio_scsi,virtio_pci,virtio_blk,virtio_net
virtio                 16384  6 virtio_console,virtio_balloon,virtio_scsi,virtio_pci,virtio_blk,virtio_net
e1000                 106496  0
bnx2                   77824  0
fuse                  106496  1
jfs                   159744  0
multipath              16384  0
linear                 16384  0
raid10                 49152  0
raid1                  40960  0
raid0                  20480  0
dm_raid                40960  0
raid456               122880  2 dm_raid
async_raid6_recov      16384  1 raid456
async_memcpy           16384  1 raid456
async_pq               16384  1 raid456
async_xor              16384  2 async_pq,raid456
async_tx               16384  5 async_pq,async_memcpy,async_xor,raid456,async_raid6_recov
dm_snapshot            45056  0
dm_bufio               28672  1 dm_snapshot
dm_mirror              20480  0
dm_region_hash         16384  1 dm_mirror
dm_log                 16384  2 dm_region_hash,dm_mirror
hid_sunplus            16384  0
hid_sony               28672  0
hid_samsung            16384  0
hid_pl                 16384  0
hid_petalynx           16384  0
hid_logitech_dj        24576  0
hid_logitech_hidpp     36864  0
hid_gyration           16384  0
sl811_hcd              20480  0
ohci_pci               16384  0
ohci_hcd               36864  1 ohci_pci
uhci_hcd               32768  0
usb_storage            57344  0
mpt3sas               233472  0
aic94xx                73728  0
libsas                 69632  1 aic94xx
qla2xxx               724992  0
megaraid_sas          126976  0
megaraid_mbox          32768  0
megaraid_mm            20480  1 megaraid_mbox
megaraid               36864  0
[... 174 lines cut ...]
snd_hda_codec          94208  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
i2c_i801               24576  0
i2c_smbus              16384  1 i2c_i801
snd_hda_core           65536  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
tbsecp3                40960  40
snd_pcm                86016  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
tas2101                16384  3 tbsecp3
gx1133                 20480  3 tbsecp3
snd_timer              32768  1 snd_pcm
xhci_pci               16384  0
dvb_core               86016  3 tbsecp3,dvb_usb,dvb_usb_dw2102
ehci_pci               16384  0
snd                    69632  6 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_timer,snd_pcm
mc                     36864  2 dvb_usb,dvb_core
e1000e                180224  0
soundcore              16384  1 snd
xhci_hcd              200704  1 xhci_pci
ehci_hcd               49152  1 ehci_pci
fan                    16384  0
thermal                20480  0
wmi                    24576  2 wmi_bmof,mxm_wmi
mlx4_ib               155648  0
ib_uverbs             114688  1 mlx4_ib
ib_core               278528  2 mlx4_ib,ib_uverbs
```