tbsdtv/linux_media #256: help on installing drivers for tbs5980 on raspberry pi

keiron26, 2021-06-30:
Hi can anyone give me a list of step by step quide on building and installing drivers for tbs5980

i have followed the 
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
make -j4
sudo make install
sudo reboot

also installed raspberrypi-kernel-headers, install patchutils and libproc-processtable-perl

no errors came up i could tell but it is still not showing up in tvheadend adapters

crazycat69, 2021-06-30:
Remove kernel media modules tree before install, see [install.sh script](https://github.com/tbsdtv/media_build/blob/master/install.sh)

keiron26, 2021-06-30:
> Remove kernel media modules tree before install, see [install.sh script](https://github.com/tbsdtv/media_build/blob/master/install.sh)

can you explain how i do that i am abit of a noob towards linux

menzbua, 2021-07-06:
Execute theses commands:
```bash
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/media
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/staging/media
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/misc/altera-stapl
sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/linux
sudo rm -r -f /lib/modules/$(uname -r)/extra
```
And then recompile the drivers. This is how i understand @crazycat69 suggestion.