tbsdtv/linux_media #257: TBS5990 After a few minutes stream cancels with error poll TIMEOUT

menzbua, 2021-07-04:
Hi,

i have the following setup:
Ubuntu 20.04 tvheadend VM on my VMWare Workstation Host
TBS5990 DVB-S Dual Tuner USB
Kernel 5.4.0-77
https://github.com/tbsdtv/linux_media drivers compiled on friday 02.07.2021
Tvheadend 4.3-1967~geb59284b8

USB Tuner is connected to the VM with an USB2.0 Hostadapter.

The Problem:
Everything is working fine. After a few minutes, mostly between 10 and 20 minutes, the stream cancels. In the syslog messages i see the following error:
```
Jul  4 08:45:44 tvheadend tvheadend[801]: linuxdvb: TurboSight TBS 5990 #1 : DVB-S #0 - poll TIMEOUT
Jul  4 08:45:45 tvheadend tvheadend[801]: linuxdvb: TurboSight TBS 5990 #1 : DVB-S #0 - retune nodata
```
The issue happens on both Tuners. This is not specific to one of the two tuners.
I think i can exclude a hardware issue because when i use my windows machine with dvbviewer everything is working.

I've played around with the tvheadend "Input buffer (bytes)", "Status period (ms)" and "Force old status" in the tuner settings. Because the default settings are not working i've increased the Input buffer to 1880000 and the Status period to 5000 and checked the Force old status box. But the problem still exists.

These kernel modules are enabled:
```
Module                  Size  Used by
md4                    16384  0
cmac                   16384  1
nls_utf8               16384  2
cifs                 1024000  2
libarc4                16384  1 cifs
fscache               372736  1 cifs
libdes                 24576  1 cifs
vmw_vsock_vmci_transport    32768  1
vsock                  36864  2 vmw_vsock_vmci_transport
dm_multipath           32768  0
scsi_dh_rdac           16384  0
scsi_dh_emc            16384  0
scsi_dh_alua           20480  0
av201x                 20480  2
cx231xx_dvb_ci         32768  2
tas2101                28672  3 cx231xx_dvb_ci
dvb_core              151552  1 cx231xx_dvb_ci
cx25840                61440  1
cx231xx               192512  1 cx231xx_dvb_ci
tveeprom               28672  1 cx231xx
cx2341x                28672  1 cx231xx
i2c_mux                16384  1 cx231xx
videobuf2_vmalloc      20480  2 dvb_core,cx231xx
videobuf2_memops       20480  1 videobuf2_vmalloc
videobuf2_v4l2         28672  1 cx231xx
videobuf2_common       61440  5 videobuf2_vmalloc,videobuf2_v4l2,dvb_core,cx231xx,videobuf2_memops
videodev              229376  5 cx2341x,videobuf2_v4l2,videobuf2_common,cx231xx,cx25840
mc                     53248  6 videodev,videobuf2_v4l2,dvb_core,videobuf2_common,cx231xx,cx25840
intel_rapl_msr         20480  0
vmw_balloon            24576  0
intel_rapl_common      24576  1 intel_rapl_msr
rapl                   20480  0
joydev                 24576  0
input_leds             16384  0
serio_raw              20480  0
ir_rc6_decoder         20480  0
rc_rc6_mce             16384  0
ftdi_sio               61440  1
usbserial              53248  3 ftdi_sio
mceusb                 40960  0
rc_core                57344  5 mceusb,ir_rc6_decoder,rc_rc6_mce,cx231xx
vmw_vmci               69632  2 vmw_balloon,vmw_vsock_vmci_transport
mac_hid                16384  0
sch_fq_codel           20480  5
ip_tables              32768  0
x_tables               40960  1 ip_tables
autofs4                45056  2
btrfs                1261568  0
zstd_compress         167936  1 btrfs
raid10                 61440  0
raid456               155648  0
async_raid6_recov      24576  1 raid456
async_memcpy           20480  2 raid456,async_raid6_recov
async_pq               24576  2 raid456,async_raid6_recov
async_xor              20480  3 async_pq,raid456,async_raid6_recov
async_tx               20480  5 async_pq,async_memcpy,async_xor,raid456,async_raid6_recov
xor                    24576  2 async_xor,btrfs
raid6_pq              114688  4 async_pq,btrfs,raid456,async_raid6_recov
libcrc32c              16384  2 btrfs,raid456
raid1                  45056  0
raid0                  24576  0
multipath              20480  0
linear                 20480  0
hid_generic            16384  0
usbhid                 57344  0
hid                   131072  2 usbhid,hid_generic
crct10dif_pclmul       16384  1
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
aesni_intel           372736  1
crypto_simd            16384  1 aesni_intel
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
glue_helper            16384  1 aesni_intel
psmouse               155648  0
vmwgfx                299008  1
ttm                   106496  1 vmwgfx
drm_kms_helper        184320  1 vmwgfx
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
drm                   491520  4 vmwgfx,drm_kms_helper,ttm
vmxnet3                57344  0
mptspi                 24576  3
mptscsih               40960  1 mptspi
mptbase                94208  2 mptspi,mptscsih
scsi_transport_spi     32768  1 mptspi
i2c_piix4              28672  0
pata_acpi              16384  0
```

When the timeout message appears in dmesg i did not see any error. The only thing i see is that:
```
[   35.884489] cx231xx 1-1:1.1: Unknown tuner type configuring SIF
[   54.539077] cx231xx 1-1:1.1: Unknown tuner type configuring SIF
```

Regards
Manuel

tydaikho, 2021-12-25:
> Hi @Neolo,
> 
> thanks for your suggestions. I don't think this is a hardware problem because on windows the card is running without any problems.
> 
> Regards Manuel

Yes, i agree with you, i saw same problem. My Astrometa DVB USB works well with standard linux kernel. my Mygica EyeTV works well too. After i installed TBS driver for my Qbox, Astrometa gets "POLL timeout" error, my Mygica (Silicon Lab si2141) get "Firmware download failed" error. So my idea, it is not hardware, that is driver matter. Is there anyway to install TBS driver only, not whole linux media ? Thank you.

crazycat69, 2021-12-25:
@@Neolo - what kernel version ?

tydaikho, 2021-12-26:
> @tydaikho - what kernel version ?

5.10, 5.14 ... I get two sticks: Astrometa DVB (CDX2837ER) and Mygica EyeTV (dvb-tuner-si2141-a10-01.fw) . both of them work well and native supported with standard kernel 5.x version. i must install TBS driver for my Qbox2 but it makes Astrometa and Mygical problems. In this case, the problems may be from tbs or linux_media, only guess that. So i  find the way to install TBS driver only without whole linux media driver to be sure.

menzbua, 2022-08-24:
I solved the issue for my case. I'm running TVHeadend with the latest driver versions from the git repo on Ubuntu 20.04 with kernel 5.4.0-124. I've changed the tuner configuration in TVHeadend. There is a config `Status Period (ms)` and i set it to **5000**. Since i've done that everything is working fine.

closed 2022-08-24 by menzbua