tbsdtv/linux_media #260: TBS5520SE + Raspberry Pi 3 Model B Rev 1.2

jojoxyz, 2021-10-15:
2019-04-08-raspbian-stretch-lite  4.14.98-v7+

Upgrade required for mismatch headers when installing raspberrypi-kernel-headers
to  4.19.66-v7+

This is the only Rpi3 OS on which I managed at least minimal success, even if it did not work.

Is there any correction solution for my compilation that will also work?

Is there any OS for Rpi3 in which these drivers can be compiled and will I be functional?

I'm currently using LibreElec, it has built-in TBS drivers. But I need a system that is flexible and I can make changes and installations in it.

I found more instructions here than to correct rc error -22, but none worked.

############## I made ##############

sudo update
sudo full-upgrade
sudo reboot

sudo apt install raspberrypi-kernel-headers
sudo reboot

sudo apt install libproc-processtable-perl
sudo apt install build-essential
sudo apt install patchutils
sudo apt install git -y
sudo apt install gcc-4.9 g++-4.9 -y
sudo apt install mc -y
sudo apt install libncurses5-dev
sudo reboot

git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media

cd media_build
make dir DIR=../media
make allyesconfig
make                                      // make  -j2  or  -j4   internal compiler error
            
sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/*
sudo make install

wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
sudo tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

sudo reboot

dmesg | grep frontend          // no output
dmesg | grep TBS                 // no output

dmesg | grep -i dvb             // error

[    3.595130] dvb_usb: disagrees about version of symbol rc_register_device
[    3.595156] dvb_usb: Unknown symbol rc_register_device (err -22)
[    3.595284] dvb_usb: disagrees about version of symbol rc_free_device
[    3.595293] dvb_usb: Unknown symbol rc_free_device (err -22)
[    3.595342] dvb_usb: disagrees about version of symbol rc_allocate_device
[    3.595350] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[    3.595446] dvb_usb: disagrees about version of symbol rc_unregister_device
[    3.595454] dvb_usb: Unknown symbol rc_unregister_device (err -22)
[    3.998823] dvb_usb: disagrees about version of symbol rc_register_device
[    3.998840] dvb_usb: Unknown symbol rc_register_device (err -22)
[    3.998924] dvb_usb: disagrees about version of symbol rc_free_device
[    3.998932] dvb_usb: Unknown symbol rc_free_device (err -22)
[    3.999012] dvb_usb: disagrees about version of symbol rc_allocate_device
[    3.999020] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[    3.999162] dvb_usb: disagrees about version of symbol rc_unregister_device
[    3.999170] dvb_usb: Unknown symbol rc_unregister_device (err -22)
[... 3587 lines cut ...]
[    3.637418] dvb_usb: disagrees about version of symbol rc_allocate_device
[    3.637435] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[    3.637535] dvb_usb: disagrees about version of symbol rc_unregister_device
[    3.637543] dvb_usb: Unknown symbol rc_unregister_device (err -22)
[    4.102268] dvb_usb: disagrees about version of symbol rc_register_device
[    4.102286] dvb_usb: Unknown symbol rc_register_device (err -22)
[    4.102385] dvb_usb: disagrees about version of symbol rc_free_device
[    4.102393] dvb_usb: Unknown symbol rc_free_device (err -22)
[    4.102480] dvb_usb: disagrees about version of symbol rc_allocate_device
[    4.102488] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[    4.102643] dvb_usb: disagrees about version of symbol rc_unregister_device
[    4.102651] dvb_usb: Unknown symbol rc_unregister_device (err -22)

pi@raspberrypi:~ $ dmesg | grep frontend
pi@raspberrypi:~ $ dmesg | grep TBS
pi@raspberrypi:~ $ lsusb -vvv | grep 5521
Couldn't open device, some information will be missing
Bus 001 Device 006: ID 734c:5521 TBS Technologies China
  idProduct          0x5521
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
pi@raspberrypi:~ $

crazycat69, 2021-10-15:
https://github.com/tbsdtv/media_build/blob/extra/install.sh

jojoxyz, 2021-10-16:
Dakujem. Vsetko funguje.

closed 2021-10-16 by jojoxyz