tbsdtv/linux_media #265: TBS 6903X not recognized

ich777, 2021-11-01:
Hi,
got an issue with a TBS6903X that is not recognized on boot.

After issuing `modprobe -vvv tbsecp3` I get this output:

```
modprobe: INFO: custom logging function 0x40be50 registered
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/mc/mc.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/v4l2-core/videodev.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-common.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-memops.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/common/videobuf2/videobuf2-vmalloc.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/dvb-core/dvb-core.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/i2c/i2c-mux.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/dvb-frontends/tas2101.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/dvb-frontends/gx1133.ko.xz 
insmod /lib/modules/5.10.28-Unraid/kernel/drivers/media/pci/tbsecp3/tbsecp3.ko.xz 
modprobe: INFO: context 0x427500 released
```

The card is running on Kernel version 5.10.28, this was also tried on a Windows machine and the card is working there, has anyone a guess what could that be?

This is the output from `lspci` from this card:
```
07:00.0 Multimedia controller [0480]: TBS Technologies DVB Tuner PCIe Card [544d:6178]
	Subsystem: Device [6903:8888]
```

crazycat69, 2021-11-01:
wrong subsystem pci id
see correct in driver source
https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/pci/tbsecp3/tbsecp3-core.c#L340

ich777, 2021-11-01:
@crazycat69 what can I do so that it actually lists the correct subsystem pci id?
Nothing was changed on the system or modified, already rebooted a few times and nothing changed.

crazycat69, 2021-11-01:
this 6903X V1 or V2 ?
try add exntry for your card to **tbsecp3_id_table**:

TBSECP3_ID(TBSECP3_BOARD_TBS6903X,0x6903,0x8888),

ich777, 2021-11-02:
@crazycat69 added the pci id and it is working now and the card is recognized and working:

```
08:00.0 Multimedia controller [0480]: TBS Technologies DVB Tuner PCIe Card [544d:6178]
	Subsystem: Device [6903:8888]
	Kernel driver in use: TBSECP3 driver
	Kernel modules: tbsecp3
```

I've attached two pictures from the card, it is a TBS6903X **V2**:
[image]
:
[image]

Is there a chance you can add this pci id?

crazycat69, 2021-11-02:
Correct id must be 6903:0021 for this card.
How windows driver 3.0.0.1 recognize you card ?

ich777, 2021-11-02:
@crazycat69 the same as on Linux:
[image]

crazycat69, 2021-11-02:
and driver installed without warnings ?

ich777, 2021-11-02:
@crazycat69 I think, I had an "unknown device" in device manager and installed manually, pointing to the directory of the unpacked driver. After this, it installed without any problem.
Do you have a guess why it has the "wrong" id?

crazycat69, 2021-11-02:
see tbs6903x.inf, this contain only PCI\VEN_544D&DEV_6178&SUBSYS_00216903

ich777, 2021-11-03:
> I just had a Phone Call with TBS distibution in Germany to clarify what the problem with this card is. 
> And indeed, they already knew about the issue.
> There has been a mistake in the firmware flashing with the last productions series of theese cards. 
> As the very friendly guy at TBS could see by the serial number of my card, my card was the second in this series of cards with the wrong hardware-id. 
> He told me, that there will be a new driver with this hardware-id included. Perhaps there will be also a firmware update, that corrects the id. But this is not sure atm.

Source: [Click](https://forums.unraid.net/topic/99027-plugin-dvb-driver/?do=findComment&comment=1050233)

crazycat69, 2021-11-03:
tbsecp3 drivers have IOCTL support for SPI flash access  (FPGA firmware bitstream). 
I have FPGA firmare dumps for all 6903x revisions, so if you want dump your current firmare and change to proper 6903x firmware - send me PM or e-mail

ich777, 2021-11-05:
Thank you @crazycat69 wrote a mail to the mail address that is linked in your Github profile, I think this issue can be closed.

closed 2021-11-05 by ich777