tbsdtv/linux_media #266: Compiling problem on SLES 15 SP3

Martin-Weiss, 2021-11-12:
For some reason I can not compile and get this error:

```
  CC [M]  /root/tbs/media_build/v4l/vb2-trace.o
  CC [M]  /root/tbs/media_build/v4l/videobuf2-v4l2.o
  CC [M]  /root/tbs/media_build/v4l/videobuf2-memops.o
  CC [M]  /root/tbs/media_build/v4l/videobuf2-vmalloc.o
/root/tbs/media_build/v4l/videobuf2-vmalloc.c:341:3: error: 'const struct dma_buf_ops' has no member named 'map'; did you mean 'mmap'?
  .map = vb2_vmalloc_dmabuf_ops_kmap,
   ^~~
   mmap
/root/tbs/media_build/v4l/videobuf2-vmalloc.c:341:9: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .map = vb2_vmalloc_dmabuf_ops_kmap,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/tbs/media_build/v4l/videobuf2-vmalloc.c:341:9: note: (near initialization for 'vb2_vmalloc_dmabuf_ops.release')
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-5.3.18-59.27/scripts/Makefile.build:288: /root/tbs/media_build/v4l/videobuf2-vmalloc.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-5.3.18-59.27/Makefile:1675: _module_/root/tbs/media_build/v4l] Error 2
make[2]: *** [../../../linux-5.3.18-59.27/Makefile:179: sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-5.3.18-59.27-obj/x86_64/default'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/tbs/media_build/v4l'
make: *** [Makefile:26: all] Error 2

```

As I am not a developer I am seeking for some help, here..

Any ideas?

Martin-Weiss, 2021-11-12:
Feedback on my own -> had to remove the backport v5.5_vb2_kmap.patch and then the compile worked!

crazycat69, 2021-11-12:
look like not-LTS kernel with forced patches from 5.4+

Martin-Weiss, 2021-11-15:
> look like not-LTS kernel with forced patches from 5.4+

Yeah - the kernel has back-ported patches. Do you know if there is a way to detect if patches in backports are required during build and auto-select them?

crazycat69, 2023-03-23:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-23 by crazycat69