tbsdtv/linux_media #267: Kernel crash with TBS5930

OnlySlon, 2022-01-05:
Hello, the tbs5930 has just arrived
No problem with driver compilation and installation
OS: Ubuntu 20.04 LTS under ESXi,  Kernel: 5.4.0-91-generic

After 4-5 sec after last line of 'dmesg -w' linux rebooted. 

> 
> [   80.941012] usb 3-2: new high-speed USB device number 5 using xhci_hcd
> [   81.478120] usb 3-2: New USB device found, idVendor=734c, idProduct=5930, bcdDevice= 0.01
> [   81.478122] usb 3-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
> [   81.513778] mc: loading out-of-tree module taints kernel.
> [   81.513853] mc: module verification failed: signature and/or required key missing - tainting kernel
> [   81.514544] mc: Linux media interface: v0.10
> [   81.519175] videodev: Linux video capture interface: v2.00
> [   81.519176] WARNING: You are using an experimental version of the media stack.
>                 As the driver is backported to an older kernel, it doesn't offer
>                 enough quality for its usage in production.
>                 Use it with care.
>                Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
>                 af95da4670b9fc4cfc6443e0e62d90429bccee18 add support for tbs7230
> [   81.523442] WARNING: You are using an experimental version of the media stack.
>                 As the driver is backported to an older kernel, it doesn't offer
>                 enough quality for its usage in production.
>                 Use it with care.
>                Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
>                 af95da4670b9fc4cfc6443e0e62d90429bccee18 add support for tbs7230
> [   81.526158] WARNING: You are using an experimental version of the media stack.
>                 As the driver is backported to an older kernel, it doesn't offer
>                 enough quality for its usage in production.
>                 Use it with care.
>                Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
>                 af95da4670b9fc4cfc6443e0e62d90429bccee18 add support for tbs7230
> [   81.527950] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in cold state, will try to load a firmware
> [   81.529105] dvb-usb: downloading firmware from file 'dvb-usb-id5930.fw'
> [   81.529106] tbs5930: start downloading TBS5930 firmware
> [   81.565831] usb 3-2: USB disconnect, device number 5
> [   81.649005] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in warm state.
> [   81.649250] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
> [   81.649271] dvbdev: DVB: registering new adapter (TurboSight TBS 5930 DVB-S/S2/S2x)
> [   81.649273] tbs5930: read eeprom failed.
> [   81.649276] dvb-usb: MAC address reading failed.
> [   81.717047] m88rs6060 0-0069: found a 'Montage m88rs6060' in cold state
> [   81.717925] m88rs6060 0-0069: downloading firmware from file 'dvb-demod-m88rs6060.fw'
> [   81.718051] m88rs6060 0-0069: firmware did not run
> [   81.737027] usb 3-2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5930 DVB-S/S2/S2x)...
> [   81.737146] dvb-usb: TurboSight TBS 5930 DVB-S/S2/S2x successfully initialized and connected.
> [   81.737173] usbcore: registered new interface driver tbs5930
> [   81.737909] dvb-usb: TurboSight TBS 5930 DVB-S/S2/S2x successfully deinitialized and disconnected.
> [   84.397089] usb 3-2: new high-speed USB device number 6 using xhci_hcd
> [   84.934114] usb 3-2: New USB device found, idVendor=734c, idProduct=5930, bcdDevice= 0.01
> [   84.934116] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [   84.934117] usb 3-2: Product: TBS 5930
> [   84.934118] usb 3-2: Manufacturer: TBS-Tech
> [   84.955677] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in cold state, will try to load a firmware
> [   84.955680] dvb-usb: downloading firmware from file 'dvb-usb-id5930.fw'
> [   84.955681] tbs5930: start downloading TBS5930 firmware
> [   85.077135] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in warm state.
> [   85.077410] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
> [   85.077667] dvbdev: DVB: registering new adapter (TurboSight TBS 5930 DVB-S/S2/S2x)
> [   85.308024] dvb-usb: MAC address: 00:22:ab:c0:61:4e
> [   85.395934] m88rs6060 0-0069: found a 'Montage m88rs6060' in cold state
> [   85.396017] m88rs6060 0-0069: downloading firmware from file 'dvb-demod-m88rs6060.fw'
> [   86.916659] m88rs6060 0-0069: found a 'Montage m88rs6060' in warm state
> [   86.916661] m88rs6060 0-0069: firmware version:30
> [   86.934931] tmp = 0x1

crazycat69, 2022-01-22:
try without ESXi

crazycat69, 2023-03-23:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-23 by crazycat69