tbsdtv/linux_media #268: stv0900 causing system to lock requiring push of reset button.

bluzee, 2022-01-07:
Greetings Crazycat.  Happy new year. 
[Prof_7301.patch.txt](https://github.com/tbsdtv/linux_media/files/7831105/Prof_7301.patch.txt)

 I have TBS 5980 and Prof 7301 in my machine.  I installed tbsdtv driver to have dvb_usb_tbsqbox2ci.  For Prof 7301 tbsdtv loads stv0900 driver as it has always been mainline assigned driver for Prof 7301.  As you know stv0900 has never worked properly for Prof 7301 but mainline refuse any patches to fix or reassign 7301 to stv090x.  Mostly problem has been 7301 would stop locking some transponder after kernel upgrade.  Minor annoyance and reminder to apply patches and recompile kernel driver.  Now much more annoying problem.  stv0900 stuck in some endless loop that spams syslog two times per second.  On first tune attempt machine locks up tight and requires the reset button to be pushed.  Mint 20.2 Kernel 5.4.0-74-generic #83-Ubuntu SMP Sat May 8 02:35:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

As a convenience to users I wonder if you might consider switching Prof 7301 to stv090x in tbsdtv.     I have given up on linux mainline ever fixing Prof 7301 issue or merging TBS drivers into mainline.

crazycat69, 2022-01-10:
Done - 2c0740d1a219f97ccc502580514e285f433d04c3
And for Prof 7500 - 5b1cafe0dfbec022c1c4d4acdd2b406f9715f1e1

Happy New Year !

closed 2022-01-10 by crazycat69

manio, 2022-02-11:
@bluzee 
Nice that you are also using this card these days. I also have this card somewhere in a shelf :D
I also had the same problems after I bought it (official stv0900 was really unreliable) and that is why I spent a really lot of time debugging this card and as the result I've created a patch which is switching the frontend. I was trying the patch to be upstreamed but I was blocked by others :(
https://patchwork.kernel.org/project/linux-media/patch/4D3358C5.5080706@skyboo.net/
Nevertheless great to meet someone with the same card :)
Btw: I don't know if you are aware but I also helped to make a IR receiver to use in-kernel universal infrared receiver for this card, which is able to decode all protocols (not only the bundled-in) remote.

cheers!

bluzee, 2022-02-12:
Yes. Your patch was used in v4l-updatelee https://gitlab.com/updatelee/v4l-updatelee and now also crazycat has accepted it into tbsdtv.  Prof 7301 and other Prof devices work excellent with stv090x driver.  They are excellent devices.  There was a patch for stv0900 that fixed the issues but it was also rejected.  I used that patch for a couple years without issue.  I've been using v4l-updatelee driver for many years.  I have problems with inconsistent locking with v4l-updatelee and TBS 5980 though so switched to tbsdtv now for regular tv tuning.  stv0900 was written for NetUp devices. Prof devices should never have been attached to stv0900.

smp79, 2023-01-31:
> And for Prof 7500 - [5b1cafe](https://github.com/tbsdtv/linux_media/commit/5b1cafe0dfbec022c1c4d4acdd2b406f9715f1e1)

It doesn't work properly with Prof 7500.  Switching between different transponders is extremely slow.

bluzee, 2023-02-01:
I don't have a Prof 7500 but everyone I know who does had switched theirs to stv090x long ago. v4l-updatelee tree also has Prof 7500 switched to stv90x.   Did you revert the patch and test with stv0900?  stv0900 could never lock a transponder with a bitrate over 65Mbps.  In North America this is most transponders. stv090x worked.

smp79, 2023-02-01:
Default stv0900 driver works fine. For high bitrate transponder locking issue I use [this patch](https://rickcaylor.websitetoolbox.com/post/stv0900core-c-patch-5481028).
stv090x driver is usable but TV channel switching is very slow.

bluzee, 2023-02-01:
Yes, I used that patch until Prof devices were switched to stv090x in v4l-updatelee.  Prof 7301 has no issues zapping with stv090x.  This is the only time I've heard anyone say 7500 is slow switching with stv090x.  Maybe Crazycat will revert Prof 7500 and apply Pendragon's patch for you.  As I mentioned, stv0900 is a total no go with my 7301 now. Attempt to tune and it locks the whole machine up. Have to hit the reset button.

smp79, 2023-02-01:
I found the reason of the slow zapping on Prof 7500. Looks like Crazycat entered the wrong parameters for prof_7500_stv090x_config.

instead of:

	.ts1_mode               = STV090x_TSMODE_PARALLEL_PUNCTURED,
	.repeater_level         = STV090x_RPTLEVEL_64,

it should be:

	.ts1_mode                = STV090x_TSMODE_DVBCI,
	.ts2_mode                = STV090x_TSMODE_SERIAL_CONTINUOUS,
	.repeater_level         = STV090x_RPTLEVEL_16,

crazycat69, 2023-02-02:
STV090x_RPTLEVEL_64 present in original patch [Prof_7301.patch.txt](https://github.com/tbsdtv/linux_media/files/7831105/Prof_7301.patch.txt)

smp79, 2023-02-02:
That patch is for 7301.
For 7500 I set it to STV090x_RPTLEVEL_16 like in [tbs-qbox2.c](https://github.com/tbsdtv/linux_media/blob/82e72d47b794898843b980632c8543f5a34cf608/drivers/media/usb/dvb-usb/tbs-qbox2.c) and [tbs5925.c](https://github.com/tbsdtv/linux_media/blob/82e72d47b794898843b980632c8543f5a34cf608/drivers/media/usb/dvb-usb/tbs5925.c) and it works perfectly now.

crazycat69, 2023-02-02:
b1742b36688b64f187cd6076fb497b0e50a8cf46