tbsdtv/linux_media #269: No signal on DVB-C with MyGica T230A

haasviktor, 2022-01-13:
Hi,

I have MyGica T230A usb tuner, kernel has recognized it, but there's no signal on DVB-C.
Also, I have T230C tuner too and it's working perfectly.

Maybe this commit is not enough for T230A to work correctly with cable?
https://github.com/tbsdtv/linux_media/commit/387caebee00671
Parent issue: https://github.com/tbsdtv/linux_media/issues/224

Do you have any idea, what is the issue?

I use kernel 5.15, this is the log:
kernel: [1070396.570050] usb 1-7: new high-speed USB device number 7 using xhci_hcd
kernel: [1070396.719586] usb 1-7: New USB device found, idVendor=0572, idProduct=689a, bcdDevice= 8.00
kernel: [1070396.719607] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: [1070396.719614] usb 1-7: Product: USB Stick
kernel: [1070396.719620] usb 1-7: Manufacturer: Gen
kernel: [1070396.719625] usb 1-7: SerialNumber: 2017.12.09
kernel: [1070396.914072] usb 1-7: dvb_usb_v2: found a 'MyGica Mini DVB-(T/T2/C) USB Stick T230A' in warm state
kernel: [1070396.914214] usb 1-7: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
kernel: [1070396.914231] dvbdev: DVB: registering new adapter (MyGica Mini DVB-(T/T2/C) USB Stick T230A)
kernel: [1070396.914238] usb 1-7: media controller created
kernel: [1070396.915016] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
kernel: [1070396.926237] i2c i2c-1: Added multiplexed i2c bus 5
kernel: [1070396.926248] si2168 1-0064: Silicon Labs Si2168-D60 successfully identified
kernel: [1070396.926251] si2168 1-0064: firmware version: D 6.0.1
kernel: [1070396.930859] si2157 5-0060: Silicon Labs Si2141 successfully attached
kernel: [1070396.930939] usb 1-7: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
kernel: [1070396.930954] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
kernel: [1070396.931519] usb 1-7: dvb_usb_v2: 'MyGica Mini DVB-(T/T2/C) USB Stick T230A' successfully initialized and connected
kernel: [1070397.019034] si2168 1-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
kernel: [1070397.019047] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
kernel: [1070397.514364] si2168 1-0064: firmware version: D 6.0.13
kernel: [1070397.520274] si2157 5-0060: found a 'Silicon Labs Si2141-A10'
kernel: [1070397.520354] si2157 5-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
kernel: [1070397.520364] si2157 5-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
kernel: [1070397.783571] si2157 5-0060: firmware version: 1.1.11
tvheadend[584]: linuxdvb: adapter added /dev/dvb/adapter0
kernel: [1070397.786818] usb 1-7: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)

No signal found by w_scan:
w_scan -f c -c HU
w_scan version 20170107 (compiled for DVB API 5.11)
using settings for HUNGARY
Country identifier HU not defined. Using defaults.
scan type CABLE, channellist 4
output format vdr-2.0
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> CABLE "Silicon Labs Si2168": very good :-))

Using CABLE frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.11
frontend 'Silicon Labs Si2168' supports
INVERSION_AUTO
QAM_AUTO
FEC_AUTO
FREQ (48.00MHz ... 870.00MHz)
SRATE (1.000MSym/s ... 7.200MSym/s)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
474000: sr6900 (time: 00:01.287) sr6875 (time: 00:02.815)
482000: sr6900 (time: 00:06.107) sr6875 (time: 00:09.403)         signal ok:    QAM_AUTO f = 482000 kHz S6875C999  (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 482000 kHz S6875C999  (0:0:0))
490000: sr6900 (time: 00:13.131)         signal ok:     QAM_AUTO f = 490000 kHz S6900C999  (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 490000 kHz S6900C999  (0:0:0))
498000: sr6900 (time: 00:17.135) sr6875 (time: 00:20.439)
506000: sr6900 (time: 00:23.735)         signal ok:     QAM_AUTO f = 506000 kHz S6900C999  (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 506000 kHz S6900C999  (0:0:0))
514000: sr6900 (time: 00:26.135) sr6875 (time: 00:29.431)
522000: sr6900 (time: 00:32.723)         signal ok:     QAM_AUTO f = 522000 kHz S6900C999  (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 522000 kHz S6900C999  (0:0:0))
......

Tvheadend log with pre-defined channels:
tvheadend[584]: linuxdvb: Unhandled ERROR_BLOCK_COUNT scale: 0
tvheadend[584]: mpegts: 762MHz in DVB-C Network - scan no data, failed
tvheadend[584]: subscription: 284D: service instance is bad, reason: No input detected
tvheadend[584]: mpegts: 762MHz in DVB-C Network - tuning on Silicon Labs Si2168 #0 : DVB-C #0
tvheadend[584]: subscription: 284E: "epggrab" subscribing to mux "762MHz", weight: 4, adapter: "Silicon Labs Si2168 #0 : DVB-C #0", network: "DVB-C Network", service: "Raw PID Subscription"
tvheadend[584]: subscription: 284D: No input source available for subscription 
tvheadend[584]: subscription: 284D: No input source available for subscription 
tvheadend[584]: mpegts: 762MHz in DVB-C Network - scan no data, failed

crazycat69, 2022-01-13:
try mainline [media_build ](https://git.linuxtv.org/media_build.git)or latest kernel 5.15+

haasviktor, 2022-01-13:
I'm using Debian 5.15.5-2 kernel.

crazycat69, 2022-01-13:
T230A not supported in 5.15 kernel ?

haasviktor, 2022-01-14:
T230A is supported in 5.15
https://github.com/torvalds/linux/commit/387caebee00671
https://kernelnewbies.org/Linux_5.15#TV_tuners.2C_webcams.2C_video_capturers

closed 2022-01-19 by crazycat69

haasviktor, 2022-01-20:
Why is it closed? 
The issue still exists.

crazycat69, 2022-01-20:
I think LNA before tuner IC prevent DVB-C receiving. Check in windows

haasviktor, 2022-01-23:
Checked in Windows and found channels.
I used the same cable and tried with different cables too.
I have no issue with T230C2 and Astrometa, but with T230A there's no signal under Linux 5.15.5

tcmaps, 2022-02-08:
myGica was always weird on DVB-C. For my original T230, I need to "hotstart" it with progDVB before using DVBviewer, or it won't get signal also. Maybe something similiar is happening here?

nwrkbiz, 2022-05-28:
Same issue here.

technorior, 2023-01-02:
Same issue with T230C under 5.15.0-56-generic

AapoTahkola, 2023-05-25:
> I think LNA before tuner IC prevent DVB-C receiving. Check in windows

My T230A worked at 6.1.13 and 6.x series kernels. However few weeks ago I get no signal at all.
Now however if I connect VMware player and connect in windows and back, it starts working. I did not find channels or get picture in windows 10.
I think there is some initialization still missing. Does anyone want me to snoop usb traffic to see what the windows driver does?