tbsdtv/linux_media #271: Ubuntu 21.10 5.13.0-25-generic x86_64 si2183 problems

xlazom00, 2022-01-18:
I am on Ubuntu 21.10 with LINUX 5.13.0-25-generic x86_64
and I started to get this recently
si2183 4-0067: err setting tuner params
si2183 4-0067: set_params failed=-11

crazycat69, 2022-01-18:
what device ?

xlazom00, 2022-01-18:
@crazycat69 TBS6209 
I rebuilded older version of modules and all works fine

mthqwork, 2022-01-18:
Look back one issue. I have the same problem but with other title. What version you built that works? 
#270

xlazom00, 2022-01-19:
media_build e9f00f91c6e72bc9aa78d7542a933c3a79178d63
media 6873542771f5ae4aa5baea33529e742fec9ca0d7
this works for me

mthqwork, 2022-01-19:
Thanks, i will test it with Debian 11 and will back soon with the results.

crazycat69, 2022-01-19:
fixed 7c4fb702fd57679ce082cbb089ca307e3f9b0682

closed 2022-01-19 by crazycat69