tbsdtv/linux_media #274: Compile fails on ALPINE 5.15.16-0-virt

bolzerrr, 2022-01-29:
Build fails continuously with following message:

`/root/TBS/media_build/v4l/lm3560.c:343:21: error: variable 'lm3560_regmap' has initializer but incomplete type
  343 | static const struct regmap_config lm3560_regmap = {
      |                     ^~~~~~~~~~~~~
/root/TBS/media_build/v4l/lm3560.c:344:3: error: 'const struct regmap_config' has no member named 'reg_bits'
  344 |  .reg_bits = 8,
      |   ^~~~~~~~
/root/TBS/media_build/v4l/lm3560.c:344:14: warning: excess elements in struct initializer
  344 |  .reg_bits = 8,
      |              ^
/root/TBS/media_build/v4l/lm3560.c:344:14: note: (near initialization for 'lm3560_regmap')
`

crazycat69, 2023-03-23:
Updated, now compatible with 4.14 - 6.2

bolzerrr, 2023-05-19:
I tried it with the current active kernel (5.15.112-0-virt), but still fails:

```bash
/home/loot/tbs/media_build/v4l/max2175.c:237:34: error: array type has incomplete element type 'struct regmap_range'
  237 | static const struct regmap_range max2175_regmap_volatile_range[] = {
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/loot/tbs/media_build/v4l/max2175.c:238:2: error: implicit declaration of function 'regmap_reg_range'; did you mean 'remap_pfn_range'? [-Werror=implicit-function-declaration]
  238 |  regmap_reg_range(0x30, 0x35),
      |  ^~~~~~~~~~~~~~~~
      |  remap_pfn_range
/home/loot/tbs/media_build/v4l/max2175.c:244:21: error: variable 'max2175_volatile_regs' has initializer but incomplete type
  244 | static const struct regmap_access_table max2175_volatile_regs = {
      |                     ^~~~~~~~~~~~~~~~~~~
/home/loot/tbs/media_build/v4l/max2175.c:245:3: error: 'const struct regmap_access_table' has no member named 'yes_ranges'
  245 |  .yes_ranges = max2175_regmap_volatile_range,
      |   ^~~~~~~~~~
/home/loot/tbs/media_build/v4l/max2175.c:245:16: warning: excess elements in struct initializer
  245 |  .yes_ranges = max2175_regmap_volatile_range,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/loot/tbs/media_build/v4l/max2175.c:245:16: note: (near initialization for 'max2175_volatile_regs')
/home/loot/tbs/media_build/v4l/max2175.c:246:3: error: 'const struct regmap_access_table' has no member named 'n_yes_ranges'
  246 |  .n_yes_ranges = ARRAY_SIZE(max2175_regmap_volatile_range),
      |   ^~~~~~~~~~~~
In file included from ./include/linux/bits.h:22,
                 from ./include/linux/ratelimit_types.h:5,
                 from ./include/linux/printk.h:10,
                 from ./include/asm-generic/bug.h:22,
                 from ./arch/x86/include/asm/bug.h:86,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/loot/tbs/media_build/v4l/config-compat.h:12,
                 from /home/loot/tbs/media_build/v4l/compat.h:10,
                 from <command-line>:
```