tbsdtv/linux_media #275: [DOCUMENTATION] Installing the TBS5930 on Ubuntu 20.04

Josvth, 2022-01-30:
Hi all! Seeing others are having issues with getting the TBS5930 driver to work I though I would share my findings over the weekend to get it to work.

Using an older Linux kernel version is key to getting it to work. I did a fresh install of Ubuntu 20.04 and immediately downgraded the kernel to 5.4 using the ubunut-mainline-kernel.sh script from here: https://github.com/pimlie/ubuntu-mainline-kernel.sh or try [Mainline Ubuntu Kernel Installer](https://ubuntuhandbook.org/index.php/2020/08/mainline-install-latest-kernel-ubuntu-linux-mint/). 

The specific command I used to install the downgrade was `sudo ubuntu-mainline-kernel.sh -i 5.4.0`. I then switched to that Kernal from the `Advanced options for Ubuntu` menu during the GRUB startup.

I haven't tried higher kernel versions yet. I did notice that compiling the driver on one kernel version, then running a higher kernel version does not work. Probably because of the headers used.

Next you need to compile and install the drivers according to the wikipage: https://github.com/tbsdtv/linux_media/wiki

Be sure to install the pre-requisites (Kernel headers, patchutils, libproc, gcc and g++) following: https://github.com/tbsdtv/linux_media/wiki/Prerequisites,-Build-errors-and-Rebuilding

Do the git clones and build the drivers (I had to sudo everything):
```
cd media_build
sudo make dir DIR=../media
sudo make allyesconfig
sudo make -j4
sudo make install
```
Then reboot. After plugging in the usb device `dmesg` should show a log like this:
```
[   31.004236] usb 3-1: new high-speed USB device number 9 using xhci_hcd
[   31.152760] usb 3-1: New USB device found, idVendor=734c, idProduct=5930, bcdDevice= 0.01
[   31.152765] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   31.152768] usb 3-1: Product: TBS 5930
[   31.152771] usb 3-1: Manufacturer: TBS-Tech
[   31.153960] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in cold state, will try to load a firmware
[   31.153964] dvb-usb: downloading firmware from file 'dvb-usb-id5930.fw'
[   31.153966] tbs5930: start downloading TBS5930 firmware
[   31.264225] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in warm state.
[   31.264445] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   31.264797] dvbdev: DVB: registering new adapter (TurboSight TBS 5930 DVB-S/S2/S2x)
[   31.457573] dvb-usb: MAC address: 00:22:ab:c0:61:4d
[   31.545234] m88rs6060 10-0069: found a 'Montage m88rs6060' in cold state
[   31.545303] m88rs6060 10-0069: downloading firmware from file 'dvb-demod-m88rs6060.fw'
```
If the lines with `dvb-usb` do not show up the driver was not correctly installed / loaded.

Hope that helps someone!

crazycat69, 2022-01-30:
[media_build/build](https://github.com/tbsdtv/media_build/blob/extra/build) script always check required packages for building and give hint what need install.
**./build --check-only** - only check required packages,

need disable IR RC support for 5.8+ kernels, see [media_build/install.sh](https://github.com/tbsdtv/media_build/blob/extra/install.sh) script

If you want switch to ubuntu 20.04 GA kernel (5.4) - https://wiki.ubuntu.com/Kernel/LTSEnablementStack#Ubuntu_20.04_LTS_-_Focal_Fossa

Josvth, 2022-01-30:
Thanks @crazycat69 for the additional information. I'll try the build scripts again with the 5.8+ kernels some time later.

crazycat69, 2023-03-23:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-23 by crazycat69