tbsdtv/linux_media #276: Ubuntu 21.04 compile - other module fails - mceusb

ben-abolins, 2022-02-14:
Hello,

I am able to compile the TBS drivers and use the TBS devices. But, after I compile the TBS drivers the remote control module 'mceusb' stops working.

System: Ubuntu 21.04 AMD64
Linux ben-desktop 5.13.0-28-generic #31-Ubuntu SMP Thu Jan 13 17:41:06 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Thanks,
Ben

crazycat69, 2022-02-14:
try recompile and install with [install.sh](https://github.com/tbsdtv/media_build/blob/extra/install.sh) script

ben-abolins, 2022-02-16:
Thank you crazycat69,

I've tried to join together the earlier instructions for git/compile the TBS drivers and the install.sh file you sent.

Have I done this in the right order?

Thanks,
Ben

sudo apt install patchutils libproc-processtable-perl git
cd ~
sudo rm -r media
sudo rm -r media_build
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
####################NEW
make stagingconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j$(nproc)
sudo rm -r -f /lib/modules/$(uname -r)/updates/extra
sudo make install
#####################NEW
wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
sudo tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

crazycat69, 2022-02-16:
look good

crazycat69, 2023-03-23:
Updated, now compatible with 4.14 - 6.2