tbsdtv/linux_media #282: TBS 6981 not working/recognized, module missing...?

ich777, 2022-05-26:
Hi , I'm compiling the driver as instructed for Unraid but a user recently reported [here](https://forums.unraid.net/topic/99027-plugin-dvb-driver/?do=findComment&comment=1131011) that his TBS 6205 works with the TBS-OS package but his TBS 6981 won't work with it where it is listed as a supported device on the Wiki.

I also instructed him to do load the drivers manually [here](https://forums.unraid.net/topic/99027-plugin-dvb-driver/?do=findComment&comment=1131062) but this results in showing that no module is available:
```
modprobe cx23885
modprobe: FATAL: Module cx23885 not found in directory /lib/modules/5.10.28-Unraid 
```

Is there something that I'm missing that needs to be done for this specific driver?

crazycat69, 2022-05-26:
Look like kernel media drivers removed and new not installed.

P.S. 6981 supported in official V4L DVB drivers, included in kernels

ich777, 2022-05-26:
@crazycat69 I find no indication that it is compiled in the TBS-OS package: [log.txt](https://github.com/tbsdtv/linux_media/files/8779250/log.txt) sure, the firmware is added but in the build process for the TBS-OS package I don't see a indication that it is compiled (the build log is from yesterday).
 

> P.S. 6981 supported in official V4L DVB drivers, included in kernels

How can I compile the official V4L DVB drivers with this driver?

crazycat69, 2022-05-26:
This official V4L DVB + TBS additional drivers

ich777, 2022-05-27:
> In your case cx23885 not compiled, maybe some dependency problems

I have everything installed like mentioned in the Wiki.
How can I check what went wrong or why it isn‘t compiled, I even don‘t see any indication that it would compile but fails.

crazycat69, 2022-05-27:
check v4l/.config after build

ich777, 2022-05-30:
@crazycat69 I see that `# CONFIG_VIDEO_CX23885 is not set` but why is that the case, I doesn't changed anything and I build with the recommended parameters from the Wiki:
```
cd media_build
make dir DIR=../media
make allyesconfig
make -j4
sudo make install
```
Except for the CPU cores, I build with 12 instead of 4.

When I change `# CONFIG_VIDEO_CX23885 is not set` to `CONFIG_VIDEO_CX23885=m` and run the compilation again then it compiles the module just fine.

Also attached the output from `.config`:
```
CONFIG_IR_FINTEK=m
CONFIG_DVB_STB6100=m
CONFIG_IR_NUVOTON=m
CONFIG_VIDEO_FB_IVTV=m
# CONFIG_VIDEO_CX25821_ALSA is not set
# CONFIG_VIDEO_MEDIATEK_VCODEC is not set
CONFIG_VIDEO_MSP3400=m
# CONFIG_VIDEO_RENESAS_VSP1 is not set
CONFIG_VIDEO_TC358743=m
CONFIG_MEDIA_TUNER_TDA18212=m
# CONFIG_VIDEO_XILINX is not set
CONFIG_VIDEO_DW9768=m
# CONFIG_VIDEO_IMX7_CSI is not set
# CONFIG_RADIO_SF16FMR2 is not set
CONFIG_MEDIA_TUNER_RDA5816=m
CONFIG_USB_GSPCA_MR97310A=m
# CONFIG_VIDEO_RCAR_DRIF is not set
CONFIG_DVB_A8293=m
CONFIG_RC_LOOPBACK=m
CONFIG_DVB_DIB7000P=m
CONFIG_IR_SERIAL=m
CONFIG_SMS_USB_DRV=m
# CONFIG_DVB_C8SECTPFE is not set
CONFIG_USB_AIRSPY=m
CONFIG_VIDEOBUF2_MEMOPS=m
CONFIG_VIDEO_TW9906=m
CONFIG_DVB_LG2160=m
CONFIG_IR_JVC_DECODER=m
CONFIG_VIDEO_ADV7170=m
CONFIG_DVB_USB_ANYSEE=m
CONFIG_DVB_USB_DIBUSB_MB=m
CONFIG_VIDEOBUF2_DMA_CONTIG=m
CONFIG_VIDEO_CCS_PLL=m
CONFIG_DVB_GX1133=m
CONFIG_SAA716X_CORE=m
CONFIG_VIDEO_ADV7175=m
CONFIG_USB_GSPCA_ETOMS=m
CONFIG_VIDEO_ADV7343=m
# CONFIG_VIDEO_PXA27x is not set
CONFIG_VIDEO_FB_IVTV_FORCE_PAT=y
CONFIG_VIDEO_BT866=m
CONFIG_IR_STREAMZAP=m
# CONFIG_RADIO_SI476X is not set
CONFIG_RADIO_TEA5764=m
CONFIG_VIDEO_TW2804=m
# CONFIG_VIDEO_OV5640 is not set
# CONFIG_VIDEO_CX18_ALSA is not set
# CONFIG_CEC_GPIO is not set
CONFIG_DVB_VES1820=m
CONFIG_DVB_USB_RTL28XXU=m
CONFIG_DVB_OR51211=m
CONFIG_VIDEO_TEA6420=m
# CONFIG_DVB_SP8870 is not set
CONFIG_MEDIA_CAMERA_SUPPORT=y
CONFIG_DVB_STID135=m
CONFIG_DVB_DYNAMIC_MINORS=y
[... 789 lines cut ...]
CONFIG_MEDIA_ATTACH=y
CONFIG_DVB_TBSECP3=m
CONFIG_MEDIA_TUNER_FC0011=m
CONFIG_USB_PWC_INPUT_EVDEV=y
CONFIG_VIDEO_TVP7002=m
# CONFIG_IR_PWM_TX is not set
CONFIG_IR_ENE=m
# CONFIG_VIDEO_MAX9286 is not set
# CONFIG_VIDEO_HANTRO_ROCKCHIP is not set
CONFIG_VIDEO_OV2680=m
CONFIG_VIDEO_TVEEPROM=m
CONFIG_VIDEO_SAA7164=m
CONFIG_DVB_USB_TBSQBOX2=m
CONFIG_VIDEO_VIVID=m
CONFIG_VIDEO_AU0828_V4L2=y
# CONFIG_VIDEO_ZORAN_LML33 is not set
# CONFIG_VIDEO_OV7740 is not set
CONFIG_RADIO_SHARK=m
CONFIG_VIDEO_CX231XX_DVB=m
# CONFIG_VIDEO_EXYNOS4_IS_COMMON is not set
CONFIG_VIDEO_CPIA2=m
CONFIG_DVB_DDBRIDGE=m
CONFIG_DVB_MAX_ADAPTERS=64
CONFIG_VIDEO_VIVID_MAX_DEVS=64
```

ich777, 2022-06-07:
Did a workaround by editing the v4l/.config file before compilation.

closed 2022-06-07 by ich777