tbsdtv/linux_media #283: Module msi001 doesn't compile

paorie, 2022-06-01:
Archlinux 5.18.1-arch1-1
gcc version 12.1.0 (GCC)

/tbs-linux_media-git/src/media_build/v4l/msi001.c:504:27: error: initialization of 'void (*)(struct spi_device *)' from incompatible pointer type 'int (*)(struct spi_device *)' [-Werror=incompatible-pointer-types]
  504 |         .remove         = msi001_remove,
      |                           ^~~~~~~~~~~~~

paorie, 2022-06-02:
Ok with kernel 5.15.44-1-lts

tryptichon, 2022-06-17:
I have the same error on Fedora 36 after the lates kernel update as well:

```
root/git/media_build/v4l/msi001.c:504:27: error: initialization of 'void (*)(struct spi_device *)' from incompatible pointer type 'int (*)(struct spi_device *)' [-Werror=incompatible-pointer-types]
  504 |         .remove         = msi001_remove,
      |                           ^~~~~~~~~~~~~
/root/git/media_build/v4l/msi001.c:504:27: note: (near initialization for 'msi001_driver.remove')
```

```
Linux version 5.18.5-200.fc36.x86_64 (mockbuild@bkernel01.iad2.fedoraproject.org) (gcc (GCC) 12.1.1 20220507 (Red Hat 12.1.1-1), GNU ld version 2.37-27.fc36) #1 SMP PREEMPT_DYNAMIC Thu Jun 16 14:51:11 UTC 2022
```

tryptichon, 2022-06-22:
Me again. I hacked together a patch for the v4l kernel code. After applying this, the code was able to compile on kernel 5.18.5-200 from Fedora 36.

[kernel_v4l_update.zip](https://github.com/tbsdtv/linux_media/files/8954755/kernel_v4l_update.zip)

Unzip the file, put it in the directory next to the folder `media` and apply patch via 
`patch -p1 -i ../kernel_v4l_update.patch` in directory `media`

tryptichon, 2022-06-25:
Kernel 5.18.6-200 on Fedora 36 works as well with this patch. Just tested. :)

TBScardOwner, 2022-06-27:
Patch works fine on Opensuse Tumbleweed with vanilla kernel 5.18.6.
Thank you Tryptichon.

closed 2023-02-19 by paorie